h2zero/NimBLE-Arduino #274: Not sending all notifies

Bascy, 2021-08-04:
We are using a single BLE characteristic to send values of several sensors to an Android App. In normal operation this is working fine.

When a device connects to our BLEService, we send that data of all 21 sensors directly one after the other, all using `characteristic.setValue(...)` and `characteristic.notify()`. All sent values are small (5 - 7 bytes)

But somehow the BLE messages after the 15th sensor are not sent, adding up to just under 100 bytes before it stops. We've monitored the messages with an app like nRFConnect but no messages are received after the 15th.
Also there are no error or warning messages when we turn on logging in NimBLE.

We tried adding delays after sending each value but this doesnt change anything.

Is there some low level buffer that might be overflowing? Or does anyone has an idea what might be causing this?

h2zero, 2021-08-04:
Possibly overflowing the msys1 buffer. You can test by increasing the count in config, default is 12.

Bascy, 2021-08-04:
Great, Increasing to 20 solved the problem

Thanks for the rapid response

closed 2021-08-04 by Bascy

h2zero, 2021-08-04:
Glad to hear it worked. FYI the delays you inserted probably would have worked as well but they would need a duration of probably half the connection interval. Increasing the buffer count is the best option though, just uses more RAM.