h2zero/NimBLE-Arduino #275: Cannot Update Advertisement Data

juehv, 2021-08-09:
I'm trying to build a BLE server which provides some changing data with advertisements. I figured, that even though I'm trying to reset the advertisement every time I want to change the data, the new data gets appended and not replaced.

Here is my code
```
char data =0;
void loop() {
  Serial.println(data);
  NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(serviceUUID);
  pAdvertising->setServiceData(serviceUUID, std::string(data,1));
  
  pAdvertising->start();
  delay(5000);
  pAdvertising->removeServiceUUID(serviceUUID);
  pAdvertising->stop();
  pAdvertising->reset();
  data=data+1;
  delay(500);
}
```

I expected the service data to be changed at every 5 seconds. But what happens is that the broadcasted service data starts with "0x01" and every five seconds another "01" is appended until the length exceeds the allowed size.

Any idea what I did wrong? And how to achieve my goal?

h2zero, 2021-08-10:
Thank you for reporting this. The code you have should work as intended, I will have a better look tomorrow when I have some more time and get back to you.

sivar2311, 2021-08-22:
The issue is located here:
```C++
pAdvertising->setServiceData(serviceUUID, std::string(data,1));
```
You (accidentally) use the **fill constructor** of std::string which is defined as
```C++
std::string(size_t n, char c);
```
"Fills the string with n consecutive copies of character c."
And this is exactly what happens in your sketch.

Instead you need use the **from buffer** constructor which is defined as
```C++
std::string (const char* s, size_t n);
```
For this you need to pass in the address of your data and it's length like so:
```C++
pAdvertising->setServiceData(serviceUUID, std::string(&data,1));
```

For more information about std::string see [here](cplusplus.com/reference/string/string/string/)

Here is my test sketch:

```C++
#include <Arduino.h>
#include <NimBLEDevice.h>

NimBLEAdvertising* pAdvertising;
NimBLEUUID serviceUUID("BEEF");
char data = 0;

void setup() {
    Serial.begin(115200);

    NimBLEDevice::init("ESP32 Advertising test");
    pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(serviceUUID);
}

void loop() {
    Serial.printf("data: %d\r\n", data);
    pAdvertising->setServiceData(serviceUUID, std::string(&data, 1));
    pAdvertising->start();

    delay(5000);

    pAdvertising->stop();
    data++;
}
```

chegewara, 2021-08-22:
IMO the issue is not real issue. You just have tyo stop and re-start advertising after changing data.

sivar2311, 2021-08-22:
The issues is `std::string(data,1)` vs `std::string(&data,1)`
Even if you stop and restart the advertising (what i am actually do in my sketch above), the std::string's fill constructor is used and will lead to the same problem.

juehv, 2021-08-24:
@chegewara if you look at my code, you will see that the advertising *is* restarted after changing data.

@sivar2311 thank you very much for your comment! That solved the issue!

closed 2021-08-24 by juehv