h2zero/NimBLE-Arduino #276: Higher Flash and Ram usage than ESP32 BLE Arduino

Nailik, 2021-08-10:
Hi,
i switched to this library from ESP32 BLE Arduino 1.0.1 in order to have a smaller programm, but it increased:

**Before**
```
RAM:   [==        ]  15.2% (used 49884 bytes from 327680 bytes)
Flash: [=====     ]  54.3% (used 1138746 bytes from 2097152 bytes)
```

<details>
  <summary>Full Output Before</summary>

```
C:\Users\kilia\.platformio\penv\Scripts\platformio.exe -c clion run --target upload -e featheresp32
Processing featheresp32 (platform: espressif32; board: featheresp32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (3.3.1) > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h,
 olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit LED Backpack Library> 1.1.8
|   |-- <Wire> 1.0.1
|   |-- <Adafruit GFX Library> 1.10.10
|   |   |-- <Adafruit BusIO> 1.8.2
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.8.2
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.8.5
|-- <HX711> 0.7.4
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.3
|-- <ESP32 BLE Arduino> 1.0.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Adafruit GFX Library> 1.10.10
|   |-- <Adafruit BusIO> 1.8.2
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
[... 684 lines cut ...]
Writing at 0x0010d280... (77 %)
Writing at 0x0011321f... (79 %)
Writing at 0x00118e58... (81 %)
Writing at 0x0011f5ed... (83 %)
Writing at 0x00128005... (85 %)
Writing at 0x0012f9a1... (87 %)
Writing at 0x00135a1c... (89 %)
Writing at 0x0013b84b... (91 %)
Writing at 0x001423ef... (93 %)
Writing at 0x00149368... (95 %)
Writing at 0x0014eec1... (97 %)
Writing at 0x001549bc... (100 %)
Wrote 1342816 bytes (778204 compressed) at 0x00010000 in 20.1 seconds (effective 534.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================= [SUCCESS] Took 86.46 seconds =============================================

Process finished with exit code 0
```

</details>

Hope you can help me decrease the size.

h2zero, 2021-08-10:
This is quite surprising. May I ask what you're including in your application? Also what arduino core version you are compiling with?

thorrak, 2021-08-11:
Are you actually using the libraries in your sketch? That flash/ram usage seems lighter than some sketches I’ve built that don’t use either radio…

Nailik, 2021-08-11:
Those are the libs i'm using:
```
adafruit/Adafruit LED Backpack Library @ ^1.1.8
	adafruit/Adafruit BusIO@^1.7.5
	adafruit/Adafruit NeoPixel @ ^1.8.5
	bogde/HX711@^0.7.4
	me-no-dev/ESP Async WebServer @ ^1.2.3
```

And yes i'm using all the libraries like the Bluetooth connection, webserver, dns and also the neopixel and hx711 and led backpack.

And the dependency graph (in the Spoiler) shows:

```
|-- <Adafruit LED Backpack Library> 1.1.8
|   |-- <Wire> 1.0.1
|   |-- <Adafruit GFX Library> 1.10.10
|   |   |-- <Adafruit BusIO> 1.8.2
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.8.2
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit NeoPixel> 1.8.5
|-- <HX711> 0.7.4
|-- <NimBLE-Arduino> 1.3.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.3
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Adafruit GFX Library> 1.10.10
|   |-- <Adafruit BusIO> 1.8.2
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <ArduinoJson> 6.18.3
|-- <Preferences> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <AsyncTCP> 1.1.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
```

I'm not sure about the arduino core version since i'm using Plaformio V 5.1.1, maybe this helps.

```
pio platform show espressif32
espressif32 ~ Espressif 32
==========================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: 3.3.1
Home: https://espressif.com/
Repository: https://github.com/platformio/platform-espressif32.git
Vendor: https://espressif.com/
License: Apache-2.0
Frameworks: arduino, espidf, pumbaa, simba

Packages
--------

Package toolchain-xtensa32
--------------------------
Type: toolchain
Requirements: ~2.50200.0
Installed: Yes
Version: 2.50200.97
Original version: 5.2.0
Description: GCC Toolchain for Xtensa32 processor

Package toolchain-xtensa32s2
----------------------------
Type: toolchain
Requirements: ~1.80400.0
Installed: No (optional)

Package toolchain-esp32ulp
--------------------------
Type: toolchain
Requirements: ~1.22851.0
Installed: No (optional)

Package toolchain-esp32s2ulp
----------------------------
Type: toolchain
Requirements: ~1.22851.0
Installed: No (optional)

Package toolchain-riscv-esp
---------------------------
Type: toolchain
Requirements: ~1.80400.0
Installed: No (optional)

Package framework-arduinoespressif32
------------------------------------
Type: framework
Requirements: ~3.10006.0
Installed: Yes
Version: 3.10006.210326
Original version: 1.0.6
Description: Arduino Wiring-based Framework for Espressif ESP32 microcontrollers

Package framework-arduino-mbcwb
-------------------------------
Type: framework
Requirements: >=2.1.1
Installed: No (optional)

Package framework-espidf
------------------------
Type: framework
Requirements: ~3.40300.0
Installed: No (optional)

Package framework-simba
-----------------------
Type: framework
Requirements: >=12.2.0
Installed: No (optional)

Package framework-pumbaa
------------------------
Type: framework
Requirements: >=2.3.0
Installed: No (optional)

Package tool-esptoolpy
----------------------
Type: uploader
Requirements: ~1.30100.0
Installed: Yes
Version: 1.30100.210531
Original version: 3.1.0
Description: Espressif ESP8266 and ESP32 serial bootloader utility

Package tool-mbctool
--------------------
Requirements: >=2.0.0
Installed: No (optional)

Package tool-openocd-esp32
--------------------------
Type: debugger
Requirements: ~2.1000.0
Installed: Yes
Version: 2.1000.20210721
Original version: 10.0
Description: Open On-Chip Debugger for Espressif ESP32

Package tool-mkspiffs
---------------------
Type: uploader
Requirements: ~2.230.0
Installed: Yes
Version: 2.230.0
Original version: 2.30
Description: Tool to build and unpack SPIFFS images

Package tool-cmake
------------------
Requirements: ~3.16.0
Installed: No (optional)

Package tool-ninja
------------------
Requirements: ^1.7.0
Installed: No (optional)

Package tool-mconf
------------------
Requirements: ~1.4060000.0
Installed: No (optional)

Package tool-idf
----------------
Requirements: ~1.0.1
Installed: No (optional)
```

h2zero, 2021-08-11:
Thanks for the info, looks like 1.0.6 for the arduino core. Unfortunately it doesn't look like PlatformIO provides a component size output command as IDF does, so that won't be able to answer the issue. I wonder if when you include the original BLE library that one of the others isn't included. 

From the list of libraries you are using the code size looks appropriate with NimBLE included.

closed 2022-01-15 by h2zero