h2zero/NimBLE-Arduino #277: cambio de libreria de ble a nimble

roger4-77, 2021-08-30:
Hi, I'm new to this, I would appreciate some help, I can't get this program to go with the new nimble library, 

bool connectToServer1(BLEAddress pAddress)
{
  M5.Lcd. setCursor (1, 200);
      M5.Lcd. setTextColor (BLUE);
      M5.Lcd. setTextSize ( 3);
      M5.Lcd.print("CONECTANDO RELE1"); 
  BLEClient*  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");
  
  pClient->setClientCallbacks(new MyClientCallback());

  if (!pClient->connect(pAddress)) {
     
    ESP.restart();
  } else {
    Serial.println(" - Connected to Flora");
   M5.Lcd. fillRect (1, 200,290,40,BLACK);
  }

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  else
  {
    Serial.println(" - Found our service");
  }

  // Obtain a reference to the characteristic in the service of the remote BLE server.

  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  uint8_t buf[4] = {0xA0, 0x01, 0x01, 0xA2};
  pRemoteCharacteristic->writeValue(buf, 4, true);

  relec1 = HIGH;
  grabarvariableeeprom1();
  delay(200);
  pClient->disconnect();

  // connected = false;
   M5.Lcd.fillScreen(BLACK);
      M5.Lcd. setCursor (1, 120);
      M5.Lcd. setTextColor (YELLOW);
      M5.Lcd. setTextSize ( 4);
      M5.Lcd.print("RELE1 ON");
  return true;

}
I have tried everything by copying from other programs and reading the library but I can't
Thanks in advance

h2zero, 2021-08-30:
Hi, you didn't specify the error you are having but I can point out that since you are creating a new client on each cal to this function you should delete it instead of only disconnecting. Just change the calls `pClient->disconnect();` to `NimBLEDevice::deleteClient(pClient); and that will stop related errors to that. 

Another potential issue is the address you are providing to the function, where are you getting it from? If hard coded you need to make sure you have the address type correct, default is 0 but some devices use 1. If the address comes from a scanned `advertisedDevice` then there should be no issue with this.

roger4-77, 2021-08-30:
Thanks for answering, the problem is that this program when I turn it to nimble, it connects me but not the service or the rest

roger4-77, 2021-08-30:
sorry I closed the thread

h2zero, 2021-08-30:
Ok, thanks. What are the values of `serviceUUID` and `charUUID`?

roger4-77, 2021-08-30:
// The remote service we wish to connect to.
static BLEUUID serviceUUID("0000FFE0-0000-1000-8000-00805F9B34FB");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("0000FFE1-0000-1000-8000-00805F9B34FB");

static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEClient* pClient;
static BLEAddress floraAddress(rele);

h2zero, 2021-08-30:
> // The remote service we wish to connect to.
static BLEUUID serviceUUID("0000FFE0-0000-1000-8000-00805F9B34FB");
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID("0000FFE1-0000-1000-8000-00805F9B34FB");

Looks like the device does not respond to 128 bit base uuid construction, instead you could try using the 16bit version:
```
// The remote service we wish to connect to.
static BLEUUID serviceUUID("FFE0");
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID("FFE1");
``` 

I will update the code in the future to try this automatically if that fixes this issue.

roger4-77, 2021-08-30:
sorry but I don't understand with the ble library it worked for me and in the example of the nimble the old bleclient comes out and it is also 128bits and thank you very much for answering

h2zero, 2021-08-30:
When requesting a service/characteristic with a uuid it sends the request to the device and if it has that uuid in it's database it will respond with the access handle. In this case the device doesn't use the 128 bit version of the handle that NimBLE sends and doesn't account for the base uuid internally so it responds false. 

In the bluedriod stack I suspect it would do the conversions internally in this situation, the NimBLE stack does not so I would add it into the library to account for this situation. This has been an issue in the past but in the opposite. Let me know if the change works for you.

roger4-77, 2021-08-30:
it work

/** NimBLE_Server Demo:
 *
 *  Demonstrates many of the available features of the NimBLE client library.
 *
 *  Created: on March 24 2020
 *      Author: H2zero
 *
*/
#include <M5Stack.h>
#include <NimBLEDevice.h>
#define rele  "F8:30:02:23:5B:41" //FLORA ADDR1

// The remote service we wish to connect to.0000FFE0-0000-1000-8000-00805F9B34FB
 //static NimBLEUUID serviceUUID("0000FFE0-0000-1000-8000-00805F9B34FB");
// The characteristic of the remote service we are interested in.
//static NimBLEUUID    charUUID("0000FFE1-0000-1000-8000-00805F9B34FB");

// The remote service we wish to connect to.
static BLEUUID serviceUUID("FFE0");
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID("FFE1");

 #include <NimBLEDevice.h>
static NimBLERemoteCharacteristic* pRemoteCharacteristic;
static NimBLEClient* pClient;

void setup() {
// put your setup code here, to run once:
Serial.begin(115200);
Serial.println("Starting NimBLE Client");
NimBLEDevice::init("");
NimBLEDevice::setPower(ESP_PWR_LVL_P9);

pClient = NimBLEDevice::createClient(NimBLEAddress("F8:30:02:23:5B:41"));
Serial.println("New client created");
pClient->connect();
Serial.print("Connected to: ");
Serial.println(pClient->getPeerAddress().toString().c_str());
Serial.print("RSSI: ");
Serial.println(pClient->getRssi());

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
   
  }
  Serial.println(" - Found our service");

  // Obtain a reference to the characteristic in the service of the remote BLE server.
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed pRemoteCharacteristic ");
    Serial.println(charUUID.toString().c_str());
   
  }
  else
  {
    Serial.println(" - Found Characteristic");
  }
  

  uint8_t buf2[4] = {0xA0,0x01,0x00,0xA1};
  pRemoteCharacteristic->writeValue(buf2, 4, true);
  
  //pClient->disconnect();
}

void loop() {
if (!pClient->isConnected()) {
    pClient->connect();
} else {
   Serial.println("Connected");
}
delay(1000);

}

but the order to turn on the relay does not work
uint8_t buf2[4] = {0xA0,0x01,0x00,0xA1};
  pRemoteCharacteristic->writeValue(buf2, 4, true);

roger4-77, 2021-08-30:
Yes that works, you are the best, I have been with this for 2 days but since I do not know, it was what you said.

// The remote service we wish to connect to.0000FFE0-0000-1000-8000-00805F9B34FB
 //static NimBLEUUID serviceUUID("0000FFE0-0000-1000-8000-00805F9B34FB");
// The characteristic of the remote service we are interested in.
//static NimBLEUUID    charUUID("0000FFE1-0000-1000-8000-00805F9B34FB");

// The remote service we wish to connect to.
static BLEUUID serviceUUID("FFE0");
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID("FFE1");

Muchas Gracias
Thank you so much

h2zero, 2021-08-31:
You're very welcome :) thanks for reporting back. I will make a note to make this automatic in the library in the future.

h2zero, 2022-04-04:
Closing as resolved. Latest release incorporates rechecking with 16bit uuid.

closed 2022-04-04 by h2zero