h2zero/NimBLE-Arduino #278: m5.begin () conflict with  BLEDevice :: init ("");

roger4-77, 2021-08-31:
good there is conflict between m5.begin () and BLEDevice :: init ("");
if I put both in setup, the esp does not continue, it does nothing

h2zero, 2021-08-31:
I don't know what m5.begin() does but I'm assuming here that it is a blocking call which does not return. 

If that is the case you could try calling it after NimBLEDevice::init().

roger4-77, 2021-08-31:
I tried it, I also tried to put the order in void loop and to continue but it does not work, it is for the m5stack to start.

chegewara, 2021-08-31:
`NimBLEDevice::init()` != `BLEDevice::init()`

roger4-77, 2021-08-31:
I do not understand. Do I have to put that or do you tell me that one is not the same as another? I have tried it with bledevice and with nimbledevice

h2zero, 2021-08-31:
Either will work actually, I have aliased all the `BLE*` classes to `NimBLE*` so that would not be the issue here.

h2zero, 2021-08-31:
I looked at the m5.begin code and it starts serial for you but you also have the call to `Serial.begin()` in your code. This may be the cause of your issue. Try removing the call to `Serial.begin()` in your code and let me know.

roger4-77, 2021-08-31:
YES IT HAS WORKED

void setup() 
{
  
// Serial.begin(115200);

NimBLEDevice::init("");
//NimBLEDevice::setPower(ESP_PWR_LVL_P9);
 M5.begin();
  M5.Power.begin();
}

BUT BEFORE WITH LIBRARY BLE IT WORKED

#include "BLEDevice.h"

void setup()
{
  Serial.begin(115200);
  delay(200);
  M5.begin();
  M5.Power.begin();

  Serial.println("EEPROM LEYENDO ");
  delay(200);

  M5.Lcd.fillScreen(BLACK);
  
  BLEDevice::init("");
  }
THANK YOU SO MUCH

h2zero, 2021-08-31:
Your welcome 😄. Strange that the old code worked like that, at least it's resolved now.

closed 2022-01-15 by h2zero