h2zero/NimBLE-Arduino #280: l2cap -> iPhone

SMorozovMinsk, 2021-09-01:
Hi ,
And sorry for this question. I'm new in Arduino programming. (But I have 20+ years of experience in C/ObjC/Swift for ios and others)

Can I use your fantastic lib to establish connection with iOS via l2cap? On the iOS side I can use special api for l2cap BLE connection. It work's fine and I can connect iPhone to iPhone via l2cap. And it shows higher speed for data transfer. Now we have an idea to use it for iPhone to ESP32 Arduino connection and data transfer between ESP32 and iPhone.
Is it possible to use NimBLE for this? May be you have some simple code for this? 

Thank you!

h2zero, 2021-09-01:
I think this is possible but I have not attempted it. The NimBLE stack code is present to work with l2cap so you would need to use the lower level functions for this as there is no CPP wrapper for them. 

The header for these functions is here https://github.com/h2zero/NimBLE-Arduino/blob/master/src/host/ble_l2cap.h. I'd be interested in hearing of your results if you try it.

ClaesClaes, 2021-09-02:
Hi, @SMorozovMinsk 
I'm using the iOS CoreBluetooth framework for BLE communication between an iPhone and an ESP32. I currently reach 32kB/s. According to this article (https://punchthrough.com/ble-throughput-part-4/) around 50kB/s should be the theoretical maximum throughput.

I would be very interested increasing my throughput so could you tell me what transfer speed you achieve with your phone-to-phone using your special L2CAP special api. Also, is this api open source and/or part of the iOS CoreBluetooth framework?

SMorozovMinsk, 2021-09-02:
@h2zero Thank you for your help! Yes I will try this and will bak to you with any results and code example.

SMorozovMinsk, 2021-09-02:
@ClaesClaes Hmmm I'm checked. iPhone to iPhone via L2CAP 28kB/s only. 1024 kB for 36 seconds. Could you share some code for 32kB/s?
2021-08-12 10:46:26 +0000
sending 1024 kilobytes start
2021-08-12 10:46:26 +0000
bytesWritten = 8192
Space is available
...
2021-08-12 10:47:02 +0000
bytesWritten = 1933
Space is available

36 seconds

ClaesClaes, 2021-09-03:
@SMorozovMinsk Ok 28kB/s is lower than what I expected, but then on the other hand I've not seen any consensus on how/why use of L2CAP and not the the higher layers would allow for higher speeds. 

For code, I've nothing I can show as I would need to extract it from the real project into something workable. However you could start by looking at an example I made for OTA file transfer (from iPhone to ESP32 BLE 4.2) here: 

[https://github.com/ClaesClaes/Arduino-ESP32-BLE-OTA-iOS-SwiftUI]

Pls note: The repository does not use NimBLE but worked fine. However after changing to NimBLE I had to make some changes and I also added a generic file transfer function. If I get time I can try to extract the relevant parts (both Arduino and Swift) into something you can look at. In the meantime shoot any Qs you might have in a DM to me as this might go off topic for this issue.

h2zero, 2021-11-21:
@ClaesClaes Sorry for the late response, I think this would benefit you without using L2CAP #283. I've seen others get above 35kB/s by using this.

closed 2026-03-17 by h2zero