h2zero/NimBLE-Arduino #281: List of ble devices in area

svdrummer, 2021-09-06:
Hello,
I have read the docs and they seem to have a lot of other calls which look for service data etc.

All i want to do is print a list of devices (beacons) around me.

I have the other ble library(arduino default library) working, however, just the basic library is using 76% on my flash (crazy)
I am looking to operate a relay when a stored beacon is near-by, so no one can steal a wheelchair.

Thank you.

h2zero, 2021-09-06:
Hi @svdrummer, for your simple use case I would suggest looking at the scan continuous example [here] (https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Scan_Continuous/NimBLE_Scan_Continuous.ino).

To that I would make a few changes,
- remove the call: `NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);`
- remove the call: `NimBLEDevice::setScanDuplicateCacheSize(200);`
- change `pBLEScan->setActiveScan(true);` to `pBLEScan->setActiveScan(false);`

That should be the simplest way to do what you need.

svdrummer, 2021-09-06:
Thank you so much. 
I am wanting to print the Address name only, Is that in a library file?
THank you

h2zero, 2021-09-07:
You're welocme!

Yes, you can change the line:
` Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());`

To:
`Serial.printf("Advertised Device: %s \n", advertisedDevice->getAddress().toString().c_str());`

That will print only the address of the devices.

svdrummer, 2021-09-07:
Your time is really appreciated and valued. Thank you.

h2zero, 2021-09-07:
No problem, hope it works well for you!

svdrummer, 2021-09-07:
Seems to scan, detect all the beacons, then stops for a long time not printing anything. Ie i can.t detect a loss of a beacon.

svdrummer, 2021-09-07:
Seems to scan, detect all the beacons, then stops for a long time not printing anything. Ie i can.t detect a loss of a beacon. Not locking up, just stopping for 20 seconds. When i scan the list.
I have a for loop which compares the beacon addresses, then sets a timer. Because the scanning seems to stop printing, my beacons timeout

h2zero, 2021-09-07:
Sounds like you want to turn off the duplicate filter. You can do this by calling `pScan->setDuplicateFilter(false);`,  let me know if that helps.

svdrummer, 2021-09-07:
I have tried all. Even the basic "scan" times out after about 30 mins.

The following works, it just missed a beacon that is set to advertise every three seconds. The beacon does get picked up most of the time. Sometimes it is missed.
I have added a watchdog timer to help with the lockups.

Here is the code so far.

``

/** Example of continuous scanning for BLE advertisements.
   This example will scan forever while consuming as few resources as possible
   and report all advertisments on the serial monitor.

   Created: on January 31 2021
        Author: H2zero

*/
//#########################################
#include <Ticker.h>
#include "NimBLEDevice.h"
//###### Beacon address added here ##############
String knownAddresses[] = {
  "84:cc:a8:5f:af:46",
  "e3:78:06:79:9f:c7",
  "me8:06:2a:65:0a:c5",
  "mc2f:58:fe:6f:a6:7d",
  "mfc:f1:36:5c:2a:4d",
  "m1b:22:33:57:ba:a4"
};
//##### local name added here ##############
String beaconnames[] = {
  "ESP32",
  "Versa 2",
  "Mod3",
  "Mod4",
  "Mod5",
  "Mod6"
};
int beacontimes[] = {0, 0, 0, 0, 0, 0};
String oldtesting = "";// added to filter same address being printed in a row
NimBLEScan* pBLEScan;
String testing = "";
int watchdogCount = 0;
Ticker secondTick;// ISR for seconds ticker

//#########################################################
//########### Watchdog ########### Watchdog ###############
void ISRwatchdog() {// keep things quick in here
  watchdogCount++;
  beacontimes[0]++;
  beacontimes[1]++;
  if (watchdogCount > 2) {
    Serial.println("/n You tripped a Watchdog!!/n");
  }
}/// end watchdog

//#########################################################
//########### MyAdvertisedDeviceCallbacks ###############
class MyAdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
      //Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
      // Serial.printf("Advertised Device: %s \n", advertisedDevice->getAddress().toString().c_str());////////////////////////////////////////
      String testing = (advertisedDevice->getAddress().toString().c_str());
      //      if (testing.equals(oldtesting)) {
      //        Serial.print("same");
      //      } else {
      //        oldtesting = testing;
      //        Serial.print("New ");
      //        Serial.println (testing);
      //      }
      for (int i = 0; i <= 6; i++) {
        if (testing.equals(knownAddresses[i])) {
          Serial.print("Found ");
          Serial.println(beaconnames[i]);
          beacontimes[i]=0;
        } else {
          //          // Serial.println (testing);
        }
      }
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");
  secondTick.attach(1, ISRwatchdog);
  /** *Optional* Sets the filtering mode used by the scanner in the BLE controller.

      Can be one of:
      CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE (0) (default)
      Filter by device address only, advertisements from the same address will be reported only once.

      CONFIG_BTDM_SCAN_DUPL_TYPE_DATA (1)
      Filter by data only, advertisements with the same data will only be reported once,
      even from different addresses.

      CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE (2)
      Filter by address and data, advertisements from the same address will be reported only once,
      except if the data in the advertisement has changed, then it will be reported again.

      Can only be used BEFORE calling NimBLEDevice::init.
  */
  //NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);////////////////////////////////////////////////////////////////////

  /** *Optional* Sets the scan filter cache size in the BLE controller.
      When the number of duplicate advertisements seen by the controller
      reaches this value it will clear the cache and start reporting previously
      seen devices. The larger this number, the longer time between repeated
      device reports. Range 10 - 1000. (default 20)

      Can only be used BEFORE calling NimBLEDevice::init.
  */
  // NimBLEDevice::setScanDuplicateCacheSize(200);//////////////////////////////////////////////////////////////////////////////

  NimBLEDevice::init("");
 //NimBLEScanResults results = pBLEScan->start(10);
  pBLEScan = NimBLEDevice::getScan(); //create new scan
  // Set the callback for when devices are discovered, no duplicates.
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), false);
  //pBLEScan->setActiveScan(true); // Set active scanning, this will get more data from the advertiser.
  pBLEScan->setActiveScan(false); //////////////////////////////////////////////////////////////////////////////////////////////
  pBLEScan->setInterval(97); // How often the scan occurs / switches channels; in milliseconds,
  pBLEScan->setWindow(37);  // How long to scan during the interval; in milliseconds.
  pBLEScan->setMaxResults(0); // was zerodo not store the scan results, use callback only.
  pBLEScan->setDuplicateFilter(false);
  
}

void loop() {
  // If an error occurs that stops the scan, it will be restarted here.
  watchdogCount = 0;
  if (pBLEScan->isScanning() == false) {
    // Start scan with: duration = 0 seconds(forever), no scan end callback, not a continuation of a previous scan.
    pBLEScan->start(0, nullptr, false);
  }
 // for (int i=0; i <= 1; i++) {
    if (beacontimes[0] > 10) {
      Serial.print(beaconnames[0]);
      Serial.println(" Beacon timeout");
    }
 // }
   delay(1000);
}
``

h2zero, 2021-09-07:
Sorry for your trouble, could you try setting the debug log level to "Debug" and provide the log output when it locks up?

svdrummer, 2021-09-07:
Where is debug level set? The esp32 /arduino IDE does not allow for a debug option for the esp32

h2zero, 2021-09-07:
This library works off the core debug level setting in Arduino IDE. Go to tools->Core Debug Level->select "Debug".

svdrummer, 2021-09-07:
I was using wemos32. No debug level there. I changed chip to ESP32 DEV1 and i can now see the debug option.
I was using core package 1.0.6 the other BLE library that comes with Arduino, has suggested to another user to switch back to board version 1.0.2. I will try both. Maybe there is a core issue.

h2zero, 2021-09-07:
It should work with any core but I'd be curious if you did find an issue in one. Glad you found the log settings, hopefully it'll point to the problem.

svdrummer, 2021-09-07:
Whilst the esp core 1.0.6 shows more data in the debug.
[BLEScan.cpp:72] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT seems to show up a lot in V1.0.6 and nothing in 1.0.2
From the code i am running point of view, my code above seems to miss quite a few beacon transmissions. Maybe i have too much going on in the callback or not scanning long enough.

svdrummer, 2021-09-08:
This is from the default library listing in Arduino. The only library i have selected is
 #include <Ticker.h>
#include "NimBLEDevice.h"
Anything I can fix?

h2zero, 2021-09-08:
I'll do some testing on this and get back to you shortly.

h2zero, 2021-09-08:
I've been testing this for 2 hours with 1.0.6 and have not seen any issues. I'll let it run overnight and check again.

svdrummer, 2021-09-08:
which code are you running, maybe i have to remove the  #include "NimBLEDevice.h" and replace with #include <NimBLEDevice.h>
I may have a library conflict

closed 2021-09-08 by svdrummer