h2zero/NimBLE-Arduino #282: LE Data Length Extension

Mr-Mime, 2021-09-06:
I have just asked myself if there is a specific reason, why Data Length Extension is not implemented / part of the library.

I see it is implemented in the espressif nimble fork.
https://github.com/espressif/esp-nimble/blob/94afe2711752ba08bef93780be1b9e5143a87159/nimble/host/src/ble_gap.c#L5615-L5618

And also is part of the official nimble stack
https://github.com/apache/mynewt-nimble/blob/87f1d1da6cabc1c5ded863544d2e4df97a3003b3/nimble/host/src/ble_gap.c#L5533-L5536

I searched in the issues, but found nothing regarding this topic.

Looking forward for a response and am interested in helping to implement it, if wanted.

h2zero, 2021-09-06:
Great question! The answer is actually quite simple, I haven't thought of it and it has not come up until now 😄. It should be pretty easy to add, and I would happy to accept a PR for this if you're interested.

Mr-Mime, 2021-09-06:
That's a good answer :D
I will have a look at it...

Mr-Mime, 2021-09-07:
@h2zero I would be happy if you could say a word to my plans.

1.  update to the newest https://github.com/espressif/esp-nimble/commit/94afe2711752ba08bef93780be1b9e5143a87159
  However I am not sure about the changes from this commit https://github.com/espressif/esp-nimble/commit/4e51509cf37d45c4308750767b72752534cd675c  
  The [`modlog.h`](https://github.com/h2zero/NimBLE-Arduino/blob/master/src/modlog/modlog.h) of this repo looks like it has not been updated in a while, but rather had some individual changes, thus I am not sure what to do here exactly.
2. add the method to the classes
  I am not that experienced with the BLE specification, but IMO the Server and Client class should have the `setDataLength()` method

h2zero, 2021-09-07:
Don't worry about 1, I have a patch almost ready to go that I hope to push tonight. I have had to make some changes in the core code for better Arduino support, also the extra files in idf need to be updated as well, it's a bit of a chore that I'm going to make easier.

For 2, I agree with your proposal, those are the only classes that would use this functionality.

Mr-Mime, 2021-09-08:
Then I'll wait for your push, before I continue.

h2zero, 2021-09-08:
Sorry I didn't get it pushed last night. The changes I've made are proving to be more complex, hopefully get it sorted tonight.

Mr-Mime, 2021-09-08:
No worries and no need to rush :)

h2zero, 2021-09-09:
Changes pushed, you should be good to go now.

closed 2021-09-13 by h2zero