h2zero/NimBLE-Arduino #284: NimBLE - Wifi gateway coexistence example

smartanimaltraining, 2021-09-10:
It would be fantastic to add a working example in this repo showing how to use NimBLE and Wifi in an application running both simultaneously.

h2zero, 2021-09-10:
There isn't much to show. The esp32 controller does all the magic where we can't see it. Is there anything specific that you would like to see? 

I suppose I could do a demo using their functions in separate tasks. For the most part though you can just use WiFi and BLE however you see fit.

smartanimaltraining, 2021-09-10:
I would like to see a transfer of a message over to  NIMBLE CLIENT for a server and back. Also using using a coexistence call would pull it all together.

Thanks for the great library you have put so much effort into!
Thanks
Wes

svdrummer, 2021-09-10:
My understanding is, WiFi share the same antenna, which means scheduled switching between the two.

h2zero, 2021-09-12:
>I would like to see a transfer of a message over to NIMBLE CLIENT for a server and back. Also using using a coexistence call would pull it all together.

There aren't any "coexistence calls", you could have a simple webserver/client getting data and just send that data over BLE or from BLE back to the webserver or MQTT or telnet etc.. 

Here is a modified version of the scan continuous example that uses telnet (and uart) for the serial output and accepts commands to start/stop scanning. It uses this telnet library: https://github.com/yasheena/telnetspy

[wifi-ble-coex.txt](https://github.com/h2zero/NimBLE-Arduino/files/7149687/wifi-ble-coex.txt)

smartanimaltraining, 2021-09-12:
Thanks so much for link and the reply.  Sorry maybe "call" was not correct  but rather a config for coexistance preference(WiFi vs BLE priority) I guess what I was thinking about coexistence was  [FAQ](https://docs.espressif.com/projects/espressif-esp-faq/en/latest/software-framework/coexistence.html)   
"Does ESP32 support coexistence between Bluetooth® and Wi-Fi?
Yes, but time-sharing control is required for ESP32’s coexistence between Wi-Fi and Bluetooth. Please go to menuconfig to enable the Wi-Fi/Bluetooth coexistence, shown as follows:

menuconfig -> Component config -> Wi-Fi -> Software controls WiFi/Bluetooth coexistence (Enable)"

and...
[esp32 q/a](https://www.esp32.com/viewtopic.php?f=13&t=6707&start=30)
1. Hardware control:
If you disable CONFIG_SW_COEXIST_ENABLE, it will use hardware control. In this mode, Bluetooth has higher priority then WiFi. It will keep the bluetooth performance, but WiFi may be effected depend on bluetooth traffic. It means while WIFI is RX a packet, if bluetooth want to TX or RX a packet, it will stop WiFi. For example, BLE is scanning and WiFi is just in connection to hear beacon.

h2zero, 2021-09-12:
The coexistence config is all done in the arduino-esp32 core. There are no settings here that change it.

smartanimaltraining, 2021-09-13:
Thanks so much for feedback and taking time to provide the example!

closed 2021-09-13 by smartanimaltraining