h2zero/NimBLE-Arduino #285: Abort client connection

itamardv, 2021-09-14:
Hi All,

NimBLE is a great lib. Very good work of the developers. Congratulations !

I'd like to know if there is some way to abort a client connection at  onConnect callback of NimBle Server.

Thanks!

h2zero, 2021-09-14:
Hi and thank you.

Yes, you can use this:
```
void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
    if (your check value status etc.. here) {
        pServer->disconnect(desc->conn_handle);
    }
}
```

itamardv, 2021-09-15:
Great h2zero !

This will disconnect only the present client trying to connect right?
Previously connected clients will be preserved ?

Thnaks

h2zero, 2021-09-15:
Yes, it will disconnect only the currently connecting client.

h2zero, 2022-04-04:
Closing as resolved.

closed 2022-04-04 by h2zero