h2zero/NimBLE-Arduino #287: setNotifications(true) on missing 2902 descriptor

sivar2311, 2021-09-16:
Hello h2zero!
I worked on the [ESP32-BLEKeyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) library  some time ago and modified it to work with both the classic BLE stack and NimBLE-Arduino stack (to reduce the resource load).

In doing so I removed some 2902 based code as NimBLE-Arduino no longer provides this and handles it internally.

This has caused the behavior that when restarting the ESP, the BLE connection is restored automatically, but the keyboard no longer works - see issue T-vK/ESP32-BLE-Keyboard#115

I was able to fix this for the non-NimBLE version by reimplementing parts of the old code:

```C++
void BleKeyboard::onConnect(BLEServer* pServer) {
  this->connected = true;

#if !defined(USE_NIMBLE)

  BLE2902* desc = (BLE2902*)this->inputKeyboard->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(true);
  desc = (BLE2902*)this->inputMediaKeys->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(true);  

#else

// how to fix this in NimBLE?

#endif // USE_NIMBLE  

}

void BleKeyboard::onDisconnect(BLEServer* pServer) {
  this->connected = false;

#if !defined(USE_NIMBLE)
  BLE2902* desc = (BLE2902*)this->inputKeyboard->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(false);
  desc = (BLE2902*)this->inputMediaKeys->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
  desc->setNotifications(false);
  advertising->start();

#else

// how to fix this in NimBLE?

#endif // !USE_NIMBLE
}
```

My question is how can I fix this behavior in the NimBLE-Arduino version since the 2902 descriptor is not available in the NimBLE-Arduino API.

h2zero, 2021-09-16:
The easiest way is to enable bonding and bond with the client on connection. If done, this process will be automatic.

sivar2311, 2021-09-16:
My knowledge of BLE is very limited. 
These lines are from `BleKeyboard::begin()`
```C++
  ...
  BLESecurity* pSecurity = new BLESecurity();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);
  ...
```

With this the bonding should be enabled already?

h2zero, 2021-09-16:
It should be enabled with that, however you need to specify that you require security from the client.

Have a look at this example https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Secure_Server/NimBLE_Secure_Server.ino

wakwak-koba, 2021-09-16:
What I previously forked for NimBLE, would you like to help?
https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard

sivar2311, 2021-09-17:
Thanks for the tips and sorry for the delay in replying.

` BLEDevice::setSecurityAuth(true, true, true);` solved the problem in NimBLE-Mode.

May I ask what the difficult way would be?

sivar2311, 2021-09-18:
The problem itself is solved (thank you very much).
If you still like and have the time - the complicated way would still interest me.

closed 2021-09-18 by sivar2311

h2zero, 2021-09-18:
>If you still like and have the time - the complicated way would still interest me.

Sorry, been busy. The more complicated way is actually just to have the client re-subscribed every time it connects. There is one alternative and that would be to use the low level NimBLE calls that will broadcast a notification regardless if a client is subscribed.

sivar2311, 2021-09-19:
Thanks for the explanation.
Unfortunately this is beyond my BLE programming skills.

The easy way has helped :)

JopWerff, 2023-12-28:
Hi h2zero,

I have a bit a problem get things done with 2902 based stuff. I tried self several things - researched a little - had a look to this keyboard stuff of sivar2311. But I didn't get it. I'm a real newbie in the area of BLE.
I searched also for examples in the NewBLE-library but nothing to find.

Can you help me?

Here my code in to convert to NimBLED from the original BLE Library:
```
    NimBLECharacteristic* pCharacteristicTEMP = pBmeService->createCharacteristic(TEMP_UUID, NIMBLE_PROPERTY::NOTIFY);

    BLEDescriptor* pBLEDescriptorTEMP = pCharacteristicTEMP->createDescriptor(BLEUUID((uint16_t)0x2902));
    pBLEDescriptorTEMP->setValue("BME temperature Celsius");
    pCharacteristicTEMP->addDescriptor(pBLEDescriptorTEMP);
```

These last 3 lines: how I have to implement this in the NimBLE-way?

Jop

h2zero, 2023-12-29:
In this library the 2902 is automatic when the characteristic has the notification property. You need to delete those 3 lines.

Spiderman124, 2024-02-02:
Sorry guys, but I can't get it to work either despite following the steps above.
I still have the problem that the keyboard only works again after pairing again on Win10 computers. 
On a Pi4 with Bullseye64 the keyboard works after every reconnection. 
Maybe this is a stupid question... but couldn't the bugfix be implemented in an update to the library?
I would be very grateful for a helpful tip 

BR SM124