h2zero/NimBLE-Arduino #288: setEncryptionLevel alternative

tommy-b-10, 2021-09-22:
Hi, am I attempting to switch over from stock BLEDevice.h to NimBLE and are having an issue with converting the code.
My device requires I have the setEncryptionLevel flag set as below to be able to pair correctly, however there doesn't seem to be an alternative in this library? 

BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);

Is there an alternative setting with this library?
Thanks!

h2zero, 2021-09-23:
Hi @tommy-b-10. Yes the setting is available, NimBLE uses a different format for this configuration. Please see the documentation on this [here](https://h2zero.github.io/esp-nimble-cpp/md_docs__migration_guide.html#autotoc_md59). 

Note that encryption is always enabled here, so the call you referenced from the bluedroid library is not necessary. 
If you need to force encryption on connection you can call `NimBLEDevice::startSecurity` from the onConnect event handler.

tommy-b-10, 2021-09-24:
Thanks for confirming that, I have discovered it's not connecting due to another issue that I'll post.

closed 2021-09-24 by tommy-b-10