h2zero/NimBLE-Arduino #289: [Question] Make name writeable

Nailik, 2021-09-22:
Hi,

i want to make the name in the Generic Access writeable.

I found `ble_svc_gap_device_name_write_access` in ble_svc_gap.c but i'm not sure from where i should get ble_gatt_access_ctxt or how i need to set it up correctly - or if there is a much simpler way to make the device name writeable.

Hope someone can help me!

h2zero, 2021-09-23:
Hi @Nailik, in the default configuration this characteristic is not writable. You can change this by adding:
`#define MYNEWT_VAL_BLE_SVC_GAP_DEVICE_NAME_WRITE_PERM 0` to the nimconfig.h file.

I have not tested this but I believe it should work.

Nailik, 2021-09-23:
Thanks, works like a charm. Yeah it's weird but i need the name to have the flag write because i pretend to be a specific device and the client just checks for all the characteristic settings.

Hope it's ok to ask another question .. 

How can i add the Device Peripheral Properties to the Generic Access Service? 
Because i can't find the Service and then use createCharacteristics .. when i try
 pServer->getServiceByUUID(NimBLEUUID(0x1800), 0) it's returning a nullptr unless i create a Generic Access Service myself

Nailik, 2021-09-23:
Uff too easy - solution is almost the same, just define those:

```
#define MYNEWT_VAL_BLE_SVC_GAP_PPCP_MIN_CONN_INTERVAL 
#define MYNEWT_VAL_BLE_SVC_GAP_PPCP_MAX_CONN_INTERVAL 
#define MYNEWT_VAL_BLE_SVC_GAP_PPCP_SLAVE_LATENCY 
#define MYNEWT_VAL_BLE_SVC_GAP_PPCP_SUPERVISION_TMO 
```

And it works

closed 2021-09-23 by Nailik