h2zero/NimBLE-Arduino #290: client example Words used

svdrummer, 2021-09-23:
Hi,
I am trying to look for a list of devices and find the battery voltage of the beacon.
I look at the example called Nimble_client and can see the words "DEAD", "BEEF" "TASTY"

Does anyone know what these even mean"

One device I am looking for has service UUID of abbaff00-e56a-484c-b832-8b17cf6cbfe8
service is !*)A
Characteristic of 2A29

no idea what the rest means.
It is a fitbit watch
TIA

chegewara, 2021-09-23:
> I look at the example called Nimble_client and can see the words "DEAD", "BEEF" "TASTY"

its just a funny usage hex values (0-9, A-F) to create UUID (TASTY is characteristic value, so it can use any chars).

svdrummer, 2021-09-23:
Dead beef feed i get that part 😀 to read a battery voltage, can anyone recommend the standard uuid or whatever for battery voltages

h2zero, 2021-09-23:
The battery service standard uuid is 0x180F.

The 0x180a you see above is device information service.

You can see the full list of assigned numbers here:

https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf

h2zero, 2022-04-04:
Closing this as a non-library issue. Please reopen as a discussion if necessary.

closed 2022-04-04 by h2zero