h2zero/NimBLE-Arduino #291: Cannot Find BLE Advertisement when scanning

govnocoder, 2021-09-23:
My ESP32-WROOM-32U cannot find a BLE advertisement these non-connectable devices (advertising mode ADV_NONCONN_IND) with name starting TPMS. My phone and a BLE scanner can find the non-connectable devices when scanning. A screenshot is below. The ESP32 does detect other BLE devices, but not this non-connectable devices. I use continuous scanning example.

govnocoder, 2021-09-23:
[image]

h2zero, 2021-09-23:
That's strange, could be a bug. Could you try scanning with `NimBLEScan::setActiveScan(false);` and let me know if it shows then?

govnocoder, 2021-09-23:
Already tried, but no result. I'll try one more time.

govnocoder, 2021-09-25:
Tried again today and was unsuccessful. I noticed one thing. The sensors appear after a while for a few seconds. Perhaps I need to increase the scan interval or, on the contrary, decrease it?

govnocoder, 2021-09-25:
[image]

h2zero, 2021-09-25:
If they do show sometimes then you may need to adjust your parameters, try setting the window and interval closer together.

govnocoder, 2021-10-15:
Solved. Just change controller from esp32-wroom-32u to esp32-wroom-32d and all work.

h2zero, 2021-10-16:
This is interesting, could you share the change you made? It may be useful for others.

govnocoder, 2021-10-19:
No changes, just new board

h2zero, 2021-10-19:
Oh, I misunderstood. I though you were able to change the BLE controller binary.