h2zero/NimBLE-Arduino #292: Connection failing with status=574 on ESP32-WROOM-32E

tommy-b-10, 2021-09-24:
Hi,

I am experiencing a strange issue where I am unable to connect to a device with my ESP32-WROOM-32E that I am able to connect to with a ESP-WROOM-32 board. Both have the exact same arduino sketch uploaded, but the below is the output log from when I attempt to connect using the 32E.

```
D NimBLEScan: "discovery complete; reason=0"
D NimBLEClient: ">> connect(fa:b4:7c:f8:ba:a1)"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connected event"
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
D NimBLEClient: "Got Client event "
E NimBLEClient: "Connection failed; status=574 "
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "Not connected to any peers"
D NimBLEClient: "<< disconnect()"
```

My BLE setup is:
```
  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(45);
  pBLEScan->setWindow(15);
  BLEDevice::setSecurityAuth(true, true, true);
  BLEDevice::setPower(ESP_PWR_LVL_P9);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5);
```

And my connection code is:
 ```
  pClient = NimBLEDevice::createClient();
  pClient->setClientCallbacks(&clientCB);
  pClient->setConnectionParams(32, 160, 0, 500);
  pClient->setConnectTimeout(5);
  pClient->connect(advDevice);
```
Is there a bug with the 32E that may be causing this? Or specific settings for this module?

Thanks!

h2zero, 2021-09-24:
That message is the controller saying there was a timing error in the connection process. Try removing the connection parameter code and the timeout. If you have success then, you'll have to re-evaluate the parameters.

tommy-b-10, 2021-09-25:
Thanks h2zero, I've tried that, and many other combination of parameters with no luck. Always able to connect using the ESP-WROOM-32 module but not with the ESP32-WROOM-32E.

Note that the device I'm connecting to (GoPro 7) does have a 'pairing' mode where both modules connect successfully, but once not in that mode only the 32 can connect. Possibly there is something with the 32E module that is not allowing the GoPro to whitelist it? (Tried many combinations of pairing the two modules with still only the 32 being successful outside of the pairing mode).

h2zero, 2021-09-25:
I wonder if the GoPro has some kind of whitelist that only allows connections from already paired devices? In that scenario the new device may be rejected from connecting and you may need to reset the bluetooth settings on to GoPro to allow a new connection, just a thought.

tommy-b-10, 2021-09-25:
That was my thought too, but I've tried just pairing the 32E, with not luck when reconnecting. The thing that's really stumping me is that the 32 is working perfectly well. I have reset the GoPro many times and still no luck with just the 32E....

My other thought is does the ESP have any hidden settings that can only be configured using IDF menuconfig or something that could be causing this?

h2zero, 2021-09-25:
32 and the 32E are equal as far as I'm aware regarding BLE/bluetooth. One other thing to try if you haven't yet is doing a full erase of the flash on the 32E as well as resetting the GoPro incase there is some bonding info that is interfering.

tommy-b-10, 2021-09-25:
I reset the GoPro and erased the flash on both 32E and 32 modules, connected the 32E and successfully paired when the GoPro was in pairing mode, then rebooted the 32E and it would not connect again until the GoPro was put back in pair mode. Then tried the 32 and it successfully paired and reconnected when rebooted.

Exact same arduino code uploaded on both boards after erasing them...

h2zero, 2021-09-25:
Seems that the 32E is not correctly storing the bond/pairing info? Strange that there would be a difference in this regard though. I'll have to think about this more.

svdrummer, 2021-09-25:
Are you 100% sure its a 32e (not a Chinese knock off) with the faulty 32d
chip? Report the issue to espressif as well.

tommy-b-10, 2021-09-25:
Hi h2zero, I think I have solved the issue, but now have even more questions!

I set the 32E's MAC address to the same as the 32 module using esp_base_mac_addr_set(), and it seems to be connecting successfully now. It seems as though the default mac address of the 32E module is not being stored correctly in the GoPro?

32E default base mac = 44:17:93:61:0f:c4
32 default base mac = ec:94:cb:49:a9:8c

Is the 32E's mac unusual in any way? Not really a NimBLE issue, but it is very strange behaviour...

tommy-b-10, 2021-09-25:
> 
> Are you 100% sure its a 32e (not a Chinese knock off) with the faulty 32d chip? Report the issue to espressif as well.
> […](#)

Quite sure, it is a custom PCB and from a reputable manufacturer. ESP-IDF reports it as 'Chip is ESP32-D0WD-V3 (revision 3)'

h2zero, 2021-09-25:
Interesting... and good troubleshooting. Perhaps the GoPro isn't deleting the info completely? Perhaps it's expecting a different address type also, don't have time to decode the type right now but there is a clue there for sure.

tommy-b-10, 2021-09-25:
Now I am experiencing another issue... I am able to connect when the GoPro is in both pair and not pairing mode, however authentication only works in pair mode. When connecting normally, it the auth callback is not called and the connection is not bonded/authenticated.

The 32 only has this issue if the 32E is paired initially to the GoPro, otherwise it works fine

mxdv, 2022-01-25:
I found that esp32 needs setOwnAddrType to be set for the gopro to connect successfully. Otherwise gopro "forgets" esp32 after disconnection.

I don't know the exact settings yet. If you reset the gopro network settings and setOwnAddrType(BLE_OWN_ADDR_RANDOM); then the ESP32 at least can connect and power on the gopro. 

I discovered this a few hours ago, so I keep experimenting.

h2zero, 2022-04-04:
Closing this as a non-library issue. Please reopen as a discussion if necessary.

closed 2022-04-04 by h2zero