h2zero/NimBLE-Arduino #294: Non-blocking onConfirmPIN

olicooper, 2021-09-25:
In the original Bluedroid implementation `onConfirmPIN` was a blocking function. I bypassed the issue by creating some custom code which would call `esp_ble_confirm_reply(bleBondReqAddr, false)` when the user pressed a button which confirms the pin was correct. As the device was waiting for user input it could continue running other functions. I also had a timeout to automatically confirm `false` after 10 seconds (I think the ble stack enforces 30 seconds, but I set my own shorter timeout).
The problem with NimBLE is that `onConfirmPIN` is included in the `NimBLEServerCallbacks` so I can't bypass the call anymore.

Is is possible to create a way to asynchronously confirm the pin request? This would be much more useful than the current method. I have my own class that wraps BLE functionality i.e. `MyBleClass : public NimBLEServerCallbacks { ... }` so ideally the added functions would be located in `NimBLEServerCallbacks` for ease of use.

Thanks 👍

h2zero, 2021-09-25:
Sounds like a good suggestion. It could be implemented by adding a function to the server that can inject the response at a later time instead of waiting for the callback to return. I'm too busy at this time to work on it, I'd happily accept a PR for this if you're interested.

chegewara, 2021-09-26:
30 seconds is from bluetooth specification.

Yes, it is possible to do it. `onConfirm` callback can be overloaded that will just inform user there is request. Then like @h2zero said, additional function to which you can pass a pin.

olicooper, 2021-09-27:
Quite busy myself too so but I will see if I can make some time to work on something.. Any hints for me? I see usages of [ble_sm_inject_io](https://github.com/h2zero/NimBLE-Arduino/blob/f0bfd4c03ef6ab2fb01f04b231f0db7deab03c36/src/nimble/nimble/host/src/ble_sm.c#L2665) in [NimBLEServer](https://github.com/h2zero/NimBLE-Arduino/blob/f0bfd4c03ef6ab2fb01f04b231f0db7deab03c36/src/NimBLEServer.cpp#L524-L584) - is that what I should be calling?

h2zero, 2021-09-27:
`ble_sm_inject_io` is what you would need to use. 

I think the event handler would need to be changed so if the callback returns an invalid pin (such as 0xFFFFFFFF) it will not inject the pin response at that time and instead we could have a function to perform the proper pin response after.

h2zero, 2024-06-04:
If you are still following this there is a PR here: https://github.com/h2zero/esp-nimble-cpp/pull/165

h2zero, 2024-06-09:
resolved by https://github.com/h2zero/NimBLE-Arduino/commit/643ce468f4d64d55560fcb5461bd6834085ab182

PR #678 (2024-06-08): [BREAKING] Async pin injections for Numeric Comparison and Passkey input
fixes #294 

Adds new functions to asynchronously send pin and passkey confirmation to allow up to 30 seconds for display and key entry/confirmation.
 
* Make NimBLEConnInfo functions const.

* Update callback functions and update client to use new functions.

* Update examples.

* Update migration guide.
closed 2024-06-09 by h2zero