h2zero/NimBLE-Arduino #295: failed to allocate ACL buffers

nicolosponziello, 2021-09-27:
I am trying to transfer a 1.1mb file from a phone app to an esp32. The file is divided into packages smaller than the MTU.
After some packages have been succeessful sent, the esp is printing this error in the console:
NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT

How can I fix it?

h2zero, 2021-09-27:
Hi @nicolosponziello, you can try adjusting the buffers here https://github.com/h2zero/NimBLE-Arduino/blob/f0bfd4c03ef6ab2fb01f04b231f0db7deab03c36/src/nimconfig.h#L182

nicolosponziello, 2021-09-27:
Thank you for the reply. I just tried to increase them to 24 but the problem is still there

h2zero, 2021-09-27:
You might need to slow down the data rate a bit. What is your process for sending this data?

nicolosponziello, 2021-09-27:
In the app I have negotiated a MTU of 500 bytes and I split the file in packets that are MTU -12 bytes. Then I send each of them by writing to the characteristic with 100ms between each other

h2zero, 2021-09-27:
Just for a test in this case, could you try writing with response request, it'll be slow but should confirm if the data rate is the issue. 150ms might be too fast as there is no guarantee that the first packet was processed before the second gets sent.

nicolosponziello, 2021-09-30:
I tried with write with response and the app managed to send all data, but as soon as the app sent the last packet, the esp logged multiple lines about failing to allocate ACL buffers

nicolosponziello, 2021-09-30:
I solved the issue by reworking the code in the esp. Somehow the problem is not related to the data transfer rate, since now the problem is gone even with 60ms delay and the default number of ACL buffers

closed 2021-09-30 by nicolosponziello