h2zero/NimBLE-Arduino #296: Issue adding service after initialization

olicooper, 2021-09-27:
I have a program that can toggle a service on/off at runtime but the service doesn't appear and I get an error in the terminal: 
`E NimBLEService: "ble_gatts_add_svcs, rc= 15, "`

I have tried disconnecting attached devices (with a 100ms delay after requesting a disconnect) and stopping advertising before adding the service but the error still appears. The only reliable solution is to turn BLE off and back on again so that the service is added as BLE is starting up rather than at a later point in time.

I have a sample project here to demonstrate. See readme for more info..
https://github.com/olicooper/arduino-nimble-service-bug

h2zero, 2021-09-27:
When adding services after initialization, the server needs to reset the gatt/gap states. This will not occur until the clients have disconnected. This is a limitation in the NimBLE core at the moment.

olicooper, 2021-09-27:
Okay thanks for the info. I can see  `NimBLEServer::resetGATT()` but it is private, how do I do reset the state? 
My code already disconnects clients by calling `NimBLEServer::disconnect(bleConnHandle, BLE_ERR_REM_USER_CONN_TERM);` then waits 100ms before stopping advertisements and creating the new service. The problem is still present despite doing that.

There is still a problem even when no clients have previously connected (the service is empty), so I think that is a separate problem?

h2zero, 2021-09-27:
` NimBLEServer::resetGATT()` is called internally when the services are changed and after all connections are disconnected, no need to call it in your code.

>My code already disconnects clients by calling NimBLEServer::disconnect(bleConnHandle, BLE_ERR_REM_USER_CONN_TERM); then waits 100ms before stopping advertisements and creating the new service. 

Not sure why you're having issues there, i'll need to test it.

h2zero, 2022-03-18:
I seem to have forgotten about this one, I assume this is still an issue with the latest release, 1.3.7?

MassimilianoMontagni98, 2022-03-18:
Yes is still here

h2zero, 2022-03-19:
I have tested the code provided. What you are seeing as the error is due to calling `pService->start()`. This call is not needed here because it will be started automatically when the gatts server is restarted. Removing that from the code removes the error. After this I can see the service gets added after the client devices have all disconnected and it shows in nRFconnect as expected.

h2zero, 2022-04-04:
Closing this, assuming resolved from my above recommendations. Please reopen if this issue is still persisting.

closed 2022-04-04 by h2zero