h2zero/NimBLE-Arduino #297: bug in scan

chegewara, 2021-10-01:
Hi
today i discovered there is some bug, but its very hard to say if it is in NimBLE stack, espressif low level driver or this library.
My observations:
- sometimes, when scan with timeout the scan never stop
- it is fairly easy to fix when we use blocking scan
- i have no idea how to "fix" it when using non blocking scan

This is quick fix for blocking scan:
```
NimBLEScanResults NimBLEScan::start(uint32_t duration, bool is_continue) {
    if(duration == 0) {
        NIMBLE_LOGW(LOG_TAG, "Blocking scan called with duration = forever");
    }

    ble_task_data_t taskData = {nullptr, xTaskGetCurrentTaskHandle(),0, nullptr};
    m_pTaskData = &taskData;

    if(start(duration, nullptr, is_continue)) {
        if(duration){
            if(ulTaskNotifyTake(pdTRUE, duration * 1000 + 100) != pdTRUE) // dont panic, maybe just period is late
                  stop();
       } else {
            // if you call this function with duration = 0 you dont know what you are doing
       }
    }

    m_pTaskData = nullptr;
    return m_scanResults;
} // start
```

h2zero, 2021-10-01:
Are you using master or release branch? I noticed this recently as well, seems like a timing issue. Any logs?

chegewara, 2021-10-01:
Im not sure whch version i am using (PIO):
`h2zero/NimBLE-Arduino @ ^1.3.1`

Yes, bunch of lgs saying that device are found with info `updated` and no `new device` which means the scan is continuing. It suppose to be  short scan, like 5 sec. Few itterations is ok, but in meantime i am switching code between wifi and LTE, by disabling/enabling AP in smartphone.

h2zero, 2021-10-01:
I've got a test running with 1 second scan times then restart 100ms after stop. 

I have not seen this issue on the esp32 before but it has happened when testing other MCU's. 

I'll let this run at work today and see if I can reproduce.

h2zero, 2021-10-01:
Yes that code is a good backstop and should added, thanks! 

I'm just hoping I can stress test this a bit to find the bug. I added a few asserts to see if I can catch it.

h2zero, 2021-10-01:
Ran the stress test for 7 hours today and had no issues with the non-blocking scan. 

I'll add your suggested changes for blocking use, if you'd like to PR it feel free.

chegewara, 2021-10-02:
Thanks, and sorry, but this is not my week, so no PR.

h2zero, 2021-10-02:
Lol no worries, I have a few other bug fixes to address as well so I'll add this to it.

closed 2021-12-19 by chegewara