h2zero/NimBLE-Arduino #299: Low level connect for NimBLEClient

DTTerastar, 2021-10-05:
My project https://github.com/ESPresense/ESPresense is trying to connect to quite a few different devices quickly and get the model number and/or name.  It sort of works but since connect doesn't return an error code I can't do retries based on the error codes.  Is there a way to make a lower level connect that allows you to get the codes?  Connect is also blocking so you can't easily do simultaneous connections.  Any other suggestions on how to make the connections more reliable would be helpful: https://github.com/ESPresense/ESPresense/blob/ab7e387c22030f80f07330912d3e9105d05e0356/lib/BleFingerprint/BleFingerprint.cpp#L303

ATM even after 10 attempts i'm only like 80% successful.  (iPhones rotate the mac every 15 minutes so it needs to work every 15 minutes)

DTTerastar, 2021-10-05:
Oh and it'd be nice to be able to avoid the logging to the console for this call:

```lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "```

corgan2222, 2021-10-05:
It would be great to have a solution for this Problem. 

[image]

h2zero, 2021-10-05:
BLE connections will only occur when the server device is listening for them. This is usually only for short periods of time after advertising intervals. If you're cycling through connections to a large number of devices it's likely some of them have stopped listening by the time you attempt connection.

NimBLE supports multiple connections but can only be actively trying to connect to 1 device at a time, that's why the call blocks.

If you want to experiment with the low level NimBLE calls you can look in NimBLEClient.cpp and look for ble_gap_connect within NimBLE itself. I'd be curious of any ideas you come up with.

The ll_pdu messages come from the espressif controller and we have no way to disable them. Only solution is to update the arduino core / IDF version to a version after espressif removed that. The NimBLE error can just be commented out if you don't want to see it.

h2zero, 2021-10-05:
Ok, I see. A get last error function added to the client?

DTTerastar, 2021-10-05:
Yeah, that'll work

DTTerastar, 2021-10-05:
I still find having callbacks in a blocking function very weird, but I know you're trying to be compatible with an older api.

h2zero, 2021-10-05:
Yes, it's not really necessary for this but it doesn't hurt much. It wasn't an easy job to put this together in the same API but it worked out 😃.

I'll be busy tonight but I'll look to add the function when I have a few minutes.

h2zero, 2021-10-06:
I can see you've already done what you need but let me know if #300 works for you.

closed 2021-10-06 by DTTerastar