h2zero/NimBLE-Arduino #302: Panic in NimBLEScan::clearResults()

DTTerastar, 2021-10-13:
CORRUPT HEAP: Bad head at 0x3ffeb5b4. Expected 0xabba1234 got 0x3ffeb650
abort() was called at PC 0x4008778d on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40090220:0x3ffe1080 0x4009049d:0x3ffe10a0 0x4008778d:0x3ffe10c0 0x400878b9:0x3ffe10f0 0x40126f9f:0x3ffe1110 0x40120a45:0x3ffe13d0 0x401209c9:0x3ffe1420 0x400951f9:0x3ffe1450 0x4008291e:0x3ffe1470 0x40087685:0x3ffe1490 0x4000bec7:0x3ffe14b0 0x401ab309:0x3ffe14d0 0x400e049d:0x3ffe14f0 0x400e063e:0x3ffe1510 0x400f110f:0x3ffe1540 0x400f5b53:0x3ffe15b0 0x400d7451:0x3ffe1760 0x400914ae:0x3ffe17e0
  #0  0x40090220:0x3ffe1080 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4009049d:0x3ffe10a0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4008778d:0x3ffe10c0 in lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
  #3  0x400878b9:0x3ffe10f0 in _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
  #4  0x40126f9f:0x3ffe1110 in _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
  #5  0x40120a45:0x3ffe13d0 in fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
  #6  0x401209c9:0x3ffe1420 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
  #7  0x400951f9:0x3ffe1450 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305
  #8  0x4008291e:0x3ffe1470 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #9  0x40087685:0x3ffe1490 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #10 0x4000bec7:0x3ffe14b0 in ?? ??:0
  #11 0x401ab309:0x3ffe14d0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #12 0x400e049d:0x3ffe14f0 in __gnu_cxx::new_allocator<unsigned char>::deallocate(unsigned char*, unsigned int) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:408
      (inlined by) std::allocator_traits<std::allocator<unsigned char> >::deallocate(std::allocator<unsigned char>&, unsigned char*, unsigned int) at /Users/dturner/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/alloc_traits.h:386
      (inlined by) std::_Vector_base<unsigned char, std::allocator<unsigned char> >::_M_deallocate(unsigned char*, unsigned int) at /Users/dturner/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:178
      (inlined by) std::_Vector_base<unsigned char, std::allocator<unsigned char> >::~_Vector_base() at /Users/dturner/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:160
      (inlined by) std::vector<unsigned char, std::allocator<unsigned char> >::~vector() at /Users/dturner/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:425
      (inlined by) NimBLEAdvertisedDevice::~NimBLEAdvertisedDevice() at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEAdvertisedDevice.h:40
      (inlined by) NimBLEScan::clearResults() at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:465
  #13 0x400e063e:0x3ffe1510 in NimBLEScan::stop() at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:408
  #14 0x400f110f:0x3ffe1540 in NimBLEClient::connect(NimBLEAddress const&, bool) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEClient.cpp:454
  #15 0x400f5b53:0x3ffe15b0 in BleFingerprint::query() at .pio/libdeps/m5atom-matrix/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp:138
  #16 0x400d7451:0x3ffe1760 in scanForDevices(void*) at src/main.cpp:339
  #17 0x400914ae:0x3ffe17e0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

h2zero, 2021-10-13:
Looks like some sort of timing issue, please try calling scan::stop() before connecting and let me know if that resolves it.

h2zero, 2021-10-14:
That option is possibly the best. I was considering a mutex or semaphore but i'd prefer not to use those.

h2zero, 2021-10-16:
@DTTerastar I've submitted #304 for this, if you would like to test it in your use I would appreciate any feedback.

DTTerastar, 2021-10-16:
I've released a version with it: https://github.com/ESPresense/ESPresense/commit/90cb008329344de0a2252a66fb522c0e22521d56

I'll monitor the uptime of the various sensors.

h2zero, 2022-01-15:
I'm assuming this is resolved and closing. Please re-open if not.

closed 2022-01-15 by h2zero

thorrak, 2022-02-07:
Interestingly, I just started encountering this issue with a project of mine which I think relates to the fact that  I'm currently performing active scans (ie. I have `pBLEScan->setActiveScan(true);` set):

```
CORRUPT HEAP: Bad head at 0x3ffce8f8. Expected 0xabba1234 got 0x3ffce968
abort() was called at PC 0x40086c3d on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f778:0x3ffcbfb0 0x4008f9f5:0x3ffcbfd0 0x40086c3d:0x3ffcbff0 0x40086d69:0x3ffcc020 0x40119443:0x3ffcc040 0x40112d2d:0x3ffcc300 0x40112cb5:0x3ffcc350 0x400943d1:0x3ffcc380 0x40081dee:0x3ffcc3a0 0x40086b35:0x3ffcc3c0 0x4000bec7:0x3ffcc3e0 0x401825b5:0x3ffcc400 0x400e692d:0x3ffcc420 0x400e69a9:0x3ffcc440 0x400d8a31:0x3ffcc460 0x400d890f:0x3ffcc480 0x400d8917:0x3ffcc4a0 0x400f783c:0x3ffcc4c0 0x40090a06:0x3ffcc4e0
  #0  0x4008f778:0x3ffcbfb0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008f9f5:0x3ffcbfd0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40086c3d:0x3ffcbff0 in lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
  #3  0x40086d69:0x3ffcc020 in _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
  #4  0x40119443:0x3ffcc040 in _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
  #5  0x40112d2d:0x3ffcc300 in fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
  #6  0x40112cb5:0x3ffcc350 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
  #7  0x400943d1:0x3ffcc380 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305
  #8  0x40081dee:0x3ffcc3a0 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:354
  #9  0x40086b35:0x3ffcc3c0 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #10 0x4000bec7:0x3ffcc3e0 in ?? ??:0
  #11 0x401825b5:0x3ffcc400 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #12 0x400e692d:0x3ffcc420 in __gnu_cxx::new_allocator<unsigned char>::deallocate(unsigned char*, unsigned int) at .pio/libdeps/esp32_wifi_tft/NimBLE-Arduino/src/NimBLEScan.cpp:413
      (inlined by) std::allocator_traits<std::allocator<unsigned char> >::deallocate(std::allocator<unsigned char>&, unsigned char*, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/alloc_traits.h:386
      (inlined by) std::_Vector_base<unsigned char, std::allocator<unsigned char> >::_M_deallocate(unsigned char*, unsigned int) at /Users/jbeeler/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:178
      (inlined by) std::_Vector_base<unsigned char, std::allocator<unsigned char> >::~_Vector_base() at /Users/jbeeler/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:160
      (inlined by) std::vector<unsigned char, std::allocator<unsigned char> >::~vector() at /Users/jbeeler/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:425
      (inlined by) NimBLEAdvertisedDevice::~NimBLEAdvertisedDevice() at .pio/libdeps/esp32_wifi_tft/NimBLE-Arduino/src/NimBLEAdvertisedDevice.h:40
      (inlined by) NimBLEScan::clearResults() at .pio/libdeps/esp32_wifi_tft/NimBLE-Arduino/src/NimBLEScan.cpp:470
  #13 0x400e69a9:0x3ffcc440 in NimBLEScan::start(unsigned int, void (*)(NimBLEScanResults), bool) at .pio/libdeps/esp32_wifi_tft/NimBLE-Arduino/src/NimBLEScan.cpp:413
  #14 0x400d8a31:0x3ffcc460 in btScanner::scan() at src/wireless/BTScanner.cpp:110
  #15 0x400d890f:0x3ffcc480 in brewpiLoop() at src/brewpi-esp8266.cpp:188
  #16 0x400d8917:0x3ffcc4a0 in loop() at src/brewpi-esp8266.cpp:202
  #17 0x400f783c:0x3ffcc4c0 in loopTask(void*) at /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
  #18 0x40090a06:0x3ffcc4e0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

The function I'm calling in my loop which is triggering the scan/crash is as follows:

```
bool btScanner::scan()
{
    if (NimBLEDevice::getScan()->isScanning())
        return false;
    if (NimBLEDevice::getScan()->start(60, nullptr, false))
        return true; //Scan successfully started.
    return false;  //Scan failed to start
}
```

My hypothesis is that having active scanning set can result in the edge case you mentioned. My guess was that the active scan is connecting to devices at the exact same time as I'm trying to clear results, and my solution was to add  a `delay(500);` right before the `getScan()->start()`. 

After adding this delay I appear to be good - going on 8 hours without a crash (where I was previously seeing one happen every few minutes!)