h2zero/NimBLE-Arduino #305: CRIT NimBLEDevice: "Resetting state; reason=19, " failed to configure restored IRK

societyofrobots, 2021-10-22:
Using Android IDE, custom board w/EPS32, 1.0.6 core. Connecting via an iPhone app.

The first time Bluetooth connects to iPhone, it works without issue:
```
D NimBLECharacteristic: "<< notify"
D NimBLECharacteristic: ">> setValue: length=1, data=63, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e"
D NimBLECharacteristic: "<< setValue"
D NimBLECharacteristic: ">> notify: length: 1"
D NimBLECharacteristicCallbacks: "onNotify: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLECharacteristicCallbacks: "onStatus: default"
D NimBLECharacteristic: "<< notify"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=12, subscribed: false"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 0"
D NimBLECharacteristicCallbacks: "onSubscribe: default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLEServer: ">> handleGapEvent: "
```

I then disconnect BLE from iPhone app:

```
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
W NimBLEAdvertising: "Advertising already active"
D NimBLEServer: ">> handleGapEvent: "
```

I then try to reconnect to BLE, spinning icon, never connects, and I get this:

```
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
E NimBLEAdvertising: "Host reset, wait for sync."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Advertising Complete"
failed to configure restored IRK
failed to configure restored IRK
failed to configure restored IRK
I NimBLEDevice: "NimBle host synced."
D NimBLEAdvertising: "Host re-synced"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
```

Any idea on what is going on here?

societyofrobots, 2021-10-30:
Yes, bonding and encryption enabled. Erasing the flash temporarily fixes the issue when it happens.

I realized the only time the problem happens is when more than one iPhone is involved. Never happens with just one iPhone. But, when I connect a second simultaneously, they'll work together for a short period and then entirely crap out. Have to delete saved connected devices on the iPhones and power cycle the PCB to get it working again.

I upgraded to 2.0.1-rc1 and still the same problem:
```
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: ">> handleGapEvent: "
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "

D NimBLEServerCallbacks: "onDisconnect(): Default"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
E NimBLEAdvertising: "Host reset, wait for sync."
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=12, subscribed: false"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 0"
D NimBLECharacteristicCallbacks: "onSubscribe: default"
D NimBLEServer: ">> handleGapEvent: "
CRIT NimBLEServer: "Disconnect - host reset, rc=19"
```

My NimBLE code here (which is called from main):
```
#include <NimBLEDevice.h>
#include <NimBLEHIDDevice.h>//for HID showing in iOS

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */  
class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
      Serial.println("\nBLE device connected.");        
      save_log(BLE_con);
      BLEDevice::startAdvertising();//adding this line allows for multiple simultaneous BLE connections
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
      Serial.println("\nBLE device disconnected.");
      save_log(BLE_dis);
    }
  /***************** New - Security handled here ********************
  ****** Note: these are the same return values as defaults ********/
    uint32_t onPassKeyRequest(){
      Serial.println("Server PassKeyRequest");
      return passkey; 
    }
    
    void onPassKeyNotify(uint32_t passkey) {
      Serial.println("Passkey Notify");
    }

    bool onConfirmPIN(uint32_t passkey){
      Serial.println("The passkey YES/NO number: ");
[... 94 lines cut ...]
  pAdvertising->addServiceUUID((uint16_t)0x1812);//this one displays in iOS
  pAdvertising->setManufacturerData("Tech");
  pAdvertising->setScanResponse(true); // must be true or else BLE name gets truncated
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter

  BLEDevice::startAdvertising();
  Serial.println("Waiting for a client connection to notify...");
}

void BLE_loop() {
    
    // disconnecting
    if (!deviceConnected && oldDeviceConnected) {
        save_log(BLE_dis);
        delay(300); // give the bluetooth stack the chance to get things ready
        oldDeviceConnected = deviceConnected;
    }
    // connecting
    if (deviceConnected && !oldDeviceConnected) {
        // do stuff here on connecting
        save_log(BLE_con);
        oldDeviceConnected = deviceConnected;
    }
}
```

h2zero, 2021-10-30:
If you erase the device and disable bonding (comment out the security settings) does the same problem occur? I'd like to narrow the scope of the issue.

societyofrobots, 2021-10-30:
Quick question... how do I delete just the saved bonds in flash?

> disable bonding (comment out the security settings)

I changed `NimBLEDevice::setSecurityAuth(true, true, true);`
to `NimBLEDevice::setSecurityAuth(false, false, false);`

Is that what you meant?

I tried it using the following process:
1) connected via iPhone settings
2) ... connected using serial app on iPhone, it asked for PIN, works fine
3) connected via Android settings
4) ... connected using serial app on Android, it asked for PIN, works fine
5) I then disconnected via Android app
6) re-connected via Android app, but it no longer works, it asks for PIN but still won't work
7) deleted BLE device in Android settings
8) went back to step 3 and it works fine again until step 6

The iPhone, the first device connected, can be disconnected and reconnected without issue.

Below are the messages received during above process:

```
D NimBLEServer: ">> handleGapEvent: "

BLE device connected.

D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=0 mtu=255"
D NimBLEServerCallbacks: "onMTUChange(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=8, subscribed: false"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=12, subscribed: true"
I NimBLECharacteristic: "New subscribe value for conn: 0 val: 1"
D NimBLECharacteristicCallbacks: "onSubscribe: default"
D NimBLEServer: ">> handleGapEvent: "
Server PassKeyRequest
D NimBLEServer: "BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0"
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
D NimBLECharacteristic: "Characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e Write event"
D NimBLECharacteristic: ">> setValue: length=1, data=53, characteristic UUID=6e400002-b5a3-f393-e0a9-e50e24dcca9e"
D NimBLECharacteristic: "<< setValue"

Received Value: S, Length: 1D NimBLECharacteristicCallbacks: "onWrite: default"
D NimBLECharacteristic: ">> setValue: length=1, data=30, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e"
D NimBLECharacteristic: "<< setValue"
D NimBLECharacteristic: ">> notify: length: 1"
D NimBLECharacteristicCallbacks: "onNotify: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLECharacteristicCallbacks: "onStatus: default"
D NimBLECharacteristic: "<< notify"
D NimBLECharacteristic: ">> setValue: length=1, data=30, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e"
D NimBLECharacteristic: "<< setValue"
D NimBLECharacteristic: ">> notify: length: 1"
D NimBLECharacteristicCallbacks: "onNotify: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLECharacteristicCallbacks: "onStatus: default"
D NimBLECharacteristic: "<< notify"
D NimBLECharacteristic: ">> setValue: length=1, data=31, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e"
D NimBLECharacteristic: "<< setValue"
D NimBLECharacteristic: ">> notify: length: 1"
D NimBLECharacteristicCallbacks: "onNotify: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLECharacteristicCallbacks: "onStatus: default"
[... 59 lines cut ...]

BLE device connected.
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
D NimBLEServerCallbacks: "onConnect(): Default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "<< handleGATTServerEvent"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "mtu update event; conn_handle=1 mtu=255"
D NimBLEServerCallbacks: "onMTUChange(): Default"
D NimBLEServer: ">> handleGapEvent: "
I NimBLEServer: "subscribe event; attr_handle=12, subscribed: true"
I NimBLECharacteristic: "New subscribe value for conn: 1 val: 1"
D NimBLECharacteristicCallbacks: "onSubscribe: default"
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Connection parameters updated."
D NimBLEServer: ">> handleGapEvent: "
Server PassKeyRequest
D NimBLEServer: "BLE_SM_IOACT_DISP; ble_sm_inject_io result: 0"
D NimBLEServer: "<< handleGATTServerEvent"
```

societyofrobots, 2021-11-16:
I changed `CONFIG_BT_NIMBLE_MAX_CONNECTIONS 5` to max of `CONFIG_BT_NIMBLE_MAX_CONNECTIONS 9` and the issue seems to have gone away. I am only using two phones so no idea why.

When initiating BLE connections with two phones at similar times, sometimes it gets corrupted and I have to delete the pairing manually on a phone. This message pops up and it resets my ESP32 during the corruption:

```
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
Encrypt connection failed - disconnecting

assert failed: static NimBLEClient* NimBLEDevice::getClientByID(uint16_t) C:\Users\X\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp:237 (0)

Backtrace:0x40083861:0x3ffdd4300x40093081:0x3ffdd450 0x40098921:0x3ffdd470 0x400f1809:0x3ffdd5a0 0x400d37e5:0x3ffdd5c0 0x400f26e6:0x3ffdd5e0 0x400f6751:0x3ffdd650 0x400f679b:0x3ffdd670 0x400f7924:0x3ffdd690 0x400fde07:0x3ffdd6f0 0x400fe98b:0x3ffdd780 0x400f732c:0x3ffdd7c0 0x400f737c:0x3ffdd820 0x400fb466:0x3ffdd880 0x400fb691:0x3ffdd8a0 0x400f9fdd:0x3ffdd8c0 0x401012b2:0x3ffdd8e0 0x400f16a3:0x3ffdd900 

ELF file SHA256: 0000000000000000

Rebooting...
```

h2zero, 2021-11-16:
If you only have 2 connections I find it strange that you would need to set the max higher than 3 (advertising counts as one for some reason). 

Any chance you can decode that backtrace?

societyofrobots, 2021-11-17:
> Any chance you can decode that backtrace?

I've tried, maybe I'm doing it wrong. I used the 'Exception Decoder' for the Arduino IDE, which normally works but is never useful for me... It just says, "Decode Failed".

I was browsing the code and found this:
```
/** @brief Un-comment to change the number of devices allowed to store/bond with */
// #define CONFIG_BT_NIMBLE_MAX_BONDS 3
```

How is this different from max connections?

h2zero, 2021-11-17:
That config value sets the maximum number of bonds that can be stored in flash (i.e paring keys) and has no relation to simultaneous connections.

Also I must apologize as I did not answer your previous question:
>Quick question... how do I delete just the saved bonds in flash?

 You can call `NimBLEDevice::deleteAllBonds();` to do this.

h2zero, 2021-11-17:
If you can find the `elf` file and post it I can attempt to decode the backtrace for you.

societyofrobots, 2021-11-17:
elf file:
https://drive.google.com/file/d/1pBlOAW_KmDUiTjt0idJ7ixMMqIVhqPTE/view?usp=sharing

> That config value sets the maximum number of bonds that can be stored in flash (i.e paring keys) and has no relation to simultaneous connections.

If I have say 10 phones that will be used to connect, should I uncomment and set it to 10? Is there a max limit?

h2zero, 2021-11-17:
Yes that would be what you need to set it to. There is no hard maximum limit that I'm aware of except using all the flash space 😄

societyofrobots, 2021-11-17:
>There is no hard maximum limit that I'm aware of except using all the flash space

Would you know off-hand how much each bond would use?

h2zero, 2021-11-17:
Not much, maybe 400 bytes?

h2zero, 2021-11-21:
@societyofrobots Sorry for the delay with this, I have decoded the backtrace and this is the result:
```
0x40083861: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 402
0x40093081: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 129
0x40098921: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x400f1809: NimBLEDevice::getClientByID(unsigned short) at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp line 237
0x400d37e5: MyServerCallbacks::onAuthenticationComplete(ble_gap_conn_desc*) at C:\Users\pikachu\AppData\Local\Temp\arduino_build_603438\sketch/NimBLE.h line 62
0x400f26e6: NimBLEServer::handleGapEvent(ble_gap_event*, void*) at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEServer.cpp line 515
0x400f6751: ble_gap_call_event_cb at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 696
0x400f679b: ble_gap_call_conn_event_cb at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 722
0x400f7924: ble_gap_enc_event at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 5841
0x400fde07: ble_sm_process_result at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_sm.c line 968
0x400fe98b: ble_sm_connection_broken at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_sm.c line 2782
0x400f732c: ble_gap_conn_broken at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1222
0x400f737c: ble_gap_rx_disconn_complete at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_gap.c line 1260
0x400fb466: ble_hs_hci_evt_disconn_complete at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 196
0x400fb691: ble_hs_hci_evt_process at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs_hci_evt.c line 859
0x400f9fdd: ble_hs_event_rx_hci_ev at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\nimble\host\src\ble_hs.c line 520
0x401012b2: nimble_port_run at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src/nimble/nimble_npl_os.h line 121
0x400f16a3: NimBLEDevice::host_task(void*) at C:\Users\pikachu\Documents\Arduino\libraries\NimBLE-Arduino\src\NimBLEDevice.cpp line 728
```

It looks like the crash occurs because the client is disconnected and therefore the handle is invalid, I will remove the assert there as it was relic of debugging that got forgotten.

As for the cause of the general problem, have you had any further success?

societyofrobots, 2021-12-01:
I found the fix to it.

In nimconfig.h, changed to these settings:
```
 #define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 7
 #define CONFIG_BT_NIMBLE_MAX_BONDS 7
```

Even though I was allowing 9 connections, it refused to connect because it wouldn't save additional bonds. So also had to increase max bonds.

As it's impossible to have more possible simultaneous encrypted connections than possible saved bonds, perhaps the library should use CONFIG_BT_NIMBLE_MAX_CONNECTIONS as MIN_BONDS?

closed 2021-12-01 by societyofrobots

h2zero, 2021-12-02:
Great, thanks! Good suggestion, I'll tweak the macros to handle this.