h2zero/NimBLE-Arduino #308: Scan eventually consumes all cycles

ab0oo, 2021-10-29:
I'm struggling to put into words what I'm seeing, but here's my best attempt.
I'm writing a simply BLE scanner that uses callbacks, based on the example code.  I've made it as simple as possible, but it's not acting the way I'd _expect_.

The code:
```
#include "NimBLEDevice.h"

NimBLEScan* pBLEScan;
long lastUpdate;
const int scanTime = 3;

class MyAdvertisedDeviceCallbacks: public NimBLEAdvertisedDeviceCallbacks {
    void onResult(NimBLEAdvertisedDevice* advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(97);
  pBLEScan->setWindow(37); // less or equal setInterval value
  pBLEScan->setMaxResults(0);
}

void loop() {
  long now = millis();
  if ( now >= lastUpdate + 1000 ) {
    Serial.printf("Tick: %ld\n", now / 1000 );
    lastUpdate = now;
  }
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  pBLEScan->clearResults(); // delete results fromBLEScan buffer to release memory
}
```

What I want to see is "Advertised Device: XXX" information printed periodically, as devices are discovered.  I'd also like to see the word "Tick" posted every second, indicating that the scan is non-blocking and happening "in the background" while my main loop() continues to work.
In this case, it seems that the pBLEScan->start() is blocking, and since my scanTime is 3 seconds, I see "Tick 3" ... "Tick 6" ... "Tick 9"...  and then... nothing.  The output looks like this:

> 15:26:53.409 -> ets Jun  8 2016 00:22:57
> 15:26:53.409 -> 
> 15:26:53.409 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 15:26:53.409 -> configsip: 0, SPIWP:0xee
> 15:26:53.409 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 15:26:53.409 -> mode:DIO, clock div:1
> 15:26:53.409 -> load:0x3fff0030,len:1420
> 15:26:53.409 -> ho 0 tail 12 room 4
> 15:26:53.409 -> load:0x40078000,len:13540
> 15:26:53.409 -> load:0x40080400,len:3604
> 15:26:53.409 -> entry 0x400805f0
> 15:26:54.044 -> Advertised Device: Name: 846B215F5FC233AEE9, Address: f3:33:c0:55:58:95, appearance: 0, manufacturer data: a705051001000000000000001b00ca, serviceUUID: 0xfe07, txPower: 0 
> 15:26:54.508 -> Advertised Device: Name: , Address: 75:6d:2b:0e:f5:29, manufacturer data: 4c001006191e23cfe2c1, txPower: 8 
> 15:26:55.115 -> Advertised Device: Name: , Address: 76:90:c3:e9:90:02, manufacturer data: e00001b9ca72c54c, serviceUUID: 0xfe9f
> 15:26:55.115 -> Service Data:
> 15:26:55.115 -> UUID: 0xfe9f, Data: nTF065F6mZs 
> 15:26:55.115 -> Advertised Device: Name: , Address: 43:16:54:3d:42:ed, manufacturer data: 4c001005491c18dbfe 
> 15:26:55.221 -> Advertised Device: Name: , Address: d7:96:ba:2b:33:e2, manufacturer data: 4c0012020000 
> 15:26:56.187 -> Advertised Device: Name: , Address: 5d:0c:66:6f:61:2e, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:26:56.664 -> Advertised Device: Name: , Address: 5a:35:34:88:dc:8a, serviceUUID: 0xfd6f
> 15:26:56.664 -> Service Data:
> 15:26:56.664 -> UUID: 0xfd6f, Data: ⸮<⸮ 
> 15:26:56.912 -> Advertised Device: Name: , Address: 4b:21:d4:48:87:b6, serviceUUID: 0xfe9f
> 15:26:56.912 -> Service Data:
> 15:26:56.912 -> UUID: 0xfe9f, Data:  
**> 15:26:56.912 -> Tick: 3**
> 15:26:56.950 -> Advertised Device: Name: , Address: 43:16:54:3d:42:ed, manufacturer data: 4c001005491c18dbfe 
> 15:26:56.950 -> Advertised Device: Name: , Address: 5a:35:34:88:dc:8a, serviceUUID: 0xfd6f
> 15:26:56.950 -> Service Data:
> 15:26:56.950 -> UUID: 0xfd6f, Data: ⸮<⸮ 
> 15:26:58.214 -> Advertised Device: Name: , Address: 76:90:c3:e9:90:02, manufacturer data: e00001b9ca72c54c, serviceUUID: 0xfe9f
> 15:26:58.214 -> Service Data:
> 15:26:58.214 -> UUID: 0xfe9f, Data: nTF065F6mZs 
> 15:26:59.290 -> Advertised Device: Name: , Address: 5d:0c:66:6f:61:2e, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:26:59.928 -> Advertised Device: Name: , Address: 75:6d:2b:0e:f5:29, manufacturer data: 4c001006191e23cfe2c1, txPower: 8 
> 15:26:59.928 -> Advertised Device: Name: , Address: f3:33:c0:55:58:95, appearance: 0, manufacturer data: a705051001000000000000001b00ca, serviceUUID: 0xfe07, txPower: 0 
**> 15:26:59.964 -> Tick: 6**
> 15:27:00.073 -> Advertised Device: Name: , Address: 43:16:54:3d:42:ed, manufacturer data: 4c001005491c18dbfe 
> 15:27:00.682 -> Advertised Device: Name: , Address: 5a:35:34:88:dc:8a, serviceUUID: 0xfd6f
> 15:27:00.682 -> Service Data:
> 15:27:00.682 -> UUID: 0xfd6f, Data: ⸮<⸮ 
> 15:27:01.043 -> Advertised Device: Name: , Address: 76:90:c3:e9:90:02, manufacturer data: e00001b9ca72c54c, serviceUUID: 0xfe9f
> 15:27:01.077 -> Service Data:
> 15:27:01.077 -> UUID: 0xfe9f, Data: nTF065F6mZs 
> 15:27:01.328 -> Advertised Device: Name: , Address: 5d:0c:66:6f:61:2e, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:27:02.952 -> Advertised Device: Name: , Address: f3:33:c0:55:58:95, appearance: 0, manufacturer data: a705051001000000000000001b00ca, serviceUUID: 0xfe07, txPower: 0 
> 15:27:02.952 -> Advertised Device: Name: , Address: 75:6d:2b:0e:f5:29, manufacturer data: 4c001006191e23cfe2c1, txPower: 8 
**> 15:27:02.988 -> Tick: 9**
> 15:27:02.988 -> Advertised Device: Name: , Address: 43:16:54:3d:42:ed, manufacturer data: 4c001005491c18dbfe 
> 15:27:03.388 -> Advertised Device: Name: , Address: 5d:0c:66:6f:61:2e, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:27:04.965 -> Advertised Device: Name: , Address: 5a:35:34:88:dc:8a, serviceUUID: 0xfd6f
> 15:27:04.965 -> Service Data:
> 15:27:04.965 -> UUID: 0xfd6f, Data: ⸮<⸮ 
> 15:27:05.430 -> Advertised Device: Name: , Address: 76:90:c3:e9:90:02, manufacturer data: e00001b9ca72c54c, serviceUUID: 0xfe9f
> 15:27:05.430 -> Service Data:
> 15:27:05.430 -> UUID: 0xfe9f, Data: nTF065F6mZs 
> 15:27:05.430 -> Advertised Device: Name: 846B215F5FC233AEE9, Address: f3:33:c0:55:58:95, appearance: 0, manufacturer data: a705051001000000000000001b00ca, serviceUUID: 0xfe07, txPower: 0 
> 15:27:09.114 -> Advertised Device: Name: , Address: 75:6d:2b:0e:f5:29, manufacturer data: 4c001006191e23cfe2c1, txPower: 8 
> 15:27:17.923 -> Advertised Device: Name: , Address: 7e:d5:fc:b0:9c:b5, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:27:35.274 -> Advertised Device: Name: , Address: d7:96:ba:2b:33:e2, manufacturer data: 4c0012020000 
> 15:27:38.533 -> Advertised Device: Name: PROV_005d13Z, Address: 24:62:ab:e2:16:ee, serviceUUID: 0000ffff-0000-1000-8000-00805f9b34fb 
> 15:27:49.808 -> Advertised Device: Name: , Address: 86:2a:fd:a8:50:c6, manufacturer data: 650001c905, serviceUUID: 0xfe78
> 15:27:49.808 -> Service Data:
> 15:27:49.808 -> UUID: 0xfdf7, Data: ⸮X⸮⸮⸮⸮V⸮⸮ziEfs5 
> 15:27:54.071 -> Advertised Device: Name: , Address: 43:16:54:3d:42:ed, manufacturer data: 4c001005491c18dbfe 
> 15:27:54.360 -> Advertised Device: Name: , Address: 76:90:c3:e9:90:02, manufacturer data: e00001b9ca72c54c, serviceUUID: 0xfe9f
> 15:27:54.360 -> Service Data:
> 15:27:54.360 -> UUID: 0xfe9f, Data: nTF065F6mZs 
> 15:27:55.217 -> Advertised Device: Name: , Address: 7e:d5:fc:b0:9c:b5, manufacturer data: 4c00021561b5c6c7bc4249c982cc3fc6832f69ae00640001a6 
> 15:27:57.163 -> Advertised Device: Name: PROV_005d13Z, Address: 24:62:ab:e2:16:ee, serviceUUID: 0000ffff-0000-1000-8000-00805f9b34fb 
> 15:28:00.645 -> Advertised Device: Name: 846B215F5FC233AEE9, Address: f3:33:c0:55:58:95, appearance: 0, manufacturer data: a705051001000000000000001b00ca, serviceUUID: 0xfe07, txPower: 0 
> 15:28:06.468 -> Advertised Device: Name: , Address: 75:6d:2b:0e:f5:29, manufacturer data: 4c001006191e23cfe2c1, txPower: 8 
> 15:28:09.374 -> Advertised Device: Name: , Address: d7:96:ba:2b:33:e2, manufacturer data: 4c0012020000 
> 15:28:09.661 -> Advertised Device: Name: , Address: 4b:21:d4:48:87:b6, manufacturer data: e00001f2ca849dc4, serviceUUID: 0xfe9f
> 15:28:09.661 -> Service Data:
> 15:28:09.661 -> UUID: 0xfe9f, Data:  
> 
Reading through that, you see I get a "tick" at 15:26:56.912, at 15:26:59.964, 15:27:02.988, and ... never again for the next 60+ seconds.  The scan appears to continue to run, but my main loop() just... disappears.
Questions:  
- where is executing disappearing to, that my simple little loop() stops looping?
- am I unrealistic in thinking there's a way to scan and continue to process at the same time.  Even with the callback, the scan appears to block execution of the main loop

BTW, I don't think this is a "bug" or "issue", but I couldn't find a more appropriate forum to ask this question.  If there is one, let me know and I'll move there.  Thanks!

John

ab0oo, 2021-10-29:
edit:  I just found that setting:

ab0oo, 2021-10-29:
Edit:
  I just found that setting:
  pBLEScan->setActiveScan(false);
  as false, not true, stops the apparent infinite block of the main loop(), but I'm still seeing the scan itself blocking for 3 seconds every time it runs.

chegewara, 2021-10-29:
Of course you are seeing it since you are using blocking version of scan.

ab0oo, 2021-10-29:
> Of course you are seeing it since you are using blocking version of scan.

<sigh>. Of course it was that easy.  Is there an example that I missed of a NON-blocking scan?

chegewara, 2021-10-29:
https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEScan.h#L65

ab0oo, 2021-11-01:
Following @chegewara 's advice, I switched to non-blocking code, which works for a while, but inevitably stops hitting the callback after some random period of time, generally less than 90 seconds.
The code is here: [https://gist.github.com/ab0oo/4c49c4efc0f2a60c0a265bc4bc39e443], running on an ESP32-core_board_v2.
I am specifically searching for a BLE beacon being emitted from the Home Assistant app on my Android phone, which bears the same leading 0x4c00 manufacturers data as a normal iBeacon.
I'm unsure why the callback just stops being called; for the first ~10 iterations of the scan, the NimBLEScanResults.getCount() returns between 11 and 14 devices.  I can't imagine they all stop chattering at once.
Thoughts?

chegewara, 2021-11-02:
Just guessing now:
- by default duplicates are ignored,
- after stop scan or before start new scan you need to make some cleanup (one or both):
https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEScan.h#L77
https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEScan.h#L75

doudar, 2021-11-02:
I think it might help (can't hurt) to add a vTaskDelay(100/portTICK_PERIOD_MS); inside your main loop. 

It's a special delay that will let other processes such as garbage collection run.

ab0oo, 2021-11-03:
> I think it might help (can't hurt) to add a vTaskDelay(100/portTICK_PERIOD_MS); inside your main loop.
> 
> It's a special delay that will let other processes such as garbage collection run.
> 
> Otherwise your loop is consuming a lot of cycles for what it's actually doing.

Added this at the end of the loop(), but didn't really see any effect.  
I'm curious what you mean by "Otherwise your loop is consuming a lot of cycles for what it's actually doing."   I know writing to the Serial port is expensive, but I didn't think pulling millis() is too costly.  I could probably stop allocating a new long every loop, and got with static allocation, but where else am I being expensive?

ab0oo, 2021-11-03:
> Just guessing now:
> 
> * by default duplicates are ignored,
> * after stop scan or before start new scan you need to make some cleanup (one or both):
>   https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEScan.h#L77
>   https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLEScan.h#L75

At the end of my scanCompleteCB, I'm calling both clearResults() and clearDuplicateCache().  Right now, the code runs anywhere between 10 and 60 seconds before it stops hitting the scan complete callback.  After that, the loop continues to "tick", but no BLE callbacks (even after I let it run for hours).
I'm totally stumped, and blocked.
I've updated my gist if anyone wants to take a swing at this.  It's a pretty simple setup to reproduce.

h2zero, 2021-11-05:
Hi @ab0oo sorry for the delayed response, I have tested your gist code and it is working fine for me. As for your original code, it looks to be fine except for this line: `pBLEScan->setMaxResults(0);`, this should be removed for this use as it will delete all completed advertisement reports.

Also the line `pBLEScan->clearResults();` should be removed as well for the same reason. When starting a new scan the old results will be deleted unless the `isContinue` flag is set.

ab0oo, 2021-11-05:
> Hi @ab0oo sorry for the delayed response, I have tested your gist code and it is working fine for me. As for your original code, it looks to be fine except for this line: `pBLEScan->setMaxResults(0);`, this should be removed for this use as it will delete all completed advertisement reports.
> 
> Also the line `pBLEScan->clearResults();` should be removed as well for the same reason. When starting a new scan the old results will be deleted unless the `isContinue` flag is set.
> 
> This is assuming the original code is your intent, if you do not want to block the loop and instead want to scan and process the advertisements as they occur then the gist code will need amending which I can explain in the comments there.

It's possible I've got wonky hardware.  I've switched over from a 4 year old ESP32-core_board_v2 to a 3 year old Heltec LORA/ESP32 and that same code just runs and runs.  Granted, the coreboard has probably been flashed in excess of 500 times, but still, what a weird failure mode.
I've made the changes you suggested, and I'll leave a couple of these boards running this weekend while I'm away camping.  We'll see how things look when I get back Sunday.
Thanks for looking at this.  I'll close the issue on Sunday after I see that things just work.

ab0oo, 2021-11-08:
I'm not convinced there's not a bug in here somewhere.  The same code that runs on the Heltec (bare ESP32-D0WDQQ6) stops hitting the callback on the Wroom32D module on the devkitc.  I've run it on two Devkitc boards with the same result.
The interesting thing is that the NimBLEScan continues to run, giving me both "Updated Advertiser" and "New Advertiser" messages long after the callback stops being called, when setting the Core Debug Level level to Info.
Even after increasing the log level to Debug, there's nothing in the output to indicate why the Callback stops getting called.  The code just "ticks" along, with the callback getting hit every 2 seconds, until suddenly it never gets called again.  The main loop keeps ticking, firing off new scans, but the callback never fires.  Also, after the callback stops getting called, I never see:
13:38:53.200 -> D NimBLEScan: "discovery complete; reason=0"
show up in the logs again, either.  (comes from https://github.com/h2zero/NimBLE-Arduino/blob/f20115c4e4f2e146dc549eed074a8635ab16dfa0/src/NimBLEScan.cpp#L138)
I've altered the code to check for running scans before I fire off a new one in the main loop():
```
  if ( !pNimBLEScan->isScanning() ) {
    pNimBLEScan->start(scanTime, *scanCompleteCB, false);
  }
```
When I do that, the NimBLEScan continues to run forever, completely ignoring the scan timeout.
[scan_log.txt](https://github.com/h2zero/NimBLE-Arduino/files/7500386/scan_log.txt)
I've attached debug output showing this.  Point of interest is here:
> 13:52:41.001 -> Scan complete at 95, 7 devices found 
> 13:52:41.071 -> D NimBLEScan: ">> start(duration=2)"
> 13:52:41.071 -> D NimBLEScan: "<< start()"
> .....
> 13:52:43.069 -> D NimBLEScan: "discovery complete; reason=0"
> 13:52:43.069 -> Scan complete at 97, 7 devices found 
> 13:52:43.178 -> D NimBLEScan: ">> start(duration=2)"
> 13:52:43.178 -> D NimBLEScan: "<< start()"

Up to this point, the scan declares itself "complete, reason=0" within a few milliseconds of the 2 second scan length elapsing.  However, starting with the scan initiated at 13:52:43.178 , we never see it complete.  It just scans on and on and on...
It looks like whatever is supposed to be calling NimBLEScan::handleGapEvent with an event type of BLE_GAP_EVENT_DISC_COMPLETE never does so.
This behavior is consistently reproducible on both of my Devkitc boards; one is a few years old and has some miles on it, the other I pulled out of the sealed shipping wrapper today.  The only thing that changes is how long it runs before it hits the bug, but it never seems to be more than 2 or 3 minutes.
Not sure where to go from here.  I'd be happy to mail you my offending devices, if you think that would matter.

h2zero, 2021-11-09:
Are you using the code in the gist posted? Any modifications? Please update that with any changes if you made any and I will test it.

ab0oo, 2021-11-09:
The gist is current and exhibits the behavior I described above.  I'm currently fighting with a "raw" ESP32 WROOM-32D module in a standalone programmer, seeing if I can recreate it.  I'm also trying to install VSCode and all the appropriate doodads in a new VM on my Mac, so it might be a while...

h2zero, 2021-11-09:
Added a comment to your gist https://gist.github.com/ab0oo/4c49c4efc0f2a60c0a265bc4bc39e443#gistcomment-3955722
Not 100% certain that is the cause but seems most likely.

ab0oo, 2021-11-09:
Same result with your suggestions.  It looks like it never hits NimBLEScan::handleGapEvent.
Since you're unable to reproduce, I'm wondering if there's a regression, or a version difference.  FWIW, here's my compiler version:

> ~ xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r1/bin/xtensa-esp32-elf-gcc --version
> xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
> 
I'm using the same version (and getting the same results) both on the native MacOS compiler, and running it under Linux on a VM.  I'm going to hunt down an older version of the xtensa tools and see if that gives me the same results.  I'll update this issue with those finding.
I've updated my gist with your code suggestions, just to stay in sync.  Again, I appreciate your interest in this.  It would be great fun to find a bug in EspressIF's compiler/library.

ab0oo, 2021-11-09:
I'm stepping though the various arduino-esp32 releases to see if I can find one that works.
I get the same results (callback stops being called) with origin/master, origin/idf-release/v4.4, and origin/release/v1.0
It won't compile clean with origin/idf-release/v3.3 (probably a compiler mis-match).
Now for the kicker:
  I tried s/NimBLE/BLE/g and using the stock BLE libraries seems to work.  The longest the NimBLE library was able to run before the callback stopped being called was about 250 seconds; I'm currently running on the BLE library, and I'm over 1000 seconds with no missed callbacks.  Not sure how that helps us find the problem, but it unblocks me for now.

h2zero, 2021-11-13:
I'll try to have a closer look into this over the weekend and get back to you.

h2zero, 2021-11-21:
@ab0oo I'm still trying to reproduce this. Would you be able to provide the platformio.ini and any other code/settings? This also seems related to the issue @chegewara posted in #297 and I would like to identify the cause.

cicciocb, 2021-11-21:
@h2zero I can reproduce the same behaviour using latest arduino-esp32 release (2.0.1)

h2zero, 2021-11-21:
I was able to reproduce this now, it appears it's an issue with platformio as it does not occur when using Arduino IDE or IDF.

cicciocb, 2021-11-21:
@h2zero I think I found the problem that is inside the function
` ble_hs_timer_sched(int32_t ticks_from_now)`

In fact, the function `ble_npl_callout_is_active(&ble_hs_timer),` returns always true even if the timer has expired.
Following the documentation of the RTOS function that is called by  `ble_npl_callout_is_active(&ble_hs_timer),`
```
**
 * TickType_t xTimerGetExpiryTime( TimerHandle_t xTimer );
 *
 * Returns the time in ticks at which the timer will expire.  If this is less
 * than the current tick count then the expiry time has overflowed from the
 * current time.
 */
```

This means that the timer can be expired if its current tick time is greater than the value reported by `ble_npl_callout_get_ticks(&ble_hs_timer)`

So, the is_active condition must be modified taking into account this condition

h2zero, 2021-11-21:
Thanks for looking into it! 

You're in the right place but the call you referenced actually results in this `return xTimerIsTimerActive(co->handle) == pdTRUE;` 

I can see that the timer reset is not being called, however I don't understand why that would happen yet as it only applies to something in platformio that causes this.

h2zero, 2021-11-21:
A little more testing with platformio config:

This works fine:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
monitor_speed = 115200
```

This fails to trigger the scan stop timer:
```[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
monitor_speed = 115200
```

cicciocb, 2021-11-21:
@h2zero 
I'm using arduino esp-32 arduino using vscode inside visual studio 2009.
As per my testing, the RTOS function xTimerIsTimerActive returns active even if the time has expired.
So, to be sure that it is still active, the value coming from xTimerGetExpiryTime must be checked against the current ticks xTaskGetTickCountFromISR

Basically the patch I did is the following
`static void
ble_hs_timer_sched(int32_t ticks_from_now)
{
    ble_npl_time_t abs_time;

    if (ticks_from_now == BLE_HS_FOREVER) {
        return;
    }

    /* Reset timer if it is not currently scheduled or if the specified time is
     * sooner than the previous expiration time.
     */
	uint32_t tt = ble_npl_time_get(); // gets current ticks
	// takes into account that the timer can be overflowed compared to the current ticks
	bool is_active = (ble_npl_callout_get_ticks(&ble_hs_timer) < tt) && ble_npl_callout_is_active(&ble_hs_timer);

    abs_time = ble_npl_time_get() + ticks_from_now;
    if (!is_active ||
            ((ble_npl_stime_t)(abs_time -
                               ble_npl_callout_get_ticks(&ble_hs_timer))) < 0) {
        ble_hs_timer_reset(ticks_from_now);
    }
}
`

h2zero, 2021-11-21:
@cicciocb You're correct, the failure is in the result from `xTimerIsTimerActive`.

Logs with working pio config:
```
Setup complete, starting scans.
reset 1
timer active: 0, exp: 316, now: 316
reset 1999
Tick: 1
Tick: 2
reset 1
Scan complete at 2, 3 devices found 
reset 1
timer active: 0, exp: 2317, now: 2323
reset 1999
timer active: 0, exp: 2323, now: 2324
reset 1998
Tick: 3
Tick: 4
reset 1
Scan complete at 4, 4 devices found  
reset 1
timer active: 0, exp: 4324, now: 4330
reset 1999
timer active: 0, exp: 4330, now: 4330
reset 1999
```

Logs from non-working pio config:
```
Setup complete, starting scans.
reset 1
timer active: 0, exp: 318, now: 318
reset 1999
Tick: 1
Tick: 2
reset 1
Scan complete at 2, 3 devices found  
reset 1
timer active: 1, exp: 2319, now: 2324  --  ?? timer should not be active, it has expired 
timer active: 1, exp: 2324, now: 2324  --  ?? timer should not be active, it has expired
```

cicciocb, 2021-11-21:
```
@h2zero Probably something changed in the RTOS with the new release; maybe the compiler itself that switched from 
gcc8_4_0-esp-2021r1 to gcc8_4_0-esp-2021r2

```

h2zero, 2021-11-21:
For interest sake I ran this again with arduino IDE with my debug logs and found something interesting.

```
Setup complete, starting scans.
reset 1
timer active: 0, exp: 299, now: 299
reset 1999
Tick: 1
Tick: 2
reset 1
Scan complete at 2, 5 devices found 
reset 1
timer active: 1, exp: 2299, now: 2303 -- ?? should not be active
timer active: 0, exp: 2304, now: 2304 -- correct now?
reset 1999

--- continues for a while

reset 1
timer active: 1, exp: 12324, now: 12327
timer active: 1, exp: 12328, now: 12328 -- now we fail
```

So I take back my earlier statement about it being platformio, definitely seems to be in arduino and or IDF, it just didn't show up until the logging was added.

h2zero, 2021-11-21:
Just created a test branch/PR for this, please test and report back if you can.

cicciocb, 2021-11-21:
Hi,
just tested and I confirm that it works.

h2zero, 2021-11-21:
Thanks for testing, I hope it does not affect any other parts of the stack.

FWIW I believe I have found the offending code in FreeRTOS here:
https://github.com/espressif/esp-idf/blob/451ce8a7ed486c6c6045189db57106ca7107abdb/components/freertos/FreeRTOS-Kernel/timers.c#L952

That calls the timer callback before clearing it as active, I'll do some tests with IDF and see, might need to make a PR there.

h2zero, 2021-11-22:
Looks like the true fix is available upstream here:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/052e364686b1cf7082b8979fecf211121402bf74/timers.c#L823

You can see the difference in the blocks:
https://github.com/espressif/esp-idf/blob/451ce8a7ed486c6c6045189db57106ca7107abdb/components/freertos/FreeRTOS-Kernel/timers.c#L848

I tested and confirmed this in IDF, so I guess the workaround will be needed to compensate for bugged releases. I'm going to try to use esp_timer instead for the master branch fix, 1.3.3 will use the workaround proposed in the PR.

ab0oo, 2021-11-22:
> Looks like the true fix is available upstream here: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/052e364686b1cf7082b8979fecf211121402bf74/timers.c#L823
>
> You can see the difference in the blocks: [...]

Thank you so much for running this down.  I apologize I got scarce at the end: my partner had a hip replacement and I've been focused on her.  I'll pull these changes into my code and look for improvements.

h2zero, 2021-11-22:
No worries, please test at your convenience. I hope your partner is recovering well!

beats0126, 2022-01-03:
hi @h2zero,
i using current version of nimble (v1.3.3)
and arduino esp32 core v2.0.2
arduino ide v1.8.16

i running a ble scan in a loop but after few hour or few day, the serial monitor keep print out these message and then the ble scan will stopped

CRIT NimBLEDevice: "Resetting state; reason=19, "
CRIT NimBLEScan: "Unable to scan - Host Reset"
Devices found: 0
Scan done!
141272                     <------- esp32 freeHeap
scan_evt timeout
scan_evt timeout

sorry but i need some help on this. thanks!

[image]

beats0126, 2022-01-03:
also, i will try use back default esp32 ble library to see it will happen same problem or not. will update to here after this.

h2zero, 2022-01-10:
@ab0oo have you been able to confirm the issue is now resolved? Please close this if so, thanks!

tonton81, 2022-01-10:
Don't use printf in the callback. I noticed the ISR doesn't like always writing to serial from that callback. If you use serial print and not the entire device data it will work properly.

h2zero, 2022-01-15:
Closing this for now as I believe it's been resolved. Please re-open if the issue occurs again.

closed 2022-01-15 by h2zero