h2zero/NimBLE-Arduino #309: Disconnected/Reconnection every 90sec with BLE Scan Sample

ArcadeD3vil, 2021-10-31:
Hi mates,
any reason for that issue? any setting to change to avoid that?
with the Refactored_original_examples/ BLE Scan this behavior is not present.....

Cheers !!

h2zero, 2021-11-25:
Hi @ArcadeD3vil, sorry for the delay, could you provide any logs? There are many reasons for disconnection and the log output, set to debug level, will greatly help the investigation.

Generally speaking, disconnections can happen if the connection supervision timeout is too low in a noisy 2.4ghz environment. Sometimes this can be compensated for in connection settings.

tonton81, 2022-01-10:
Some devices act like this, when I did the Morimoto XBT controller, if you don't do a read/write within the first 30 seconds of connection, the XBT will disconnect you. If you do do a read/write within first 30 seconds of connection, you will remain connected indefinately even if you're idle and not talking to it.

h2zero, 2022-04-04:
Closing for now, please reopen if you're still having this issue.

closed 2022-04-04 by h2zero