h2zero/NimBLE-Arduino #310: How can I avoid duplicate reads during scanning?

Zimbu98, 2021-11-04:
I am running the following code:

```
#include <Arduino.h>

#include <NimBLEDevice.h>
#include <NimBLEAdvertisedDevice.h>
#include "NimBLEEddystoneURL.h"
#include "NimBLEEddystoneTLM.h"
#include "NimBLEBeacon.h"

#define ENDIAN_CHANGE_U16(x) ((((x)&0xFF00) >> 8) + (((x)&0xFF) << 8))

int scanTime = 5; //In seconds
BLEScan *pBLEScan;

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
    void onResult(BLEAdvertisedDevice *advertisedDevice)
    {
      if (advertisedDevice->haveName())
      {
        Serial.print("Device name: ");
        Serial.println(advertisedDevice->getName().c_str());
        Serial.println("");
      }
    }
};

void setup()
{
  Serial.begin(115200);
  Serial.println("Scanning...");

  NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);
  
  /** *Optional* Sets the scan filter cache size in the BLE controller.
   *  When the number of duplicate advertisements seen by the controller
   *  reaches this value it will clear the cache and start reporting previously
   *  seen devices. The larger this number, the longer time between repeated
   *  device reports. Range 10 - 1000. (default 20)
   *
   *  Can only be used BEFORE calling NimBLEDevice::init.
   */
    NimBLEDevice::setScanDuplicateCacheSize(300); //setting this to 20 results in lots of duplicates
  
    NimBLEDevice::init("");
  
  //BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99); // less or equal setInterval value
  
}

void loop()
{
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false); 
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults(); // delete results fromBLEScan buffer to release memory
  delay(10000); //pause for x seconds
}

```

So it is a 5 second scan, then 10 second pause. I want the scan to return **only one data report from each nearby iBeacon**. 

About half of the time, I am getting one report for each beacon, but the other half of the time, I am getting some duplicates. 

Is there something I can do to avoid duplicates?

Thanks for a great library!

h2zero, 2021-11-05:
Hi @Zimbu98  looks like you will need to adjust the scan filter settings. As per the example [here](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/NimBLE_Scan_Continuous/NimBLE_Scan_Continuous.ino).

```
/** *Optional* Sets the filtering mode used by the scanner in the BLE controller.
 *
 *  Can be one of:
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE (0) (default)
 *  Filter by device address only, advertisements from the same address will be reported only once.
 *
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DATA (1)
 *  Filter by data only, advertisements with the same data will only be reported once,
 *  even from different addresses.
 *
 *  CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE (2)
 *  Filter by address and data, advertisements from the same address will be reported only once,
 *  except if the data in the advertisement has changed, then it will be reported again.
 *
 *  Can only be used BEFORE calling NimBLEDevice::init.
*/
  NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);

/** *Optional* Sets the scan filter cache size in the BLE controller.
 *  When the number of duplicate advertisements seen by the controller
 *  reaches this value it will clear the cache and start reporting previously
 *  seen devices. The larger this number, the longer time between repeated
 *  device reports. Range 10 - 1000. (default 20)
 *
 *  Can only be used BEFORE calling NimBLEDevice::init.
 */
  NimBLEDevice::setScanDuplicateCacheSize(200);
```

Adjusting the `setScanDuplicateCacheSize` setting as shown above to a higher value may be what you need (MAX 1000).

Zimbu98, 2021-11-05:
Hi H20,

Thanks for your fast response.

Re the scan filters:

Only the first one, CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE, compiles for me. 

If I try to use CONFIG_BTDM_SCAN_DUPL_TYPE_DATA or CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE , I always get the following error:

**error: 'CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE' was not declared in this scope
     NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);**

I am using NimBLE-Arduino Version 1.3.1 library, installed via the library manager in Arduino IDE. 

I have also tried adjusting the cache size but not seen any improvement. But maybe the cache size would help if I could use one of the other scan filters without errors.

h2zero, 2021-11-05:
That's odd, what version of the arduino core are you using?

Zimbu98, 2021-11-05:
1.8.15

h2zero, 2021-11-05:
I can confirm the issue with that, added to my bug list. You can just use the number value those macros convert to `CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE (2)` for that case just use `2`.

Zimbu98, 2021-11-05:
Just checked and:

NimBLEDevice::setScanFilterMode(1);

compiles with no errors. I will play around with these two optional filters and the cache size tomorrow to try to find the perfect balance. 

Thanks!

h2zero, 2021-11-20:
Just an update to this, the macros have been added and should now work.

h2zero, 2022-04-04:
Closing as resolved.

closed 2022-04-04 by h2zero