h2zero/NimBLE-Arduino #313: Comiplation Error with freertos

gktval, 2021-11-24:
Hi, I am attempting to compile the library using the Arduino Nano 33 BLE. When compiling, I am getting an error that it cannot find FreeRTOS.h. I installed FreeRTOS through the library manager, but the error remains. Do I need to place freeRTOS elsewhere or should it be automatically installed when NimBLE is installed? Thanks

[image]

h2zero, 2021-11-25:
Hi @gktval, sorry but at the moment this library only supports the esp32 arduino core. 

I have a nearly ready core for arm processors that I started months ago and have not had time to complete it yet but hope to soon.

It will be announced here when its available.

gktval, 2021-11-25:
Ah. That's good to hear.  I will close this then.

closed 2021-11-25 by gktval

h2zero, 2021-11-27:
@gktval Since you asked about this I thought I'd let you know that the core is now available if you're interested. Please see #314