h2zero/NimBLE-Arduino #314: Not just for esp32 anymore!

h2zero, 2021-11-27:
The recent development in the master branch has added support for using this library on Nordic nRF5 devices.

A few months ago I started making an Arduino core to support this new functionality but ran out of time to finish it. Today I'm happy to announce that the new core is available (in development only form) for anyone that would like to check it out.

You can find the project here https://github.com/h2zero/n-able. I would love to have anyone interested try it out, and if you'd like to do some development work, I'm looking for some collaborators to help finish it up!

Now you can use this library for your ultra low power BLE devices! So far I have one of my nrf51 projects running with this for nearly 4 months on a CR2032 😄 .

hdpklm, 2021-12-16:
interesed in developer

h2zero, 2021-12-17:
> interesed in developer

Please post in the issue thread here https://github.com/h2zero/n-able/issues/1 and let me know what you have for devices to test/develop with.

hdpklm, 2021-12-17:
> and if you'd like to do some development work, I'm looking for some collaborators to help finish it up!
I have no other than esp32. but I can buy one. and do some testing and developing the project in my free time, and I have some ideas to include in the project, it is based on BLE with OTA in the bootloader. to reduce the size of the load file and you don't need to include and recompile the ble and ota file every time you build a new version. but this idea needs a lot of work in the microcontroller infrastructure.
but it is very necessary in the developer

h2zero, 2021-12-17:
Agreed, that was in the plan I have for it in the long term. I would be more than happy to have your assistance.

hdpklm, 2021-12-17:
ok very good. we can talk tomorrow if you have time

h2zero, 2021-12-18:
If you're on gitter it would be best to converse there.

hdpklm, 2021-12-19:
ok, I create a gitter accoun, but i not sure how to use, send me an invitation or eveny think

closed 2024-06-08 by h2zero