h2zero/NimBLE-Arduino #315: BLERemoteService* pRemoteService = pClient->getService(serviceUUID) fails on refactored BLE_client, but works on original

CharlesRDavis, 2021-12-02:
Using BLE_client from the "stock" BLE, and modifying only the BLEUUIDs to match an actual device, I can connect, find service, find characteristic. I.e. it works.

If I use the refactored one in the NimBLE examples,  it fails after connect.   
 BLERemoteService* pRemoteService = pClient->getService(serviceUUID); always returns a null pointer. 

I would very much like to use NimBLE, but no matter what I do I cannot get past this. 

Any help would be most appreciated.
Charles Davis
[BLE_clients.zip](https://github.com/h2zero/NimBLE-Arduino/files/7644989/BLE_clients.zip)

h2zero, 2021-12-03:
Hi @CharlesRDavis, the problem you are seeing is simply due to using the 128bit base uuid version of the service uuids. The device you're connecting to does not respond to those and instead expects the 16bit uuid.

Change these:
```
static BLEUUID serviceUUID("0000ff00-0000-1000-8000-00805f9b34fb"); //xiaoxiang bms original module
static BLEUUID charUUID("0000ff01-0000-1000-8000-00805f9b34fb"); //xiaoxiang bms original module
```
to:
```
static BLEUUID serviceUUID("ff00"); //xiaoxiang bms original module
static BLEUUID charUUID("ff01"); //xiaoxiang bms original module
```

The reason this works with bluedroid, I suspect, is that it will convert those uuids to 16bit when the 128bit base is used somewhere in it's stack, which NimBLE does not. I plan to add this in the future.

CharlesRDavis, 2021-12-03:
Totally solved it. Great. Thanks so much.
Charles Davis

closed 2021-12-03 by CharlesRDavis