h2zero/NimBLE-Arduino #316: `Client busy` error after first conversation

8none1, 2021-12-03:
I think that this is a support request rather than a bug report, sorry.

I'm trying to communicate with some cheapo BTLE enabled LED lights.  I've had reasonable success, and NimBLE has help me enormously.  I am not a good C++ developer.  These lights are also really finicky and have proven to be unreliable.

I've been rewriting my code to try and stay more aligned with the Client example and generally things are working pretty well.
I've hit a situation now that I can't resolve, and I'm looking for some ideas.

I can connect to the light, send a command, that command is acted on and callbacks fire etc.  However, when I try and send a second command, "turn off" for example, I get the error `"Client busy"`.
I've tracked this back to here:  https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/src/NimBLEClient.cpp#L188

I expect that this is something to do with `m_pTaskData != nullptr` not being null.  Which I then figured was because I still have a callback registered.

If I call `disconnect()` then I can reconnect successfully and issue more commands.

Is that simply how it works?  If you have a callback registered then you can't do anything else until that routine is finished?

Here's some logs in case it's useful:
```
Received mqtt message
Turning on
Created new client
D NimBLEClient: ">> connect(78:82:a4:00:05:1e)"
GAP procedure initiated: connect; peer_addr_type=0 peer_addr=78:82:a4:00:05:1e scan_itvl=16 scan_window=16 itvl_min=12 itvl_max=12 latency=10 supervision_timeout=100 min_ce_len=0 max_ce_len=0 own_addr_type=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
0x0d 0x20 0x19 0x10 0x00 0x10 0x00 0x00 0x00 0x1e 0x05 0x00 0xa4 0x82 0x78 0x00 0x0c 0x00 0x0c 0x00 0x0a 0x00 0x64 0x00 0x00 0x00 0x00 0x00 
I NimBLEClient: "Connection established"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
oD NimBLEClient: "Got Client event "
I NimBLEClient: "Connected event"
GATT procedure initiated: exchange mtu
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0xff 0x00 
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
0x16 0x20 0x02 0x00 0x00 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=16 data=0x0c 0x00 0x05 0x00 0x12 0x3d 0x08 0x00 0x07 0x00 0x07 0x00 0x00 0x00 0xc8 0x00 
L2CAP - rxed signalling msg: 0x12 0x3d 0x08 0x00 0x07 0x00 0x07 0x00 0x00 0x00 0xc8 0x00 
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 7, MaxInterval: 7, Latency: 0, Timeout: 200"
n connect callback
D NimBLEClient: "<D<oin g socme coonnenctionn paeramsc stutff()"

Connected to device: D NimBLECli7ent: "Accepted peer params"
GAP procedure initiated: connection parameter update; conn_handle=0 itvl_min=7 itvl_max=7 latency=0 supervision_timeout=200 min_ce_len=0 max_ce_len=0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0013 len=14
0x13 0x20 0x0e 0x00 0x00 0x07 0x00 0x07 0x00 0x00 0x00 0xc8 0x00 0x00 0x00 0x00 0x00 
host tx hci data; handle=0 length=10
ble_hs_hci_acl_tx(): 0x00 0x00 0x0a 0x00 0x06 0x00 0x05 0x00 0x13 0x3d 0x02 0x00 0x00 0x00 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x03 0x60 0x00 
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=0 mtu=96"
8:82:a4:00:05:1e
RSSI: D NimBLEClient: ">> getRssi()"
ble_hs_hci_cmd_send: ogf=0x05 ocf=0x0005 len=2
0x05 0x14 0x02 0x00 0x00 
-110
D NimBLEClient: ">> getService: uuid: 0xffd0"
D NimBLEClient: ">> retrieveServices"
GATT procedure initiated: discover service by uuid; uuid=0xffd0
host tx hci data; handle=0 length=13
ble_hs_hci_acl_tx(): 0x00 0x00 0x0d 0x00 0x09 0x00 0x04 0x00 0x06 0x01 0x00 0xff 0xff 0x00 0x28 0xd0 0xff 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x07 0x0a 0x00 0x0f 0x00 
D NimBLEClient: "Service Discovered >> status: 0 handle: 10"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService(): 0xffd0"
host tx hci data; handle=0 length=13
ble_hs_hci_acl_tx(): 0x00 0x00 0x0d 0x00 0x09 0x00 0x04 0x00 0x06 0x10 0x00 0xff 0xff 0x00 0x28 0xd0 0xff 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x06 0x10 0x00 0x0a 
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
[... 66 lines cut ...]
host tx hci data; handle=0 length=11
ble_hs_hci_acl_tx(): 0x00 0x00 0x0b 0x00 0x07 0x00 0x04 0x00 0x08 0x09 0x00 0x09 0x00 0x03 0x28 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x08 0x09 0x00 0x0a 
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: -1"
D NimBLERemoteService: "<< Characteristic Discovered"
D NimBLERemoteService: "<< retrieveCharacteristics()"
D NimBLERemoteCharacteristic: ">> writeValue(), length: 3"
GATT procedure initiated: write no rsp; att_handle=9 len=3
ble_att_clt_tx_write_cmd(): 0xcc:0x23:0x33host tx hci data; handle=0 length=10
ble_hs_hci_acl_tx(): 0x00 0x00 0x0a 0x00 0x06 0x00 0x04 0x00 0x52 0x09 0x00 0xcc 0x23 0x33 
Correctly wrote data to device without crashing!
Power on worked
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connection parameters updated."
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0035 len=5
0x35 0x0c 0x05 0x01 0x00 0x00 0x02 0x00 
Received mqtt message
D NimBLEClient: ">> connect(78:82:a4:00:05:1e)"
E NimBLEClient: "Client busy, connected to 78:82:a4:00:05:1e, id=0"
Tried to reconnect to known device, but failed
Power off failed

```

Thanks for your work on NimBLE.  I've just ordered an NRF dongle to have a play with that too, excited to see NimBLE supported there too.

h2zero, 2021-12-03:
Hi @8none1, from your log it appears you're trying to connect to the device when you already have a connection to it, that's the meaning of the error message.

I can also see why this device may be a bit difficult to work with. It's using very tight connection parameters. You could try rejecting it's connection parameter update request and see if that helps maintain connection better.

8none1, 2021-12-03:
Yeah, I think I am trying to re-connect to a device with an existing connection, so I'm trying to re-connect like this (I copied this from the example client):

```
...
...
bool do_write(NimBLEAddress bleAddr, const uint8_t* payload, size_t length){
    digitalWrite(LED, !digitalRead(LED));
    // New reusing clients code starts here
    NimBLEClient * pClient = nullptr;
    if (NimBLEDevice::getClientListSize()) {
        // There are some existing clients available, try and find one to reuse
        pClient = NimBLEDevice::getClientByPeerAddress(bleAddr);
        if (pClient) {
            if (!pClient->connect(bleAddr, false)) {
                Serial.println("Tried to reconnect to known device, but failed");
                return false;
            } else {
                Serial.println("Correctly reconnected to device");
            }
        } else {
            Serial.println("No existing client, trying an old one");
            pClient = NimBLEDevice::getDisconnectedClient();
        }
    }
...
...
```

I'm seeing `Serial.println("Tried to reconnect to known device, but failed");` on the screen, so it looks like it's correctly trying to reuse an existing connection, but `connect` on that fails (presumably because it's already connected).

So... what I tried was commenting out the `return false` as seen here:
https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/examples/NimBLE_Client/NimBLE_Client.ino#L142

and hey presto!  :magic_wand: I am connected, my callbacks work, and I can send new commands to the device!

Does that mean the example is wrong?  It shouldn't be returning `false`?  If so, I can prepare a PR if it will help.

Next I will experiment :scientist: with changing the connection parameters to see if I can make it more reliable at range.

Thanks!

h2zero, 2021-12-04:
The return false in the example is correct, it's saying the connection failed. 

I suspect the fault is actually due to the disconnect event not yet being triggered, which can happen due to latency. 
How fast are you sending the commands to the device? 

For confirmation I suggest adding a loop after the disconnect call testing the return value of `pClient->isConnected()` until false before returning from that function.

8none1, 2021-12-04:
Aha! I think I understand now. :tada: 

Indeed, I am not disconnecting from the device - because I want to listen to the notifications for as longs as possible.  AIUI `disconnect()` will tear down the client and remove the notification callback.

So my code uses the "reconnect using an existing client if one exists" code from the example without the guarantee that the connection has been closed before being called again. The example code expects the client to have been disconnected before being called again.
The assumption is that if a call to `connect()` fails it's because it could not connect - not because perhaps it's already connected.

So it's as simple as wrapping the `getClientByPeerAddress()` section in a `if (!pClient->isConnected())` statement to catch the situation where one doesn't need to connect after getting a client.  If `getClientByPeerAddress()` returns a client, we check to see if that client is already connected, and if it is then skip over the `connect()` parts.

```
...
...
bool do_write(NimBLEAddress bleAddr, const uint8_t* payload, size_t length){
    digitalWrite(LED, !digitalRead(LED));
    NimBLEClient * pClient = nullptr;
    if (NimBLEDevice::getClientListSize()) {
        // There are some existing clients available, try and find one to reuse
        pClient = NimBLEDevice::getClientByPeerAddress(bleAddr);
        if (pClient) {
            // It's possible (and in our case, likely) that we're still connected to the device, so we need to deal with that here
            if (!pClient->isConnected()) {
                if (!pClient->connect(bleAddr, false)) {
                    Serial.println("Tried to reconnect to known device, but failed");
                    // The connection failed.  It can fail for two reasons; 1) it's already connected or 2) it can't connect
                    return false;
                } else {
                    Serial.println("Correctly reconnected to device");
                }
            } else {
                Serial.println("Found existing connected client, so using that");
            }
        } else {
            Serial.println("No existing client, trying an old one");
            pClient = NimBLEDevice::getDisconnectedClient();
        }
    }
...
...
```
This seems to have done the trick!

Thank you for the help.  NimBLE-Arduino is a great project.

closed 2021-12-04 by 8none1

h2zero, 2021-12-04:
Glad you found a solution!