h2zero/NimBLE-Arduino #317: Proposal to change the memory allocation to support modules with or without PSRAM

cicciocb, 2021-12-04:
Hi @h2zero,
if  CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL is defined and the module does not have PSRAM, the program crashes.

I propose to slightly change the memory allocation routines in order to allocate the memory even if the PSRAM is not available.
This will enable to put the same code in modules with or without PSRAM.

```
IRAM_ATTR void *nimble_platform_mem_malloc(size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    //return heap_caps_malloc(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
	return heap_caps_malloc_prefer(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_malloc_prefer(size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return malloc(size);
#endif
}

IRAM_ATTR void *nimble_platform_mem_calloc(size_t n, size_t size)
{
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_calloc(n, size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    //return heap_caps_calloc(n, size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
	return heap_caps_calloc_prefer(n, size, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
#elif CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_calloc_prefer(n, size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#else
    return calloc(n, size);
#endif
}
```

h2zero, 2021-12-04:
This proposal would be better served if made to IDF. We shouldn't make core file changes like this unless there are serious bugs or support issues. It's hard enough to maintain the current ones 😄.

cicciocb, 2021-12-05:
@cicciocb 
I don't know how to send the proposal to IDF, but I understand your point of view ;-)

h2zero, 2021-12-05:
@cicciocb I mean to propose the change to espressif here https://github.com/espressif/esp-idf. The file you want to change is maintained there.

closed 2022-01-15 by h2zero