h2zero/NimBLE-Arduino #318: Controlling the level of logs in NimBLE for ArduinoIDE 1.8.16

jaggil, 2021-12-05:
Hello,
After many tests, I can't get the NimBLE library logs to have the level I want.
In the serial output I always have "I NimBLEScan" and "D NimBLEScan" logs every time a BLE device is detected:
I NimBLEScan: "New advertiser: 7a: 0a: 85: 19: a4: 74"
D NimBLEScan: "erase device: 7a: 0a: 85: 19: a4: 74"
I NimBLEScan: "New advertiser: 46: 82: a0: 34: 74: a0"
D NimBLEScan: "erase device: 46: 82: a0: 34: 74: a0"
.
.
..........
With the function "Serial.printf (" Level log NimBLEScan =% d \ n ", esp_log_level_get (" NimBLEScan "));" I always get "Level log NimBLEScan = 1"

In my last tests I have the following configuration:
#define CONFIG_LOG_MAXIMUM_LEVEL ESP_LOG_NONE
#define LOG_LOCAL_LEVEL ESP_LOG_NONE
#define CONFIG_BT_NIMBLE_LOG_LEVEL 5

But I keep getting prints of all the BLE devices in my environment which are many:
None of the options I have tried have worked for me
Please, how should I set the log levels for NimBLE correctly?
Thanks in advance
My development data is:
OS = Windows10
ArduinoIDE: 1.8.16
ESP32 SDK: 2.0.1
NimBLE: 1.3.3

h2zero, 2021-12-05:
Hi @jaggil the current release uses the Arduino core debug level set in the Arduino IDE menu. This has been changed in the master branch to set a more appropriate macro setting in nimconfig.h `NIMBLE_CPP_DEBUG_LEVEL`. If you'd rather continue using the current release you can take a look at https://github.com/h2zero/NimBLE-Arduino/blob/1.3.3/src/NimBLELog.h and make the changes you'd like there.

h2zero, 2021-12-07:
Glad the master branch solution is working for you. 

I will be making a release of this branch at some point in the future as a 2.x version due to planned breaking changes, which the log change is one of.

I don't have a timeline for this to be complete as there is still a long list of todo items and testing and my time is more limited lately.

closed 2021-12-15 by jaggil