h2zero/NimBLE-Arduino #321: Exception was unhandled

doudar, 2021-12-19:
I hadn't seen this one previously. Happened on a unit boot during testing. It very well could be in my code.
`
N 18451 BLE_Client: "trying to connect to PM"
N 18452 BLE_Client: "Forming a connection to:  3c:61:05:30:2a:66"
N 18453 BLE_Client: " - Created client"
N 19485 BLE_Common: "06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 21455 BLE_Client: " - Connected to server"
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.`

Exception decoder:
`
PC: 0x401aeffe: put_le16 at .pio/libdeps/release/NimBLE-Arduino/src/nimble/porting/nimble/src/endian.c line 28
EXCVADDR: 0x00000001`
`
Decoding stack results
0x401aeffb: put_le16 at .pio/libdeps/release/NimBLE-Arduino/src/nimble/porting/nimble/src/endian.c line 28
0x400e9cc8: ble_hs_hci_cmd_send_buf at .pio/libdeps/release/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_cmd.c line 78
0x400e97a9: ble_hs_flow_tx_num_comp_pkts at .pio/libdeps/release/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_flow.c line 95
0x400e97c3: ble_hs_flow_event_cb at .pio/libdeps/release/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_flow.c line 116
0x400ef8aa: nimble_port_run at .pio/libdeps/release/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h line 132
0x400dfe57: NimBLEDevice::host_task(void*) at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEDevice.cpp line 791
`

h2zero, 2021-12-19:
I don't know where this could be coming from. Looks like something wasn't initialized or is NULL, but the backtrace is indicating an error in flow control which is "under the hood" stack stuff.

Is this repeatable?

doudar, 2021-12-20:
> I don't know where this could be coming from. Looks like something wasn't initialized or is NULL, but the backtrace is indicating an error in flow control which is "under the hood" stack stuff.
> 
> Is this repeatable?

Unfortunately I’ve only seen it a couple times. I’ll let you know if anything concrete can be connected to it.

doudar, 2021-12-21:
Okay, I think it's from calling `pClient->getRssi();` immediately after server connection.  Occasionally causes the exception.

h2zero, 2021-12-22:
Interesting, the backtrace does not indicate anything related to that. I'll see if I can confirm this.

doudar, 2021-12-22:
It could be coincidence but I haven't seen it since commenting out:
https://github.com/doudar/SmartSpin2k/blob/6501084d291c4df8bec11148f4511bde1a13c192/src/BLE_Client.cpp#L214

h2zero, 2021-12-22:
That makes sense, you should consider checking the return value at https://github.com/doudar/SmartSpin2k/blob/6501084d291c4df8bec11148f4511bde1a13c192/src/BLE_Client.cpp#L212

If it returns `false` and you try to get the rssi value there could be a failure (also the rest of the code wouldn't work there).

doudar, 2021-12-26:
Thanks @h2zero . This seems to be the issue for certain. Closing.

closed 2021-12-26 by doudar