h2zero/NimBLE-Arduino #322: Getting guru error while using setvalue in loop

yashram96, 2021-12-27:
i want update status of lamp. so using  setvalue in loop. But i am getting error 
"Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled" please help me

h2zero, 2021-12-27:
I can help you with this but I need more info. Can you share your code?

peetjh, 2022-01-25:
I am having the same issue, below is my code.
`#include <NimBLEDevice.h>

NimBLECharacteristic* pCharacteristic = NULL;
#define BLEDescriptorCallbacks  NimBLEDescriptorCallbacks
bool deviceConnected = false;
bool oldDeviceConnected = false;
unsigned long startTime = 0;
int ledPin = 2;
int ledState = LOW;
int btCount = 0;

class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
      deviceConnected = true;
      Serial.println("DeviceConnected");
    }

    void onDisconnect(NimBLEServer* pServer) {
      deviceConnected = false;
      Serial.println("DeviceDisconnected");
    }
};
class DescriptorCallbacks : public NimBLEDescriptorCallbacks {
    void onWrite(NimBLEDescriptor* pDescriptor) {
        std::string dscVal((char*)pDescriptor->getValue(), pDescriptor->getLength());
    }
    void onRead(NimBLEDescriptor* pDescriptor) {
        //Serial.print(pDescriptor->getUUID().toString().c_str());
        //Serial.println(" Descriptor read");
    }    
};

void setup()
{
    Serial.begin(115200);
    pinMode(2, OUTPUT);   //NEEDED FOR BOOT AND LED FLASHING
    digitalWrite(2, LOW);   //NEEDED FOR BOOT
    NimBLEDevice::init("ESP32_TEST");
    
    NimBLEServer *pServer = NimBLEDevice::createServer();
    NimBLEService *pService = pServer->createService("331F1570-8A00-9A28-CBE1-E1D4212D53EB");
    NimBLECharacteristic *pCharacteristic = pService->createCharacteristic("331F1571-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
    NimBLECharacteristic *pCharacteristic1 = pService->createCharacteristic("331F1572-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE_NR);
    
    pService->start();
    
    pCharacteristic->setValue("OOO");
    NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID("331F1570-8A00-9A28-CBE1-E1D4212D53EB"); 
    pAdvertising->start(); 
}
void loop()
{
      if (deviceConnected == false){
      unsigned long loopTime = millis() - startTime; //Calculate loop time
      if (loopTime >= 2000) //Check that the time is 1 second
      {startTime = millis();
        if(ledState == LOW){
          ledState = HIGH;
        }
        else{
          ledState = LOW;
        }
        digitalWrite(ledPin, ledState);
        Serial.println(ledState);
        pCharacteristic->setValue("XXX");
      }
        btCount = 0; 
      }
      else{
        if (btCount == 0){
        ledState = HIGH;
        digitalWrite(ledPin, ledState);
        Serial.println(ledState);
        btCount = 1;
      } 
    }
}`

h2zero, 2022-01-25:
Your issue is not initializing the global:
 `NimBLECharacteristic* pCharacteristic = NULL;` 

and then using it in the loop:
`pCharacteristic->setValue("XXX");`

Changing this:
```
NimBLECharacteristic *pCharacteristic = pService->createCharacteristic("331F1571-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
```

To:
```
pCharacteristic = pService->createCharacteristic("331F1571-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
```

Will solve your problem.

peetjh, 2022-01-25:
Thanks, that fixed it.

h2zero, 2022-04-04:
Closing as resolved.

closed 2022-04-04 by h2zero