h2zero/NimBLE-Arduino #323: Broken characteristic state in `onReadCB`

TheRealMazur, 2021-12-27:
Hi, first of all, big thanks for maintaining the project! 
I'm running into a problem with [my project](https://github.com/TheRealMazur/LinakDeskEsp32Controller) after upgrading some dependencies, including moving NimBLE version from 1.2 to 1.3.
It seems to me that  the RemoteService is in a wrong state (client being `null`) [here](https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLERemoteCharacteristic.cpp#L537):
`uint16_t conn_id = characteristic->getRemoteService()->getClient()->getConnId();`

I'm not sure how to investigate what is the reason of the error - the logic in my project hasn't changed, just the dependencies.
It happens when I'm [subscribing](https://github.com/TheRealMazur/LinakDeskEsp32Controller/blob/98cd7d58e6e4767a01f3bac45bfc649761ecb7f1/lib/LinakDeskCore/DeskController.cpp#L89) to a characteristic right after [reading its value](https://github.com/TheRealMazur/LinakDeskEsp32Controller/blob/98cd7d58e6e4767a01f3bac45bfc649761ecb7f1/lib/LinakDeskCore/DeskController.cpp#L81).

Could you give me some pointers on what to try next?

Here are the logs right before the crash and the stacktrace:
```
HTTP server started
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0035 len=5
0x35 0x0c 0x05 0x01 0x00 0x00 0x01 0x00 
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
GATT procedure initiated: read long; att_handle=26
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x0a 0x1a 0x00 
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x001a, props:  0x12, 01"
D NimBLERemoteCharacteristic: ">> getDescriptor: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: 99fa0021-338a-1024-8a49-009c0215f78a"
GATT procedure initiated: discover all characteristics; start_handle=26 end_handle=31
host tx hci data; handle=0 length=11
ble_hs_hci_acl_tx(): 0x00 0x00 0x0b 0x00 0x07 0x00 0x04 0x00 0x08 0x1a 0x00 0x1f 0x00 0x03 0x28 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x0b 0x51 0x04 0x00 0x00 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401b60db  PS      : 0x00060e30  A0      : 0x80186fb8  A1      : 0x3ffdbd00  
A2      : 0x00000000  A3      : 0x00000009  A4      : 0x00000009  A5      : 0x00060023  
A6      : 0x007daf30  A7      : 0x003fffff  A8      : 0x8018fa51  A9      : 0x3ffdbd00  
A10     : 0x00000002  A11     : 0x00000000  A12     : 0x00060e20  A13     : 0x3ffc80e4  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40189264  LEND    : 0x40189270  LCOUNT  : 0x00000010  

Backtrace:0x401b60d8:0x3ffdbd000x40186fb5:0x3ffdbd20 0x4018cf8b:0x3ffdbd60 0x4018db6d:0x3ffdbd80 0x4018e73c:0x3ffdbdb0 0x40189778:0x3ffdbde0 0x40189294:0x3ffdbe00 0x401910d6:0x3ffdbe30 0x4018fa64:0x3ffdbe60 0x4018fa6f:0x3ffdbe80 0x40197372:0x3ffdbea0 0x40185f51:0x3ffdbec0 
  #0  0x401b60d8:0x3ffdbd00 in NimBLERemoteService::getClient() at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteService.cpp:254
  #1  0x40186fb5:0x3ffdbd20 in NimBLERemoteCharacteristic::onReadCB(unsigned short, ble_gatt_error const*, ble_gatt_attr*, void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:541
  #2  0x4018cf8b:0x3ffdbd60 in ble_gattc_read_long_cb at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_gattc.c:3086
  #3  0x4018db6d:0x3ffdbd80 in ble_gattc_read_long_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_gattc.c:3180
  #4  0x4018e73c:0x3ffdbdb0 in ble_gattc_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_gattc.c:4612
  #5  0x40189778:0x3ffdbde0 in ble_att_clt_rx_read at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_att_clt.c:478
  #6  0x40189294:0x3ffdbe00 in ble_att_rx at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_att.c:503
  #7  0x401910d6:0x3ffdbe30 in ble_hs_hci_evt_acl_process at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:929
  #8  0x4018fa64:0x3ffdbe60 in ble_hs_process_rx_data_queue at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_hs.c:238
  #9  0x4018fa6f:0x3ffdbe80 in ble_hs_event_rx_data at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_hs.c:532
  #10 0x40197372:0x3ffdbea0 in ble_npl_event_run at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble_npl_os.h:121
      (inlined by) nimble_port_run at .pio/libdeps/esp32dev/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c:81
  #11 0x40185f51:0x3ffdbec0 in NimBLEDevice::host_task(void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEDevice.cpp:728

ELF file SHA256: 0000000000000000

Rebooting...
```

h2zero, 2021-12-27:
Thank you for providing those details. This is quite curious as to how it could happen. The backtrace certainly is saying that the client pointer is invalid. 

I do not know how that is possible unless you have deleted the client or something has corrupted the remote service data.

Does your code delete the client instance at any time? 

Is this a random occurrence or does it happen every time?

TheRealMazur, 2021-12-28:
Thanks for the quick response!
As mentioned, I haven't changed the logic of my code, just the dependencies. I do not delete the client - I have a deleter provided in the unique ptr, but I experimented with adding logging to it, and as expected, didn't saw it being called.
To answer the rest of your questions: this is not random, it happens every time my code reaches the places mentioned above (It happens when I'm [subscribing](https://github.com/TheRealMazur/LinakDeskEsp32Controller/blob/98cd7d58e6e4767a01f3bac45bfc649761ecb7f1/lib/LinakDeskCore/DeskController.cpp#L89) to a characteristic right after [reading its value](https://github.com/TheRealMazur/LinakDeskEsp32Controller/blob/98cd7d58e6e4767a01f3bac45bfc649761ecb7f1/lib/LinakDeskCore/DeskController.cpp#L81).)
I've tested with the current master branch with the same results:
```
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x001a, props:  0x12, 01"
D NimBLERemoteCharacteristic: ">> getDescriptor: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: 99fa0021-338a-1024-8a49-009c0215f78a"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40186b59  PS      : 0x00060e30  A0      : 0x8018c836  A1      : 0x3ffdc070  
A2      : 0x00000001  A3      : 0x3ffc7d64  A4      : 0x3ffdc0d8  A5      : 0x3ffcf02c  
A6      : 0x3ffc4108  A7      : 0x00000000  A8      : 0x3ffc7d64  A9      : 0x3ffdc070  
A10     : 0x00000001  A11     : 0x00000009  A12     : 0x00000009  A13     : 0x00060023  
A14     : 0x007b605c  A15     : 0x003fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x40188cf4  LEND    : 0x40188d00  LCOUNT  : 0x00000010  

Backtrace:0x40186b56:0x3ffdc0700x4018c833:0x3ffdc0b0 0x4018d419:0x3ffdc0d0 0x4018df0c:0x3ffdc100 0x40189208:0x3ffdc130 0x40188d24:0x3ffdc150 0x401907d2:0x3ffdc180 0x4018f218:0x3ffdc1b0 0x4018f223:0x3ffdc1d0 0x40196536:0x3ffdc1f0 0x40185be4:0x3ffdc210 
  #0  0x40186b56:0x3ffdc070 in NimBLERemoteCharacteristic::onReadCB(unsigned short, ble_gatt_error const*, ble_gatt_attr*, void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:534
  #1  0x4018c833:0x3ffdc0b0 in ble_gattc_read_long_cb at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3086
  #2  0x4018d419:0x3ffdc0d0 in ble_gattc_read_long_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3180
[...]
```

I'm starting to wonder if it could be due to the `arg` being wrong, and therefore the `pTaskData`. Just for experimentation, I've commented out [checking the conn_id](https://github.com/h2zero/NimBLE-Arduino/blob/master/src/NimBLERemoteCharacteristic.cpp#L537-L541) and it crashed again in the same moment, due to `LoadProhibited` in L561: `xTaskNotifyGive(pTaskData->task);`

h2zero, 2021-12-28:
Thanks for answering. Something strange is happening and I'm looking into it. Your logs are showing that the `readvalue()` call was completed but the stack is still calling the callback for some reason. When that occurs the pointer here https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/src/NimBLERemoteCharacteristic.cpp#L535 is invalid and causes the crash.

If you don't mind, could you move this log https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/src/NimBLERemoteCharacteristic.cpp#L543 to line 535, before the pointer assignment? I'd like to see what the status code is when this happens.

TheRealMazur, 2021-12-28:
Here are the logs with the moved log macro (I've added some logs from the setup that show a successful read of the characteristic (the crash happens during a user command handling, after a successful setup):
```
connected
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
Current height: 735 mm
Getting height and printing it took: 117ms
After waiting 0.00 secs , WiFi connection result is 3
Local IP: 192.168.0.122
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40186b68  PS      : 0x00060630  A0      : 0x8018c832  A1      : 0x3ffdbcb0  
A2      : 0x00000000  A3      : 0x3ffc7d64  A4      : 0x3ffdbd18  A5      : 0x3ffcf02c  
A6      : 0x3ffc4108  A7      : 0x3f41c94b  A8      : 0x00000000  A9      : 0x3ffdbc60  
A10     : 0x00000046  A11     : 0x3ffdbf50  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x007daec8  A15     : 0x003fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x40091cd9  LEND    : 0x40091ce9  LCOUNT  : 0xfffffff9  

Backtrace:0x40186b65:0x3ffdbcb00x4018c82f:0x3ffdbcf0 0x4018d415:0x3ffdbd10 0x4018df08:0x3ffdbd40 0x40189204:0x3ffdbd70 0x40188d20:0x3ffdbd90 0x401907ce:0x3ffdbdc0 0x4018f214:0x3ffdbdf0 0x4018f21f:0x3ffdbe10 0x40196532:0x3ffdbe30 0x40185be0:0x3ffdbe50 
  #0  0x40186b65:0x3ffdbcb0 in NimBLERemoteCharacteristic::onReadCB(unsigned short, ble_gatt_error const*, ble_gatt_attr*, void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:535
  #1  0x4018c82f:0x3ffdbcf0 in ble_gattc_read_long_cb at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3086
  #2  0x4018d415:0x3ffdbd10 in ble_gattc_read_long_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3180
  #3  0x4018df08:0x3ffdbd40 in ble_gattc_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:4612
```
As you said, we seem to be exiting `readValue` too soon. Let me know if you think of any log lines that might be useful - it's a quick repro for me. Thanks for looking into it!

h2zero, 2021-12-28:
Thanks!, One more thing if you don't mind. Could you try the log move with 1.2? I know as you said it won't crash but if you could catch the log at the same point in the code it would give me the data to come up with a fix. The status here is `0` when I would expect it to be `14`, something in the core might have changed.

h2zero, 2021-12-28:
Did some more investigating. seems the cause is possibly in the `ulNotifiyTake()` call. Your project is using arduino core 2.0.0, which contains a new FreeRTOS version. This has been problematic already, #311. So it does not surprise me that there are other errors with this, I will need to look into it more.

TheRealMazur, 2021-12-28:
Thank you.
I was just in the middle of writing a comment saying that recreating the last working setup using 1.2 will be difficult for me. I was using some branches of https://github.com/platformio/platform-espressif32 and https://github.com/espressif/arduino-esp32 that got deleted...
I guess that's the cost of trying to be on the cutting edge 😁 
Let me know if you want me to test something on the 2.0 core :)

h2zero, 2021-12-28:
Ok, so I've been trying to reproduce this using arduino and idf with no luck so far. One thing that my be helpful if you could try it is changing https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/src/NimBLERemoteCharacteristic.cpp#L488

to 
```
uint32_t ret = ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
ESP_LOGE(LOG_TAG, "ret = %u", ret);
```

So we can see what the return value is here.

h2zero, 2021-12-28:
Looked through your code a bit more for some ideas to reproduce and noticed you are using asyncWebServer. Were you able to stabilize this in the past? Really surprising if so, BLE+async are a known problematic combination and my somehow be the cause judging by your logs:
```
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

TheRealMazur, 2021-12-28:
Re:
> Ok, so I've been trying to reproduce this using arduino and idf with no luck so far. One thing that my be helpful if you could try it is changing [...]

log:
```
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
GATT procedure initiated: read long; att_handle=26
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x0a 0x1a 0x00 
E (12347) NimBLERemoteCharacteristic: ret = 1
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
```

TheRealMazur, 2021-12-28:
Re:
> Looked through your code a bit more for some ideas to reproduce and noticed you are using asyncWebServer. Were you able to stabilize this in the past? Really surprising if so, BLE+async are a known problematic combination and may somehow be the cause judging by your logs:
> 
> ```
> HTTP server started
> D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
> D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
> I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> ```

It worked okay-ish 😁 I don't mind if the HTTP response times out, as long as it executes the command and moves my desk 😅 
I agree that it's not the most stable thing, but with the previous dependencies it worked quite reliably. I'm actually in the process of creating some forks of the dependencies to restore the previous functionality, so I should be able to provide those 1.2 logs in a while after all. 🙂

TheRealMazur, 2021-12-28:
Re:
> Thanks!, One more thing if you don't mind. Could you try the log move with 1.2? I know as you said it won't crash but if you could catch the log at the same point in the code it would give me the [...]

Old dependencies (NimBLE 1.2) logs:
```
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
GATT procedure initiated: read long; att_handle=26
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x0a 0x1a 0x00 
txed att command: read req; conn=0 handle=0x001a
Number of Completed Packets: num_handles=1
handle:0 pkts:1
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x0b 0x15 0x13 0x00 0x00 
rxed att command: read rsp; conn=0 
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
[...] (other successful reads and writes follow)
```

h2zero, 2021-12-28:
Thank you! So far I still cannot reproduce this, the return code is correct but the task is unblocked somehow before the callback is called. I have no idea how that could happen unless there is something elsewhere that releases the task...?

I can't really explain this, I don't think anything has changed in this regard between 1.2 and 1.3.

TheRealMazur, 2021-12-28:
Oh, just to clarify - those logs come from the whole old dependencies setup, which also doesn't use Arduino 2.0... I cannot simply return to NimBLE 1.2, because as I understood it " Some recent updates were made to work with the arduino core 2.0.0" ([link](https://github.com/h2zero/NimBLE-Arduino/issues/247#issuecomment-846575068))

h2zero, 2021-12-28:
You could use this commit 46deed9d6088673eab69e866ee6a604b443844b2 with 2.0, that is what the comment referred to.

Would be interesting if it worked.

TheRealMazur, 2021-12-28:
NimBLE @ https://github.com/h2zero/NimBLE-Arduino/commit/46deed9d6088673eab69e866ee6a604b443844b2 with the new dependency stack is still giving me:
```
connected
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
GATT procedure initiated: read long; att_handle=26
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x0a 0x1a 0x00 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x0b 0xa1 0x12 0x00 0x00 
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
Current height: 1101 mm
Getting height and printing it took: 126ms
After waiting 0.00 secs , WiFi connection result is 3
Local IP: 192.168.0.122
HTTP server started
ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0035 len=5
0x35 0x0c 0x05 0x01 0x00 0x00 0x01 0x00 
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
GATT procedure initiated: read long; att_handle=26
host tx hci data; handle=0 length=7
ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x0a 0x1a 0x00 
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x001a, props:  0x12, 01"
D NimBLERemoteCharacteristic: ">> getDescriptor: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: 99fa0021-338a-1024-8a49-009c0215f78a"
GATT procedure initiated: discover all descriptors; chr_val_handle=26 end_handle=31
host tx hci data; handle=0 length=9
ble_hs_hci_acl_tx(): 0x00 0x00 0x09 0x00 0x05 0x00 0x04 0x00 0x04 0x1b 0x00 0x1f 0x00 
ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x0b 0xa1 0x12 0x00 0x00 
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401b6063  PS      : 0x00060030  A0      : 0x80186f80  A1      : 0x3ffdbcf0  
A2      : 0x00000000  A3      : 0x3f406d5f  A4      : 0x3f41ce03  A5      : 0x3ffdbcc0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80099b98  A9      : 0x3ffdbc90  
A10     : 0x00000001  A11     : 0x3f406d5f  A12     : 0x3f41ce03  A13     : 0x3ffdbcf0  
A14     : 0x3ffdbcc0  A15     : 0x0000000c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40091cd9  LEND    : 0x40091ce9  LCOUNT  : 0xfffffff9  

Backtrace:0x401b6060:0x3ffdbcf00x40186f7d:0x3ffdbd10 0x4018cf33:0x3ffdbd50 0x4018db15:0x3ffdbd70 0x4018e6e4:0x3ffdbda0 0x40189720:0x3ffdbdd0 0x4018923c:0x3ffdbdf0 0x4019107e:0x3ffdbe20 0x4018fa0c:0x3ffdbe50 0x4018fa17:0x3ffdbe70 0x401972f2:0x3ffdbe90 0x40185fc5:0x3ffdbeb0 
  #0  0x401b6060:0x3ffdbcf0 in NimBLERemoteService::getClient() at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteService.cpp:240
  #1  0x40186f7d:0x3ffdbd10 in NimBLERemoteCharacteristic::onReadCB(unsigned short, ble_gatt_error const*, ble_gatt_attr*, void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:494
  #2  0x4018cf33:0x3ffdbd50 in ble_gattc_read_long_cb at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/host/src/ble_gattc.c:3086
[...]
```

h2zero, 2021-12-28:
Damn, sorry for this. What I am going to do is create a debug branch with extra logging for you to try. I'll add extra logs around all the the task notification stuff so we can hopefully find what is causing it.

TheRealMazur, 2021-12-28:
Sounds great! Thanks again, I really appreciate the amount of effort you are putting into this repo! 🙂

h2zero, 2021-12-28:
Thanks, no problem! I don't tolerate bugs well :)

Here is the branch https://github.com/h2zero/NimBLE-Arduino/tree/debug-tasks
I look forward to seeing the logs.

TheRealMazur, 2021-12-28:
Here you go - again, happy read during the end of setup, and crash after HTTP call:
```
connected
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "readValue WAIT"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "onReadCB GIVE 14"
D NimBLERemoteCharacteristic: "readValue CONTINUE"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
Current height: 1101 mm
Getting height and printing it took: 117ms
After waiting 0.00 secs , WiFi connection result is 3
Local IP: 192.168.0.122
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "readValue WAIT"
D NimBLERemoteCharacteristic: "readValue CONTINUE"
D NimBLERemoteCharacteristic: "<< readValue length: 0 rc=0"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x001a, props:  0x12, 01"
D NimBLERemoteCharacteristic: ">> getDescriptor: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: 99fa0021-338a-1024-8a49-009c0215f78a"
D NimBLERemoteCharacteristic: "getEndHandle WAIT"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40186ce8  PS      : 0x00060030  A0      : 0x8018ca32  A1      : 0x3ffdbc60  
A2      : 0x00000000  A3      : 0x3ffc7d64  A4      : 0x3ffdbcc8  A5      : 0x3ffcf02c  
A6      : 0x3ffc4108  A7      : 0x3f41ca63  A8      : 0x00000001  A9      : 0x3ffdbc10  
A10     : 0x00000046  A11     : 0x3ffdbef8  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x007dae70  A15     : 0x003fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x40091cd9  LEND    : 0x40091ce9  LCOUNT  : 0xfffffff9  

Backtrace:0x40186ce5:0x3ffdbc600x4018ca2f:0x3ffdbca0 0x4018d615:0x3ffdbcc0 0x4018e108:0x3ffdbcf0 0x40189404:0x3ffdbd20 0x40188f20:0x3ffdbd40 0x401909ce:0x3ffdbd70 0x4018f414:0x3ffdbda0 0x4018f41f:0x3ffdbdc0 0x40196732:0x3ffdbde0 0x40185cc0:0x3ffdbe00 
  #0  0x40186ce5:0x3ffdbc60 in NimBLERemoteCharacteristic::onReadCB(unsigned short, ble_gatt_error const*, ble_gatt_attr*, void*) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLERemoteCharacteristic.cpp:543
  #1  0x4018ca2f:0x3ffdbca0 in ble_gattc_read_long_cb at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3086
  #2  0x4018d615:0x3ffdbcc0 in ble_gattc_read_long_rx_read_rsp at .pio/libdeps/esp32dev/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:3180
```

h2zero, 2021-12-28:
Well.... that's no good either 😢. Nothing unblocked the task and yet it didn't block at all, I'm confused now lol. I have never seen this behavior before, I'll need to think about this more.

TheRealMazur, 2021-12-28:
No worries, I've found a way to build my project again with the old dependencies, so I will move mainline to that and treat the Arduino 2.0/NimBLE 1.3 as a feature branch. 
Take your time and let me know if you want me to test something again 🙂

h2zero, 2021-12-28:
Perhaps a clue here https://github.com/espressif/arduino-esp32/issues/2445

TheRealMazur, 2021-12-28:
Hey, a small update:
as I now very little about freertos tasks running underneath Arduino core on ESP32, I've decided to learn more about it.
In the implementation of `ulTaskNotifyTake` I've found this comment:
https://github.com/espressif/esp-idf/blob/5624dffc52271389376352974ba5911963ee207b/components/freertos/FreeRTOS-Kernel/tasks.c#L5360
Which lead me to the following experiment:
I've added 
```
ulTaskNotifyValueClear(xTaskGetCurrentTaskHandle(), UINT_MAX);
```
right before 
https://github.com/h2zero/NimBLE-Arduino/blob/4a6111d9639053caddc398f67b3dd455b26eb3aa/src/NimBLERemoteCharacteristic.cpp#L488

and it didn't crash!
here are the logs including the ones from debug branch:
```
connected
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "readValue WAIT"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "onReadCB GIVE 14"
D NimBLERemoteCharacteristic: "readValue CONTINUE"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
Current height: 1101 mm
Getting height and printing it took: 216ms
After waiting 0.00 secs , WiFi connection result is 3
Local IP: 192.168.0.122
HTTP server started
D NimBLERemoteCharacteristic: ">> readValue(): uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x1a"
D NimBLERemoteCharacteristic: "readValue WAIT"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "Got 4 bytes"
I NimBLERemoteCharacteristic: "Read complete; status=14 conn_handle=0"
D NimBLERemoteCharacteristic: "onReadCB GIVE 14"
D NimBLERemoteCharacteristic: "readValue CONTINUE"
D NimBLERemoteCharacteristic: "<< readValue length: 4 rc=0"
D NimBLERemoteCharacteristic: ">> setNotify(): Characteristic: uuid: 99fa0021-338a-1024-8a49-009c0215f78a, handle: 26 0x001a, props:  0x12, 01"
D NimBLERemoteCharacteristic: ">> getDescriptor: uuid: 0x2902"
D NimBLERemoteCharacteristic: ">> retrieveDescriptors() for characteristic: 99fa0021-338a-1024-8a49-009c0215f78a"
D NimBLERemoteCharacteristic: "getEndHandle WAIT"
D NimBLERemoteCharacteristic: "Next Characteristic >> status: 0 handle: 29"
D NimBLERemoteCharacteristic: "nextCharCB GIVE 14"
D NimBLERemoteCharacteristic: "getEndHandle CONTINUE"
D NimBLERemoteCharacteristic: "retrieveDescriptors WAIT"
D NimBLERemoteCharacteristic: "Descriptor Discovered >> status: 0 handle: 27"
D NimBLERemoteDescriptor: ">> NimBLERemoteDescriptor()"
[...]
```

It seems that the previous read somehow didn't clean the state.

h2zero, 2021-12-29:
Thanks for that! 

That's very interesting, the implementation used is a binary one in which the task notification value starts as 0 (taken) and gets incremented to 1 when the "give" is processed. 

The task value is then decremented after the task is released to run and should be 0 at that point, this would ensure a block on the next call to "take".

So the question becomes, why does this not happen? Is something else "giving" the value or is there a bug that is not decrementing it? Since I was unable to reproduce this I'm leaning towards the former. If so, now we have to ask where and why is this being set to 1?

Great work digging into this, we'll sort this out.

h2zero, 2022-01-08:
@TheRealMazur could you try the bugfix-task-notify branch in #326 and let me know if that works for you?

TheRealMazur, 2022-01-08:
Hey, yes, changing my dependency to the `bugfix-task-notify` fixes the issue 😁 

Let me know (or just close this issue so that I get pinged) when the fix gets to mainline, so I can switch my project to use it 😄
Thanks again for the good work!

h2zero, 2022-01-08:
Great, thanks for the help nailing it down. I still haven't found the root cause but this works for now. Luckily the problem doesn't exist in arduino core 1.0.6 as it does not have the `ulTaskNotifyValueClear` function.

h2zero, 2022-01-10:
Amazingly I was able to reproduce this in IDF, which allowed for some debugging. A few quick asserts lead me to this line: https://github.com/espressif/esp-idf/blob/5c33570524118873f7bd32490c7a0442fede4bf8/components/esp_timer/src/esp_timer.c#L413

Seems the new esp_timer library doesn't play nice with task notifications or there is something going on with the task pointer. I will merge the PR for now as it also resolves #324 but I may need to look at reworking this in the future.

closed 2022-01-10 by h2zero