h2zero/NimBLE-Arduino #327: writeValue response

tonton81, 2022-01-10:
I am working on an SP608E device, where writeValue doesn't work unless it is set to write response, which is fine. I tried looking for documentation for the response, if we get any, but google hasn't been of much help. Is there really response data or is that just a communication flag?

h2zero, 2022-01-10:
Yes there is a response, it is handled internally. The difference is the call to write will block until the response comes whereas the non-response write returns immediately. The response is just an acknowledgement code, no data.

tonton81, 2022-01-10:
is there a way to make it non blocking? 😁

h2zero, 2022-01-10:
>is there a way to make it non blocking?

Sure, but then you would have to handle waiting for the response before writing again and another callback would be required so that you would know when that happens. 

>or do we really need that response to just ignore it?

The purpose of it is to allow the server to respond with a error if necessary, so it's not an ignored response. That's why the function returns a bool value to indicate success/failure.