h2zero/NimBLE-Arduino #329: Task watchdog got triggered (ESP32)

joysfera, 2022-01-10:
With a very slightly modified basic `Nimble_scan_continuous.ino` example I'm getting watchdog invoked restarts on ESP32 within less than two hours since starting the program. They look as follows:

```
E (82202) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (82202) task_wdt:  - IDLE (CPU 0)
E (82202) task_wdt: Tasks currently running:
E (82202) task_wdt: CPU 0: ble
E (82202) task_wdt: CPU 1: loopTask
E (82202) task_wdt: Aborting.

abort() was called at PC 0x400eda4d on core 0

Backtrace:0x4008372d:0x3ffbe8dc |<-CORRUPTED
```

The code looks as follows. The scan() function is called every second, just in case the scan stopped, but so far the scan runs just fine for 30, 60 or even 90 minutes (it's random) but then the watchdog steps in and reboots the device.
```
void setup()
{
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true );
  pBLEScan->setInterval(100);    // How often the scan occurs / switches channels; in milliseconds (default 100)
  pBLEScan->setWindow(30);       // How long to scan during the interval; in milliseconds.
  pBLEScan->setActiveScan(true); // Set active scanning, this will get more data from the advertiser.
  pBLEScan->setMaxResults(0);    // do not store the scan results, use callback only
}

void scan()
{
    if (!pBLEScan->isScanning()) {  // Check if scan already in progress
        pBLEScan->clearResults();
        if (pBLEScan->start(0, nullptr, false)) {
            Serial.println("Scan started.");
        }
        else {
            Serial.println("Scan failed to start.");
        }
    }
}
```

BTW, when I tried to change the scan time (first parameter of `pBLEScan->start()`) from 0 to say 60 the watchdog restarted the device within 5-10 minutes.

h2zero, 2022-01-11:
Hi @joysfera, can you share some details? What version/branch of this repo are you using? Arduino core version? I've been trying to reproduce this using the master branch without success on the latest arduino core.

joysfera, 2022-01-11:
Latest released versions, i.e. Arduino core 2.0.2 and NimBLE-Arduino 1.3.3.

joysfera, 2022-01-11:
Hardware is [LilyGO TTGO T-Display](https://www.aliexpress.com/item/33048962331.html) (ESP32 with a LCD display).

h2zero, 2022-01-11:
Thanks, I'll keep trying to reproduce it, hopefully I can catch it at some point. Can you share a log with debug enabled?

joysfera, 2022-01-11:
Inside of `onResult` I display the found data on the LCD. Wondering if there could be some lockup when using SPI from within `onResult()`. Any idea?

h2zero, 2022-01-11:
That is quite likely the problem, I would suggest using a queue or some other mechanism to save off the data and update the display in a separate thread (or in loop).

joysfera, 2022-01-11:
> Can you share a log with debug enabled?

I have just started a build with Core Debug level set to "Debug" (instead of "None"). Now I need to wait for 100 minutes (several last crashes occured almost exactly after 100 minutes of run).

> The SPI is quite likely the problem, I would suggest using a queue or some other mechanism to save off the data and update the display in a separate thread (or in loop).

Will do after this build crashes, just in case I could provide some useful debug log.

After testing with a queue will post an update here. Thanks for now!

h2zero, 2022-01-11:
Yep, looks like the crash happens just as the callback is invoked. Quite likely the problem.

joysfera, 2022-01-11:
After removing the display calls from `onResult()` it's been running for 200+ minutes straight, no stability issues whatsoever.
Seems like we've nailed it and this issue can be closed.

The takeaway is to better not call any other functions from `onResult` on ESP32 as it is a callback from another core and using say SPI can result in a lock-up (that watchdog will take care of as could be seen from the logs I posted above).

h2zero, 2022-01-11:
Yeah, anything that takes time or uses the same resources should not be done in the callback for sure. Glad that you found the cause!

closed 2022-01-11 by h2zero