h2zero/NimBLE-Arduino #331: Platform IO error - Warning! Could not find the package with 'h2zero/NimBLE-Arduino @ 1.3.4' requirements for your system 'darwin_x86_64'

mc-hamster, 2022-01-11:
Hi,

I get the following error:

Library Manager: Installing h2zero/NimBLE-Arduino @ 1.3.4
Warning! Could not find the package with 'h2zero/NimBLE-Arduino @ 1.3.4' requirements for your system 'darwin_x86_64'

[image]

I have a m1 Mac running macOS 12.0.1 with PlatformIO 5.2.4

thorrak, 2022-01-11:
Hey there -- I think this is the issue I was encountering which prompted me to open PR #330 as if you look at [Platformio's library page](https://registry.platformio.org/libraries/h2zero/NimBLE-Arduino/installation) for NimBLE-Arduino it shows v1.3.1 which matched the version in library.properties. No idea if that's actually the fix, though, as I'm not familiar with how platformio's library finder triggers updates.

mc-hamster, 2022-01-11:
I’ll pull that into my fork and try it out. Thanks!

thorrak, 2022-01-11:
> I’ll pull that into my fork and try it out. Thanks!

I don't think this will have any impact on your fork, unfortunately, as from taking a quick glance at the platformio docs it seems the change only impacts when @h2zero next republishes the library using the [pio package publish](https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html) command. (Even then though, that's based on my rudimentary understanding which comes with exactly 0 seconds of experience, in this case!)

If you need v1.3.4, for now you can get it by replacing `h2zero/NimBLE-Arduino @ 1.3.4` with `https://github.com/h2zero/NimBLE-Arduino.git#1.3.4` in your `platformio.ini` file.

h2zero, 2022-01-11:
I've tried to look into this but I don't understand how the platformio registry discovers the releases. I've never had to publish manually before, it's just been picked up automatically. I think something has changed in their library registry, possibly it only reads from the default branch now?

I see arduino has picked it up correctly https://www.arduino.cc/reference/en/libraries/nimble-arduino/

h2zero, 2022-01-11:
@thorrak I'm reluctant to merge your PR until I have a better understanding of the process the pio registry uses, I don't want it pulling in the master branch or some other unexpected outcome.

On the plus side, it looks like this repo is in the top 100 now, so that's cool 😄

thorrak, 2022-01-11:
> @thorrak I'm reluctant to merge your PR until I have a better understanding of the process the pio registry uses, I don't want it pulling in the master branch or some other unexpected outcome.
> 
> On the plus side, it looks like this repo is in the top 100 now, so that's cool 😄

Considering my PR is based on a wild guess as to how Platformio operates, I would be too. :) 

Congrats on the popularity - it’s well deserved! I have at least two projects that would not exist if this hadn’t been released. (The default libraries were far too big for 4MB of flash!)

h2zero, 2022-01-11:
Maybe @ivankravets could offer a clue as to how to proceed?

ivankravets, 2022-01-11:
Hi all,

Thanks for pinging me. Glad to help. 

> s but I don't understand how the platformio registry discovers the releases. I've never had to publish manually before, it's just been picked up automatically. 

Yes, the previous version of PlatformIO Registry had a crawler that monitored the library's manifest every 24 hours for the version changes. That was a trigger to auto-publish a new version if manifest's `version` field matches with the release tag.

Currently, this crawler works in legacy mode and was disabled in 2020 for the new libraries. So, new libraries are published by their owners and they maintain updates.

In the case of this library, the legacy crawler still monitors its manifest https://github.com/h2zero/NimBLE-Arduino/blob/master/library.properties from the default branch. The latest version is `version=1.3.1` which corresponds to https://registry.platformio.org/libraries/h2zero/NimBLE-Arduino

-------

There are two options to resolve this issue:

a) Easiest, just update version in https://github.com/h2zero/NimBLE-Arduino/blob/master/library.properties and legacy crawler will pick a new version
b) as @thorrak mentioned, you can publish the version manually using [pio package publish](https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html) command. Please note, once you published a new version manually, the legacy crawler will be disabled for this library. You later will need to publish new versions each time.

P.S: Thanks for the great library! 🙏

h2zero, 2022-01-11:
Awesome, thanks for the quick response @ivankravets ! Nice to have clarity on the issue, looks like @thorrak 's PR will do the trick then 😄

h2zero, 2022-01-12:
This appears to be fixed now, thanks everyone!

closed 2022-01-12 by h2zero