h2zero/NimBLE-Arduino #332: Memory leak in "Init-Deinit" cycle

VNovytskyi, 2022-01-12:
Hello!
I write an application which **init** and **deinit** BLE peripheral device. When I started testing the memory usage I found the memory leak in my sketch.  I started deleting code from the sketch to find the source of the problem. I ended up with the following code:
```c
#include <NimBLEDevice.h>

#define BLE_LOOP_TEST_COUNT       100
#define BLE_LOOP_TEST_DELAY       10
#define FREE_HEAP_PRINT_INTERVAL  1000

void setup() {
  Serial.begin(115200);

  for (uint32_t i = 0; i < BLE_LOOP_TEST_COUNT; ++i) {
    NimBLEDevice::init("NimBLE-Arduino");
    Serial.print("\nBLE init finished: ");
    Serial.println(ESP.getFreeHeap());

    vTaskDelay(BLE_LOOP_TEST_DELAY);

    NimBLEDevice::deinit(true);
    Serial.print("BLE deinit finish: ");
    Serial.println(ESP.getFreeHeap());
  }
}

void loop() {
  static uint32_t prev_print_heap_time = 0;
  if (millis() - prev_print_heap_time > FREE_HEAP_PRINT_INTERVAL) {
    prev_print_heap_time = millis();
    Serial.print("Free heap: ");
    Serial.println(ESP.getFreeHeap());
  }
}
```

Results of executing (Serial monitor):
14:00:45.534 -> BLE init finished: 232408
14:00:45.534 -> BLE deinit finish: 258988
14:00:45.721 -> 
14:00:45.721 -> BLE init finished: 232304
14:00:45.768 -> BLE deinit finish: 258884
14:00:45.957 -> 
14:00:45.957 -> BLE init finished: 232200
14:00:45.957 -> BLE deinit finish: 258780
14:00:46.144 -> 
14:00:46.144 -> BLE init finished: 232096
14:00:46.144 -> BLE deinit finish: 258676
14:00:46.331 -> 
14:00:46.331 -> BLE init finished: 231992
14:00:46.331 -> BLE deinit finish: 258572
14:00:46.518 -> 
14:00:46.518 -> BLE init finished: 231888
14:00:46.518 -> BLE deinit finish: 258468
14:00:46.706 -> 
14:00:46.706 -> BLE init finished: 231776
14:00:46.753 -> BLE deinit finish: 262828
14:00:46.941 -> 
14:00:46.941 -> BLE init finished: 231672
14:00:46.941 -> BLE deinit finish: 262716
14:00:47.128 -> 
14:00:47.128 -> BLE init finished: 231568
14:00:47.128 -> BLE deinit finish: 258148
14:00:47.315 -> 
14:00:47.315 -> BLE init finished: 231464
14:00:47.315 -> BLE deinit finish: 258044
14:00:47.501 -> 
14:00:47.501 -> BLE init finished: 231360
14:00:47.501 -> BLE deinit finish: 257940
14:00:47.689 -> 
14:00:47.689 -> BLE init finished: 231256
14:00:47.737 -> BLE deinit finish: 257836
14:00:47.878 -> 
14:00:47.878 -> BLE init finished: 231152
14:00:47.925 -> BLE deinit finish: 257732
14:00:48.112 -> 
14:00:48.112 -> BLE init finished: 231048
14:00:48.112 -> BLE deinit finish: 257628
14:00:48.301 -> 
14:00:48.301 -> BLE init finished: 230944
14:00:48.301 -> BLE deinit finish: 257524
14:00:48.490 -> 
14:00:48.490 -> BLE init finished: 230840
14:00:48.490 -> BLE deinit finish: 257420
14:00:48.676 -> 
14:00:48.676 -> BLE init finished: 230736
14:00:48.676 -> BLE deinit finish: 261780
14:00:48.865 -> 
14:00:48.865 -> BLE init finished: 230632
14:00:48.912 -> BLE deinit finish: 257212
14:00:49.052 -> 
14:00:49.052 -> BLE init finished: 230528
14:00:49.099 -> BLE deinit finish: 257108
14:00:49.289 -> 
14:00:49.289 -> BLE init finished: 230424
14:00:49.289 -> BLE deinit finish: 257004
14:00:49.479 -> 
14:00:49.479 -> BLE init finished: 230320
14:00:49.479 -> BLE deinit finish: 261364
14:00:49.667 -> 
14:00:49.667 -> BLE init finished: 230216
14:00:49.667 -> BLE deinit finish: 261260
14:00:49.853 -> 
14:00:49.853 -> BLE init finished: 230112
14:00:49.853 -> BLE deinit finish: 256692
14:00:50.041 -> 
14:00:50.041 -> BLE init finished: 230008
14:00:50.041 -> BLE deinit finish: 256588
14:00:50.226 -> 
14:00:50.226 -> BLE init finished: 229904
14:00:50.272 -> BLE deinit finish: 256484
14:00:50.460 -> 
14:00:50.460 -> BLE init finished: 229800
14:00:50.460 -> BLE deinit finish: 256380
14:00:50.648 -> 
14:00:50.648 -> BLE init finished: 229696
14:00:50.648 -> BLE deinit finish: 256276

Results of executing (Serial ploter):
[image]

I use ESP32 and the latest release **1.3.4**
[image]

h2zero, 2022-01-12:
Thanks for reporting and the test code. I have confirmed the issue in the latest release and seems to be resolved in the master branch. It's likely due to the timer initialization in the core code, which has be experimentally changed in the master branch.

VNovytskyi, 2022-01-12:
Thank you for your fast reply! I hope my debug information helps you. Are you fix this problem in the next release?

h2zero, 2022-01-12:
Yes, the next release will include this fix. Your debug information was great!

VNovytskyi, 2022-01-12:
Ok, thank you! Can I feel free to close this issue?

closed 2022-01-12 by VNovytskyi