h2zero/NimBLE-Arduino #333: Rebooting after init - deinit and conflict with ESPasyncwebserver

Martenz, 2022-01-12:
Hi I'm using a Lilygo T5 v2.4.1 with epaper 1.54.

I'm trying to combine in a background task a switch from wifi to BLE services because I'm not able to have both working at same time (any example of this working would very helpful if anyone managed to do it). If I have the BLE on the wifi does not notify clients and after a while the esp32 reboots.

To solve temporary this I put in my main freertos task a switch that does an Init and Deinit of bluetooth but when I enable the wifi (so Deinit BLE) and then disable wifi (do a Init BLE again) the esp32 reboots with the following error:

`Backtrace: 0x40091044:0x3ffe5b60 0x4009155f:0x3ffe5b80 0x400fc923:0x3ffe5ba0 0x400d646b:0x3ffe5be0 0x400d6831:0x3ffe5c20 0x400922d2:0x3ffe5c50
  #0  0x40091044:0x3ffe5b60 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4009155f:0x3ffe5b80 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:721
  #2  0x400fc923:0x3ffe5ba0 in NimBLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at lib/NimBLE-Arduino-master/src/NimBLEDevice.cpp:681
  #3  0x400d646b:0x3ffe5be0 in start_ble(String) at src/ble.h:113
  #4  0x400d6831:0x3ffe5c20 in taskBluetooth(void*) at src/main.cpp:541
  #5  0x400922d2:0x3ffe5c50 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)`

Most likely I'm doing something wrong, any hint? Or maybe there is still a bug ?

Thanks in advance for your time.

regards,

Martino

thorrak, 2022-01-13:
Do you have a link to your code? What happens when you have WiFi and BLE running simultaneously?

I have WiFi/BLE coexistence working on a project that I manage, though admittedly it did take a bit of [tweaking configuration options](https://github.com/thorrak/tiltbridge/blob/master/platformio.ini#L45) to get it stable.

I'm not sure if these still are required (or if they had any effect at all) but it might be worth a shot!

Martenz, 2022-01-13:
> tweaking configuration options

Thanks a lot! will give a try to your configurations, I will share the code and some more logs in next days.

h2zero, 2022-01-13:
AsyncWebServer + BLE is not easy to get stable for reasons unknown to me at this time. @thorrak I'm impressed you have seem to have cracked the code for this combo in your project. I keep threatening to investigate the instability but it's a large task for little time.

thorrak, 2022-01-13:
> AsyncWebServer + BLE is not easy to get stable for reasons unknown to me at this time. @thorrak I'm impressed you seem to have cracked the code for this combo in your project. I keep threatening to investigate the instability but it's a large task for little time.

Just don't ask me what that code _was_! I'm cargo-culting it at the moment for my other projects, and including all the task-pinning, flags, and working library versions to make it replicable. One day I may try to pin it down, but for now at least I'm not questioning a good thing!

h2zero, 2024-06-04:
Closing as I believe this is not related to this repo specifically.

closed 2024-06-04 by h2zero