h2zero/NimBLE-Arduino #338: Crashes in v1.3.5 when connecting/disconnecting/writing to/reading from BLE device

thorrak, 2022-01-19:
After upgrading from v1.3.4 to v1.3.5 in one of my projects, I'm periodically getting crashes that look like the following:

```
      3318 V: Connecting to device at 60:a4:23:f1:ed:72
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400932bc  PS      : 0x00060130  A0      : 0x80093396  A1      : 0x3ffcc4f0  
A2      : 0x01000000  A3      : 0x00000001  A4      : 0x00000d3f  A5      : 0x80000020  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000003  A9      : 0x3ffcc4b0  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffbd394  A13     : 0x00000000  
A14     : 0x3ffbd42c  A15     : 0x3ffbb170  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x01000014  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400932bc:0x3ffcc4f0 0x40093393:0x3ffcc530 0x40090b2a:0x3ffcc560
  #0  0x400932bc:0x3ffcc4f0 in prvProcessReceivedCommands at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #1  0x40093393:0x3ffcc530 in prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #2  0x40090b2a:0x3ffcc560 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```
Interestingly, the crashes always invoke something timer-related and only happen when connecting to/reading/writing to/disconnecting from a BLE device. Here's another crash when trying to send data to a device:

```
    615991 V: Connecting to device at 60:a4:23:f1:ed:72
    616315 V: Connected to: 60:a4:23:f1:ed:72, RSSI: -66
    616316 V: Sending command to unlock device (write_gatt_attr)
    617676 Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009377d  PS      : 0x00060f30  A0      : 0x800932c6  A1      : 0x3ffcc4d0  
A2      : 0x3ffdbc05  A3      : 0x00000000  A4      : 0x3ffbd394  A5      : 0x00000000  
A6      : 0x3ffbd3cc  A7      : 0x3ffbb170  A8      : 0x343ffdbc  A9      : 0x00003300  
A10     : 0x78000000  A11     : 0x00000000  A12     : 0x3ffcb64c  A13     : 0x00000001  
A14     : 0x00060f20  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00003308  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4009377d:0x3ffcc4d0 0x400932c3:0x3ffcc4f0 0x40093393:0x3ffcc530 0x40090b2a:0x3ffcc560
  #0  0x4009377d:0x3ffcc4d0 in uxListRemove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:214
  #1  0x400932c3:0x3ffcc4f0 in prvProcessReceivedCommands at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #2  0x40093393:0x3ffcc530 in prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #3  0x40090b2a:0x3ffcc560 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

And another:

```
      6104 V: Connected to: 60:a4:23:f1:ed:72, RSSI: -72
      7469 V: Read port 0, characteristic f1849d4b-a033-4d97-a988-3259a3498b95 as value: 18925
Preapring to disconnectGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400933cc  PS      : 0x00060933  A0      : 0x800ed6a1  A1      : 0x3ffce6e0  
A2      : 0x010114e9  A3      : 0x3ffc0fa4  A4      : 0x3ffce6f4  A5      : 0x00000001  
A6      : 0x00060b20  A7      : 0x00000000  A8      : 0x800933cc  A9      : 0x3ffce6b0  
A10     : 0x00000001  A11     : 0x3ffce8fc  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x010114fd  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  

ELF file SHA256: 0000000000000000

Backtrace: 0x400933cc:0x3ffce6e0 0x400ed69e:0x3ffce700 0x400d87dd:0x3ffce750 0x400d8cc5:0x3ffce770 0x400d8cfd:0x3ffce7d0 0x400d90be:0x3ffce800 0x400d978e:0x3ffce820 0x40104c2c:0x3ffce840 0x40090b2a:0x3ffce860
  #0  0x400933cc:0x3ffce6e0 in xTimerIsTimerActive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:957
  #1  0x400ed69e:0x3ffce700 in NimBLEClient::disconnect(unsigned char) at .pio/libdeps/m5stack-atom-lite/NimBLE-Arduino@src-21d326ddac6723274e648607cb71496d/src/NimBLEClient.cpp:471
      (inlined by) NimBLEClient::disconnect(unsigned char) at .pio/libdeps/m5stack-atom-lite/NimBLE-Arduino@src-21d326ddac6723274e648607cb71496d/src/NimBLEClient.cpp:357
  #2  0x400d87dd:0x3ffce750 in disconnectFromServer(NimBLEClient*) at src/ktGATT.cpp:410
  #3  0x400d8cc5:0x3ffce770 in read_gatt_attr(unsigned char, char const*, unsigned short&) at src/ktGATT.cpp:410
```

Rolling back to v1.3.4 results in the code becoming stable again.

h2zero, 2022-01-19:
@thorrak Thanks for reporting this, seems the FreeRTOS timer nightmare never ends. Master branch has an additional/different fix for this could you try that and let me know if it resolves the issue?

thorrak, 2022-01-19:
It's not crashing quite as quickly, but unfortunately it still has issues:

```
    162329 V: Connecting to device at 60:a4:23:f1:ed:72
ASSERT_ERR((((uint16_t)nextptr << 0) & ~((uint16_t)0x00007FFF)) == 0), in lld_pdu.c at line 85
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400871ec  PS      : 0x00060a34  A0      : 0x8004afde  A1      : 0x3ffc1450  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3ffbdbb4  
A6      : 0x3ffbdbb4  A7      : 0x00000000  A8      : 0x800871ec  A9      : 0x3ffc1430  
A10     : 0x0000005f  A11     : 0x0000005f  A12     : 0x00000010  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40087194  LEND    : 0x4008719b  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40009203  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400871ec

ELF file SHA256: 0000000000000000

Backtrace: 0x400871ec:0x3ffc1450 0x4004afdb:0x3ffc1470 0x4004b884:0x3ffc14a0 0x4008c1d7:0x3ffc14d0 0x4008a7f7:0x3ffc1510 0x400879b6:0x3ffc1550 0x4008ce7e:0x3ffc1570 0x4008d92b:0x3ffc1590 0x4008df55:0x3ffc15b0 0x401be173:0x3ffbcb50 0x40191cfb:0x3ffbcb70 0x400922d9:0x3ffbcb90 0x40090b2a:0x3ffbcbb0
  #0  0x400871ec:0x3ffc1450 in r_assert_err at ??:?
  #1  0x4004afdb:0x3ffc1470 in ?? ??:0
  #2  0x4004b884:0x3ffc14a0 in ?? ??:0
  #3  0x4008c1d7:0x3ffc14d0 in r_lld_pdu_tx_prog at ??:?
  #4  0x4008a7f7:0x3ffc1510 in r_lld_evt_schedule at ??:?
  #5  0x400879b6:0x3ffc1550 in r_ea_finetimer_isr at ??:?
  #6  0x4008ce7e:0x3ffc1570 in r_rwbt_isr at ??:?
  #7  0x4008d92b:0x3ffc1590 in r_rwbtdm_isr_wrapper at intc.c:?
  #8  0x4008df55:0x3ffc15b0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #9  0x401be173:0x3ffbcb50 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #10 0x40191cfb:0x3ffbcb70 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #11 0x400922d9:0x3ffbcb90 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #12 0x40090b2a:0x3ffbcbb0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Core 1 register dump:
PC      : 0x401be176  PS      : 0x00060b34  A0      : 0x80191cfe  A1      : 0x3ffbd0e0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80090d3c  A5      : 0x3ffbd000  
A6      : 0x00000000  A7      : 0x3ffbcde0  A8      : 0x00060023  A9      : 0x3ffbcc50  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80090d3c  A13     : 0x3ffe7b30  
A14     : 0x00000000  A15     : 0x3ffbcde0  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401be176:0x3ffbd0e0 0x40191cfb:0x3ffbd100 0x400922d9:0x3ffbd120 0x40090b2a:0x3ffbd140
  #0  0x401be176:0x3ffbd0e0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #1  0x40191cfb:0x3ffbd100 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #2  0x400922d9:0x3ffbd120 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #3  0x40090b2a:0x3ffbd140 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

h2zero, 2022-01-19:
That looks like a completely different error, hmm...

thorrak, 2022-01-19:
For what it's worth, it's no longer crashing consistently within 3-4 GATT reads/writes. I'm struggling to get it to crash after the one I quoted above.

h2zero, 2022-01-19:
I think I know the fix for this, would you like to volunteer for testing? I'll make a branch for it shortly.

thorrak, 2022-01-19:
Of course!

h2zero, 2022-01-19:
Thanks! Here it is:
https://github.com/h2zero/NimBLE-Arduino/tree/bugfix-test

This is based on 1.3.5 instead of master.

thorrak, 2022-01-19:
It's back to crashing consistently.

```
     13280 V: Read port 0 as value: 18925
Preapring to disconnectGuru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40093786  PS      : 0x00060330  A0      : 0x800932c6  A1      : 0x3ffcc4d0  
A2      : 0x3ffe252e  A3      : 0x00000000  A4      : 0x3ffbd394  A5      : 0x00000000  
A6      : 0x3ffbd41c  A7      : 0x3ffbb170  A8      : 0x0000b33f  A9      : 0x00133ffe  
A10     : 0x25600000  A11     : 0x3ffcc500  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00134002  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40093786:0x3ffcc4d0 0x400932c3:0x3ffcc4f0 0x40093393:0x3ffcc530 0x40090b2a:0x3ffcc560
  #0  0x40093786:0x3ffcc4d0 in uxListRemove at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:215
  #1  0x400932c3:0x3ffcc4f0 in prvProcessReceivedCommands at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #2  0x40093393:0x3ffcc530 in prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/timers.c:485
  #3  0x40090b2a:0x3ffcc560 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

h2zero, 2022-01-19:
That's very strange. I can't think of any other differences between the master branch and 1.3.5 that are related. I'll rebase the IPC changes to the master branch and see if that eliminates the issues.

h2zero, 2022-01-19:
Ok 1 more try, new branch:
https://github.com/h2zero/NimBLE-Arduino/tree/bugfix-test2

thorrak, 2022-01-19:
It's back to being stable - as it was with master after the first crash. I'll keep trying to get it to crash, but this may fix it.

h2zero, 2022-01-19:
Great, that was a bug that I have heard of before and thought those changes would fix it but I had no way to test. It's very timing related so if you don't have any crashing I would appreciate the feedback so I can incorporate the changes.

I have also identified a patch in the master branch related to the original crashing: 4cacd370bcd252c0fecc91a9c92b356a76ae9136 I will update the release with this as well

svdrummer, 2022-01-19:
I wonder how good the free RTOS system really is. In line testing is
showing the ESP32 is not slower than the ESP8266. Not to mention all the
time library writers are spending of fixing things.

h2zero, 2022-01-19:
@svdrummer To be fair, it has worked really well up until recently, but in this case it was my fault for not back porting the full fix to the release branch. The newer issues were do to espressif using a newer FreeRTOS version that has some flaws  (thanks amazon?). Those flaws have been fixed in the upstream FreeRTOS releases but espressif is behind on this and I have raised the issues, so far without response.

h2zero, 2022-01-19:
@thorrak Just pushed release 1.3.6, hopefully all is well now 😄

m-menon, 2022-01-21:
I had the exact same issue as mentioned in the issue. I am using PlatformIO and for the time being I decided to rollback to 1.3.4. I will wait for PlatformIO to show up 1.3.6 in the updates and then try that out. I will test and post should I come across any issues. 

Btw  can't thank @h2zero enough for this excellent library. It's not just one of the best libraries but his support in getting issues solved has been nothing short of AWESOME!!!

thorrak, 2022-01-21:
> I had the exact same issue as mentioned in the issue. I am using PlatformIO and for the time being I decided to rollback to 1.3.4. I will wait for PlatformIO to show up 1.3.6 in the updates and then [...]

1.3.6 seems to be working perfectly for me thus far!

You may already know this, but if you want to use it before Platformio's registry picks it up, you can replace `h2zero/NimBLE-Arduino @ 1.3.5` with `https://github.com/h2zero/NimBLE-Arduino.git#1.3.6` in platformio.ini

h2zero, 2022-01-21:
> I had the exact same issue as mentioned in the issue. I am using PlatformIO and for the time being I decided to rollback to 1.3.4. I will wait for PlatformIO to show up 1.3.6 in the updates and then try that out. I will test and post should I come across any issues.

If it's still not up I guess I'll start manually publishing releases to pio, which is a shame, I liked that it's been automatic so far.

> Btw can't thank @h2zero enough for this excellent library. It's not just one of the best libraries but his support in getting issues solved has been nothing short of AWESOME!!!

Thanks! You're very welcome.

> 1.3.6 seems to be working perfectly for me thus far!

That's what I like to hear! 😃

m-menon, 2022-01-21:
> 1.3.6 seems to be working perfectly for me thus far!
>
> You may already know this, but if you want to use it before Platformio's registry picks it up, you can replace `h2zero/NimBLE-Arduino @ 1.3.5` [...]

Thanks for that github link tip @thorrak - I didn't really know that. I will try out that method. I learnt something new today :v:

And to @h2zero - thanks a ton. You are awesome! :thumbsup:

h2zero, 2022-01-21:
> There is probably a way to do this automatically using GitHub actions if you end up going down that route, for what it's worth

You're probably right. I've never manually published to the pio registry before so I'm need to read up on it.

h2zero, 2022-01-22:
I manually published the release to the platformio registry. It could take 24hrs, let's see if it works 😄. 
I'll get on the github actions for this process for the next release.

h2zero, 2022-01-22:
Looks like it was instantly updated, hopefully all is well now.

h2zero, 2022-04-04:
Closing as resolved.

closed 2022-04-04 by h2zero