h2zero/NimBLE-Arduino #341: HEX Value received is not what is expected

peetjh, 2022-01-27:
In my sketch I am sending values over BLE and the HEX value received is not what it should be. It is sending two additional zeros on the end of the HEX value. Below is a portion of my code for testing. The last couple lines are my set value commands, I put it in loop with a delay so you can see it change.

`#include <BLE2902.h>
#include <NimBLEDevice.h>

NimBLECharacteristic* pCharacteristic = NULL;
NimBLECharacteristic* pCharacteristic2 = NULL;
bool deviceConnected = false;
bool oldDeviceConnected = false;
int ledPin = 2;
int ledState = LOW;

static NimBLEServer* pServer;

class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
        deviceConnected = true;
        NimBLEDevice::startAdvertising();
    };
      
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
        pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
    };
    void onDisconnect(NimBLEServer* pServer) {
        //Serial.println("Client disconnected - start advertising");
        deviceConnected = false;    
        NimBLEDevice::startAdvertising();
    };
    void onMTUChange(uint16_t MTU, ble_gap_conn_desc* desc) {
    };
};

/** Handler class for descriptor actions */    
class DescriptorCallbacks : public NimBLEDescriptorCallbacks {
    void onWrite(NimBLEDescriptor* pDescriptor) {
        std::string dscVal((char*)pDescriptor->getValue(), pDescriptor->getLength());
    };

    void onRead(NimBLEDescriptor* pDescriptor) {
    };
};

void setup() {
  Serial.begin(115200);
    pinMode(2, OUTPUT); 
    digitalWrite(2, LOW); 
  
    /** sets device name */
    NimBLEDevice::init("ESP32");

    /** Optional: set the transmit power, default is 3db */
    //NimBLEDevice::setPower(ESP_PWR_LVL_P6); /** +6db */

    pServer = NimBLEDevice::createServer();
    pServer->setCallbacks(new ServerCallbacks());

    NimBLEService* pService = pServer->createService("331F1570-8A00-9A28-CBE1-E1D4212D53EB");
    pCharacteristic = pService->createCharacteristic("331F1571-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);

    pCharacteristic2 = pService->createCharacteristic("331F1572-8A00-9A28-CBE1-E1D4212D53EB",
             NIMBLE_PROPERTY::WRITE_NR);
    pService->start();

    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();

    pAdvertising->addServiceUUID(pService->getUUID());
    pAdvertising->setScanResponse(true);
    pAdvertising->start();
}

void reset()
{
}

void loop() {
     //FLASH LED IF 1 SECOND HAS PASSED
    {
      if (deviceConnected == false){
        if(ledState == LOW){
          ledState = HIGH;
        }
        else{
          ledState = LOW;
          digitalWrite(ledPin, ledState);
          Serial.println(ledState); 
          delay(1000);
        }
      } 
      else{
        ledState = HIGH;
        digitalWrite(ledPin, ledState);
        Serial.println(ledState);
        pCharacteristic->setValue("3.6@");pCharacteristic->notify(); delay(2500);
        pCharacteristic->setValue("5.6@");pCharacteristic->notify(); delay(2500);
        }
      } 
}`

h2zero, 2022-01-28:
>setValue("5.6@")

Try this by wrapping the value in a std::string construction.

`setValue(std::string("5.6@"));`

peetjh, 2022-01-28:
perfect, thanks for the assist.

h2zero, 2022-01-28:
You're welcome. In the future this will not be needed as I have created a new class for handling values.

h2zero, 2022-04-04:
New class implemented in master branch. Closing this now as resolved.

closed 2022-04-04 by h2zero