h2zero/NimBLE-Arduino #342: Request of setting notify_callback as class instance.

asukiaaa, 2022-01-29:


sivar2311, 2022-01-29:
You can use `std::bind` to set a class member function as callback function:

```C++
  void charaSubscribeNotification(NimBLERemoteCharacteristic* pChara) {
    if (pChara->canNotify()) {
      charaPrintId(pChara);
      Serial.println(" canNotify ");
      if (pChara->subscribe(true, std::bind(&Controller::notifyCB, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3, std::placeholders::_4), true)) { 
        Serial.println("set notifyCb");
      } else {
        Serial.println("failed to subscribe");
      }
    }
```

asukiaaa, 2022-01-29:
@sivar2311 
Thank you for the information.
I could call!

I tried calling scanEndCB with using the bind function but I cannot pass compilcation.
Could you give me some advice?
```cpp
class Controller {
  void startScan() {
    scanning = true;
    auto pScan = NimBLEDevice::getScan();
    pScan->setAdvertisedDeviceCallbacks(advDeviceCBs);
    pScan->setInterval(45);
    pScan->setWindow(15);
    Serial.println("Start scan");
    // pScan->start(scanTime, scanEndedCB);
    pScan->start(scanTime,
                 std::bind(&Core::scanEndedCB, this, std::placeholders::_1)); // <<<<<<<<<< ======== Here
  }

  void scanEndedCB(NimBLEScanResults results) {
    Serial.println("Scan Ended");
    scanning = false;
  }
};
```

I got the following error.
```
In file included from src/main.cpp:3:0:
/home/asuki/pio/shared/XboxSeriesXControllerESP32/src/XboxSeriesXControllerESP32.hpp: In member function 'void XboxSeriesXControllerESP32::Core::startScan()':
/home/asuki/pio/shared/XboxSeriesXControllerESP32/src/XboxSeriesXControllerESP32.hpp:203:76: error: no matching function for call to 'NimBLEScan::start(uint32_t&, std::_Bind_helper<false, void (XboxSeriesXControllerESP32::Core::*)(NimBLEScanResults), XboxSeriesXControllerESP32::Core*, const std::_Placeholder<1>&>::type)'
                  std::bind(&Core::scanEndedCB, this, std::placeholders::_1));
                                                                            ^
In file included from /home/asuki/pio/shared/NimBLE-Arduino/src/NimBLEDevice.h:22:0,
                 from /home/asuki/pio/shared/XboxSeriesXControllerESP32/src/XboxSeriesXControllerESP32.hpp:3,
                 from src/main.cpp:3:
/home/asuki/pio/shared/NimBLE-Arduino/src/NimBLEScan.h:65:25: note: candidate: bool NimBLEScan::start(uint32_t, void (*)(NimBLEScanResults), bool)
     bool                start(uint32_t duration, void (*scanCompleteCB)(NimBLEScanResults), bool is_continue = false);
```

I triec casting but it did not solve the problem.
```cpp
    pScan->start(scanTime,
                 (void (*)(NimBLEScanResults))std::bind(
                     &Core::scanEndedCB, this, std::placeholders::_1));
```
```
In file included from src/main.cpp:3:0:
/home/asuki/pio/shared/XboxSeriesXControllerESP32/src/XboxSeriesXControllerESP32.hpp: In member function 'void XboxSeriesXControllerESP32::Core::startScan()':
/home/asuki/pio/shared/XboxSeriesXControllerESP32/src/XboxSeriesXControllerESP32.hpp:204:69: error: invalid cast from type 'std::_Bind_helper<false, void (XboxSeriesXControllerESP32::Core::*)(NimBLEScanResults), XboxSeriesXControllerESP32::Core*, const std::_Placeholder<1>&>::type {aka std::_Bind<std::_Mem_fn<void (XboxSeriesXControllerESP32::Core::*)(NimBLEScanResults)>(XboxSeriesXControllerESP32::Core*, std::_Placeholder<1>)>}' to type 'void (*)(NimBLEScanResults)'
                      &Core::scanEndedCB, this, std::placeholders::_1));
                                                                     ^
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1
```

asukiaaa, 2022-01-29:
I found the reason because scanCompleteCB is not defined with std::function.
https://github.com/h2zero/NimBLE-Arduino/pull/343/files

sivar2311, 2022-01-29:
Sorry, I was away for a few hours.

For a "pure" function pointer std::bind does not work.
That's why I prefer to use std::function as a callback parameter.
This also allows the use of lambda functions.

asukiaaa, 2022-01-29:
Thank you for the information.
I also succeeded in compilation with using lambda expression.

```cpp
      if (pChara->subscribe(
              true,
              [this](NimBLERemoteCharacteristic* pRemoteCharacteristic,
                     uint8_t* pData, size_t length, bool isNotify) {
                notifyCB(pRemoteCharacteristic, pData, length, isNotify);
              },
              true)) {
        Serial.println("set notifyCb");
      } else {
        Serial.println("failed to subscribe");
      }
```

`std::function` and `std::bind` are new for me.
Thank you.

h2zero, 2024-06-04:
This should be resolved with the new `NimBLEScan::start` in the master branch which has a new scan callback class.

closed 2024-06-04 by h2zero