h2zero/NimBLE-Arduino #344: Client memory corruption

aovestdipaperino, 2022-01-31:
I don't have a clue yet on why this happens, but very often the client side handle value of a characteristic gets corrupted.
This is the flow:
scan
connect
get service
get characteristic
write value

I added a debug statement in NimBLEClient.
Half of the times, after reboot, the handle value of the characteristic gets corrupted, so notifications don't reach the app.

I plan to add more logging statements trying to catch what causes this corruption, but any pointers help here.

h2zero, 2022-02-02:
>Half of the times, after reboot, the handle value of the characteristic gets corrupted, so notifications don't reach the app.

I'm not sure I understand, what are you rebooting when this happens? Do you have any logs of this?

h2zero, 2024-06-04:
Closing as stale. Please reopen if still having this issue.

closed 2024-06-04 by h2zero