h2zero/NimBLE-Arduino #350: Memory leak after BLE Client

fronik, 2022-02-09:
Hello!
I can not cope for several days with a memory leak after launch BLE Client.
Can u help me?

Test log:
```
Before BLE Client RAM:264440 Timer:10317
After  BLE Client RAM:264312 Timer:15321
Before BLE Client RAM:264312 Timer:25322
After  BLE Client RAM:264252 Timer:30324
Before BLE Client RAM:264252 Timer:40325
After  BLE Client RAM:264192 Timer:45328
Before BLE Client RAM:264192 Timer:55329
After  BLE Client RAM:264132 Timer:60331
Before BLE Client RAM:264132 Timer:70332
After  BLE Client RAM:264072 Timer:75334
Before BLE Client RAM:264072 Timer:85335
After  BLE Client RAM:264012 Timer:90338
Before BLE Client RAM:264012 Timer:100339
After  BLE Client RAM:263952 Timer:105341
Before BLE Client RAM:263952 Timer:115342
After  BLE Client RAM:263892 Timer:120344
Before BLE Client RAM:263892 Timer:130345
After  BLE Client RAM:263832 Timer:135348
Before BLE Client RAM:263832 Timer:145349
After  BLE Client RAM:263772 Timer:150351
Before BLE Client RAM:263772 Timer:160352
After  BLE Client RAM:263712 Timer:165354
Before BLE Client RAM:263712 Timer:175355
After  BLE Client RAM:263652 Timer:180358
Before BLE Client RAM:263652 Timer:190359
After  BLE Client RAM:263592 Timer:195361
Before BLE Client RAM:263592 Timer:205362
After  BLE Client RAM:263532 Timer:210364
Before BLE Client RAM:263532 Timer:220365
After  BLE Client RAM:263472 Timer:225368
```

main.cpp
```
#include <Arduino.h>
//#include <WiFi.h>
#include <NimBLEDevice.h>
 
int temp;       int moisture;       int light;      int conductivity;   int timer;
const char *val;
String Msg;
NimBLEAddress MiFloraMAC("C4:7C:8D:62:09:89");

bool GetSensorData2(NimBLEAddress pAddress){

    NimBLEClient* pClient = nullptr;
    NimBLERemoteService *pService = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicW = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicR = nullptr;
    NimBLEUUID serviceUuid("00001204-0000-1000-8000-00805f9b34fb");
    pClient = NimBLEDevice::createClient();
      /** Set initial connection parameters: These settings are 15ms interval, 0 latency, 120ms timout.
      *  These settings are safe for 3 clients to connect reliably, can go faster if you have less
      *  connections. Timeout should be a multiple of the interval, minimum is 100ms.
      *  Min interval: 12 * 1.25ms = 15, Max interval: 12 * 1.25ms = 15, 0 latency, 51 * 10ms = 510ms timeout
      */
      pClient->setConnectionParams(12,12,0,51);
      /** Set how long we are willing to wait for the connection to complete (seconds), default is 30. */
      pClient->setConnectTimeout(5);
            //(pClient->connect(&device))
            if (pClient->connect(pAddress)) {
                pService = pClient->getService("00001204-0000-1000-8000-00805f9b34fb");
                
                if (pService != nullptr) {
                    pCharacteristicW = pService->getCharacteristic("00001a00-0000-1000-8000-00805f9b34fb");
                    
                    if (pCharacteristicW != nullptr) {
                      
                      if (pCharacteristicW->canWrite()) {
                        
                        uint8_t buf[2] = {0xA0, 0x1F};
                        pCharacteristicW->writeValue(buf, 2, true);

                        pCharacteristicR = pService->getCharacteristic("00001a01-0000-1000-8000-00805f9b34fb");
                        if (pCharacteristicR != nullptr) {

                          if (pCharacteristicR->canRead()) {
                            std::string value = pCharacteristicR->readValue();
                            val = value.c_str();
                            for (int i = 0; i < 16; i++) {
                              Serial.print((int)val[i], HEX);
                              Serial.print(" ");
                            }
                            Serial.println("");
                            if (((val[0] + val[1] * 256) / 10) < 100 && (val[7]) < 101 && (val[3] + val[4] * 256) < 60000 && (val[8] + val[9] * 256) < 5000) {
                              temp = (val[0] + val[1] * 256) / 10;
                              moisture = val[7];
                              light = val[3] + val[4] * 256;
                              conductivity = val[8] + val[9] * 256;
                              Serial.printf("MiFlora: 💧:%i  🌡:%i  💩:%i  🌞:%i", moisture, temp, conductivity, light);   
                              pClient->disconnect();
                              //NimBLEDevice::deleteClient(pClient);
                              return true;
                            }
                          }
                        }
                        // print or do whatever you need with the value
                      }
                    }
                }
            } else {
            // failed to connect
            }
pClient->disconnect();
NimBLEDevice::deleteClient(pClient);
return false;
}

void setup() {
  Serial.begin(115200);
  Serial.println(" ");

  NimBLEDevice::init("");
  NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */
  NimBLEScan *pScan = NimBLEDevice::getScan();
  NimBLEScanResults results = pScan->start(10);
}

void loop(){
timer = esp_timer_get_time();
Msg = ((String)"--> Before BLE Client RAM:" + ESP.getFreeHeap() + " Timer:" + timer/1000);  Serial.println(Msg); //terminal.println(Msg); 
GetSensorData2(MiFloraMAC);
timer = esp_timer_get_time();
Msg = ((String)"--> After  BLE Client RAM:" + ESP.getFreeHeap() + " Timer:" + timer/1000);  Serial.println(Msg););
delay(10000);
}
```

platformio.ini
```
[env:Flora Control]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_speed = 921600
board_build.partitions = min_spiffs.csv
lib_deps = 
	h2zero/NimBLE-Arduino @ 1.3.6
build_flags = -W
```

h2zero, 2022-02-10:
I think your issue is here:
```
pClient->disconnect();
//NimBLEDevice::deleteClient(pClient);
return true;
```

You've returned without deleting the client instance and on the next call you're creating a new one. 
FYI:  you do not need to call disconnect before deleting the client, it's handled internally.

fronik, 2022-02-10:
```
pClient->disconnect();
//NimBLEDevice::deleteClient(pClient);
return true;
```
It does not help.
```
--> Before BLE Client RAM:265384 Timer:10312
--> After  BLE Client RAM:265140 Timer:15317
--> Before BLE Client RAM:265140 Timer:25318
--> After  BLE Client RAM:264960 Timer:30321
--> Before BLE Client RAM:264960 Timer:40322
--> After  BLE Client RAM:264780 Timer:45325
--> Before BLE Client RAM:264780 Timer:55325
--> After  BLE Client RAM:264600 Timer:60328
```

I tested a lot with the disconnect() and deleteClient(pClient) combinations before writed here.

fronik, 2022-02-10:
Additionally

if use `NimBLEAddress MiFloraMAC("00:00:00:00:00:00")` the problem persists.
Leakage occurs before the survey of the sensor

thorrak, 2022-02-10:
> Additionally
> 
> if use `NimBLEAddress MiFloraMAC("00:00:00:00:00:00")` the problem persists. Leakage occurs before the survey of the sensor
> 
> For Example, if use:
> …
> Leakage memory exactly the same

I think that’s what @h2zero was suggesting in his earlier comment - do you still have a problem if you change your code to delete the client? Example:

```
 bool GetSensorData2(NimBLEAddress pAddress){
     NimBLEClient* pClient = nullptr;
    NimBLERemoteService *pService = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicW = nullptr;
     NimBLERemoteCharacteristic *pCharacteristicR = nullptr;
     NimBLEUUID serviceUuid("00001204-0000-1000-8000-00805f9b34fb");
     pClient = NimBLEDevice::createClient();
     //pClient->disconnect();
      NimBLEDevice::deleteClient(pClient);

     return false;
}
```

fronik, 2022-02-10:
> I think that’s what @h2zero was suggesting in his earlier comment - do you still have a problem if you change your code to delete the client? Example:

I tried it
> I tested a lot with the disconnect() and deleteClient(pClient) combinations before writed here

Code:
```
bool GetSensorData2(NimBLEAddress pAddress){
    NimBLEClient* pClient = nullptr;
    NimBLERemoteService *pService = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicW = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicR = nullptr;
    NimBLEUUID serviceUuid("00001204-0000-1000-8000-00805f9b34fb");
    pClient = NimBLEDevice::createClient();
	// pClient->disconnect();
	NimBLEDevice::deleteClient(pClient);
	return false;
}

```
Result:
```
--> Before BLE Client RAM:264560 Timer:10315
--> After  BLE Client RAM:264432 Timer:10315
--> Before BLE Client RAM:264432 Timer:20316
--> After  BLE Client RAM:264372 Timer:20316
--> Before BLE Client RAM:264372 Timer:30317
--> After  BLE Client RAM:264312 Timer:30317
--> Before BLE Client RAM:264312 Timer:40318
--> After  BLE Client RAM:264252 Timer:40318
--> Before BLE Client RAM:264252 Timer:50319
--> After  BLE Client RAM:264192 Timer:50319
--> Before BLE Client RAM:264192 Timer:60320
--> After  BLE Client RAM:264132 Timer:60320
```

If use `NimBLEDevice::deleteClient(pClient);` leake 40 bytes
If use `pClient->disconnect();` leake 180 bytes

fronik, 2022-02-11:
> решает ли PR [# 352](https://github.com/h2zero/NimBLE-Arduino/pull/352) вашу проблему?

Yes, is good work with `NimBLEDevice::deleteClient(pClient);`

Examples:
Code, good working:
```
// pClient->disconnect();
NimBLEDevice::deleteClient(pClient);
```
Result:
```
--> Before BLE Client RAM:264400 Timer:10316
--> After  BLE Client RAM:264332 Timer:10317
--> Before BLE Client RAM:264332 Timer:20318
--> After  BLE Client RAM:264332 Timer:20318
--> Before BLE Client RAM:264332 Timer:30319
--> After  BLE Client RAM:264332 Timer:30319
--> Before BLE Client RAM:264332 Timer:40320
--> After  BLE Client RAM:264332 Timer:40320
--> Before BLE Client RAM:264332 Timer:50321
--> After  BLE Client RAM:264332 Timer:50321
--> Before BLE Client RAM:264332 Timer:60322
--> After  BLE Client RAM:264332 Timer:60322
--> Before BLE Client RAM:264332 Timer:70323
--> After  BLE Client RAM:264332 Timer:70323
```

Code, leak keep:
```
pClient->disconnect();
// NimBLEDevice::deleteClient(pClient);
```
Result:
```
--> Before BLE Client RAM:264400 Timer:10313
--> After  BLE Client RAM:264156 Timer:10314
--> Before BLE Client RAM:264156 Timer:20315
--> After  BLE Client RAM:263976 Timer:20315
--> Before BLE Client RAM:263976 Timer:30316
--> After  BLE Client RAM:263796 Timer:30316
--> Before BLE Client RAM:263796 Timer:40317
--> After  BLE Client RAM:263612 Timer:40317
--> Before BLE Client RAM:263612 Timer:50318
--> After  BLE Client RAM:263432 Timer:50318
--> Before BLE Client RAM:263432 Timer:60319
--> After  BLE Client RAM:263252 Timer:60319
--> Before BLE Client RAM:263252 Timer:70320
--> After  BLE Client RAM:263072 Timer:70320
--> Before BLE Client RAM:263072 Timer:80321
--> After  BLE Client RAM:262892 Timer:80321
```

**And works well on the example of the MiFlora Sensor**
```
bool GetSensorData2(NimBLEAddress pAddress){
    NimBLEClient* pClient = nullptr;
    NimBLERemoteService *pService = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicW = nullptr;
    NimBLERemoteCharacteristic *pCharacteristicR = nullptr;
    
    NimBLEUUID serviceUuid("00001204-0000-1000-8000-00805f9b34fb");
    pClient = NimBLEDevice::createClient();
      /** Set initial connection parameters: These settings are 15ms interval, 0 latency, 120ms timout.
      *  These settings are safe for 3 clients to connect reliably, can go faster if you have less
      *  connections. Timeout should be a multiple of the interval, minimum is 100ms.
      *  Min interval: 12 * 1.25ms = 15, Max interval: 12 * 1.25ms = 15, 0 latency, 51 * 10ms = 510ms timeout
      */
      pClient->setConnectionParams(12,12,0,51);
      /** Set how long we are willing to wait for the connection to complete (seconds), default is 30. */
[... 62 lines cut ...]
--> After  BLE Client RAM:264308 Timer:44941
--> Before BLE Client RAM:264308 Timer:54941
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
--> After  BLE Client RAM:264308 Timer:56592
--> Before BLE Client RAM:264308 Timer:66592
5 1 0 0 0 0 0 0 0 0 2 3C 0 FB 34 9B 
MiFlora: 💧:0  🌡:26   💩:0  🌞:0
--> After  BLE Client RAM:264308 Timer:68011
--> Before BLE Client RAM:264308 Timer:78011
4 1 0 0 0 0 0 0 0 0 2 3C 0 FB 34 9B 
MiFlora: 💧:0  🌡:26   💩:0  🌞:0
--> After  BLE Client RAM:264308 Timer:78871
--> Before BLE Client RAM:264308 Timer:88871
5 1 0 20 0 0 0 0 0 0 2 3C 0 FB 34 9B 
MiFlora: 💧:0  🌡:26   💩:0  🌞:32
--> After  BLE Client RAM:264308 Timer:89881
--> Before BLE Client RAM:264308 Timer:99881
6 1 0 34 0 0 0 0 0 0 2 3C 0 FB 34 9B 
MiFlora: 💧:0  🌡:26   💩:0  🌞:52
--> After  BLE Client RAM:264308 Timer:100921
--> Before BLE Client RAM:264308 Timer:110922
5 1 0 0 0 0 0 0 0 0 2 3C 0 FB 34 9B 
MiFlora: 💧:0  🌡:26   💩:0  🌞:0
--> After  BLE Client RAM:264308 Timer:114901
```

DTTerastar, 2022-02-16:
Now that's i'm calling deleteClient instead of disconnect I think this is going to bite me as well.  Any reason not to merge it into 1.3.7?

h2zero, 2022-02-16:
Both PR's will be merged and released.

closed 2022-02-16 by h2zero