h2zero/NimBLE-Arduino #353: Exception in findAdvField

DTTerastar, 2022-02-12:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ef260  PS      : 0x00060030  A0      : 0x800ef4b3  A1      : 0x3ffe2610
A2      : 0x00000000  A3      : 0x00000016  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x801a9a10  A9      : 0xad567800
A10     : 0x000000ba  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x3f404860
A14     : 0xad567800  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0xad567800  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5

ELF file SHA256: 0000000000000000

Backtrace: 0x400ef260:0x3ffe2610 0x400ef4b0:0x3ffe2630 0x400f3d8e:0x3ffe2650 0x400f565a:0x3ffe28b0 0x400d2448:0x3ffe28d0 0x400dfff1:0x3ffe28f0 0x400e3552:0x3ffe2930 0x400e842d:0x3ffe29a0 0x400e84d9:0x3ffe29e0 0x400e8725:0x3ffe2a00 0x400e6fdd:0x3ffe2a20 0x400ee14e:0x3ffe2a40 0x400df617:0x3ffe2a60 0x400908e6:0x3ffe2a80
  #0  0x400ef260:0x3ffe2610 in NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #1  0x400ef4b0:0x3ffe2630 in NimBLEAdvertisedDevice::getServiceDataCount() at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #2  0x400f3d8e:0x3ffe2650 in BleFingerprint::fingerprint(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #3  0x400f565a:0x3ffe28b0 in BleFingerprint::seen(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #4  0x400d2448:0x3ffe28d0 in BleFingerprintCollection::onResult(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #5  0x400dfff1:0x3ffe28f0 in NimBLEScan::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEScan.cpp:413
  #6  0x400e3552:0x3ffe2930 in ble_gap_disc_report at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_gap.c:6168
      (inlined by) ble_gap_rx_adv_report at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_gap.c:1499
  #7  0x400e842d:0x3ffe29a0 in ble_hs_hci_evt_le_adv_rpt at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:501
  #8  0x400e84d9:0x3ffe29e0 in ble_hs_hci_evt_le_meta at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:301
  #9  0x400e8725:0x3ffe2a00 in ble_hs_hci_evt_process at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:859
  #10 0x400e6fdd:0x3ffe2a20 in ble_hs_event_rx_hci_ev at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble_npl_os.h:109
  #11 0x400ee14e:0x3ffe2a40 in ble_npl_event_run at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c:130
      (inlined by) nimble_port_run at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c:81
  #12 0x400df617:0x3ffe2a60 in NimBLEDevice::host_task(void*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEDevice.cpp:855 (discriminator 1)
  #13 0x400908e6:0x3ffe2a80 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-02-12:
Got another one:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ef260  PS      : 0x00060230  A0      : 0x800ef4b3  A1      : 0x3ffe2b50
A2      : 0x00000000  A3      : 0x00000016  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x801a9a10  A9      : 0xad567800
A10     : 0x000000ba  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x3f404860
A14     : 0xad567800  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0xad567800  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5

ELF file SHA256: 0000000000000000

Backtrace: 0x400ef260:0x3ffe2b50 0x400ef4b0:0x3ffe2b70 0x400f3d8e:0x3ffe2b90 0x400f565a:0x3ffe2df0 0x400d2448:0x3ffe2e10 0x400dfff1:0x3ffe2e30 0x400e3552:0x3ffe2e70 0x400e842d:0x3ffe2ee0 0x400e84d9:0x3ffe2f20 0x400e8725:0x3ffe2f40 0x400e6fdd:0x3ffe2f60 0x400ee14e:0x3ffe2f80 0x400df617:0x3ffe2fa0 0x400908e6:0x3ffe2fc0
  #0  0x400ef260:0x3ffe2b50 in NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #1  0x400ef4b0:0x3ffe2b70 in NimBLEAdvertisedDevice::getServiceDataCount() at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #2  0x400f3d8e:0x3ffe2b90 in BleFingerprint::fingerprint(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #3  0x400f565a:0x3ffe2df0 in BleFingerprint::seen(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #4  0x400d2448:0x3ffe2e10 in BleFingerprintCollection::onResult(NimBLEAdvertisedDevice*) at .pio/libdeps/esp32-verbose/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #5  0x400dfff1:0x3ffe2e30 in NimBLEScan::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEScan.cpp:413
  #6  0x400e3552:0x3ffe2e70 in ble_gap_disc_report at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_gap.c:6168
      (inlined by) ble_gap_rx_adv_report at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_gap.c:1499
  #7  0x400e842d:0x3ffe2ee0 in ble_hs_hci_evt_le_adv_rpt at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:501
  #8  0x400e84d9:0x3ffe2f20 in ble_hs_hci_evt_le_meta at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:301
  #9  0x400e8725:0x3ffe2f40 in ble_hs_hci_evt_process at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c:859
  #10 0x400e6fdd:0x3ffe2f60 in ble_hs_event_rx_hci_ev at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble_npl_os.h:109
  #11 0x400ee14e:0x3ffe2f80 in ble_npl_event_run at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c:130
      (inlined by) nimble_port_run at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/porting/nimble/src/nimble_port.c:81
  #12 0x400df617:0x3ffe2fa0 in NimBLEDevice::host_task(void*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEDevice.cpp:855 (discriminator 1)
  #13 0x400908e6:0x3ffe2fc0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-02-12:
A user thinks the issue started in 1.3.5: 

https://github.com/ESPresense/ESPresense/issues/241#issuecomment-1037249329

h2zero, 2022-02-13:
I'm still unable to reproduce this. I'm not sure what the error could be as that code has not been changed in a very long time.

h2zero, 2022-02-13:
I've opened PR #354, it may or may not help with this. If you want to give it a test and let me know, I'd appreciate it.

DTTerastar, 2022-02-14:
What am I doing wrong?

```
Linking .pio/build/m5atom-matrix/firmware.elf
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onConnParamsUpdateRequest(NimBLEClient*, ble_gap_upd_params const*)':
/Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/vector.tcc:444: multiple definition of `NimBLEClientCallbacks::onConnParamsUpdateRequest(NimBLEClient*, ble_gap_upd_params const*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):/Users/dturner/Code/ESPresense/.pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEClient.cpp:1192: first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onPassKeyRequest()':
NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks16onPassKeyRequestEv+0x0): multiple definition of `NimBLEClientCallbacks::onPassKeyRequest()'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks16onPassKeyRequestEv+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onAuthenticationComplete(ble_gap_conn_desc*)':
NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): multiple definition of `NimBLEClientCallbacks::onAuthenticationComplete(ble_gap_conn_desc*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onAuthenticationComplete(ble_gap_conn_desc*)':
NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): multiple definition of `NimBLEClientCallbacks::onConnect(NimBLEClient*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onAuthenticationComplete(ble_gap_conn_desc*)':
NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): multiple definition of `NimBLEClientCallbacks::onDisconnect(NimBLEClient*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks24onAuthenticationCompleteEP17ble_gap_conn_desc+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClientCallbacks::onConfirmPIN(unsigned int)':
NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks12onConfirmPINEj+0x0): multiple definition of `NimBLEClientCallbacks::onConfirmPIN(unsigned int)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN21NimBLEClientCallbacks12onConfirmPINEj+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClient::dcTimerCb(ble_npl_event*)':
NimBLEClient.cpp:(.text._ZN12NimBLEClient9dcTimerCbEP13ble_npl_event+0x0): multiple definition of `NimBLEClient::dcTimerCb(ble_npl_event*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN12NimBLEClient9dcTimerCbEP13ble_npl_event+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClient::handleGapEvent(ble_gap_event*, void*)':
NimBLEClient.cpp:(.text._ZN12NimBLEClient14handleGapEventEP13ble_gap_eventPv+0x0): multiple definition of `NimBLEClient::handleGapEvent(ble_gap_event*, void*)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN12NimBLEClient14handleGapEventEP13ble_gap_eventPv+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClient::NimBLEClient(NimBLEAddress const&)':
NimBLEClient.cpp:(.text._ZN12NimBLEClientC2ERK13NimBLEAddress+0x0): multiple definition of `NimBLEClient::NimBLEClient(NimBLEAddress const&)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN12NimBLEClientC2ERK13NimBLEAddress+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClient::NimBLEClient(NimBLEAddress const&)':
NimBLEClient.cpp:(.text._ZN12NimBLEClientC2ERK13NimBLEAddress+0x0): multiple definition of `NimBLEClient::NimBLEClient(NimBLEAddress const&)'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(NimBLEClient.cpp.o):NimBLEClient.cpp:(.text._ZN12NimBLEClientC2ERK13NimBLEAddress+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(NimBLEClient.cpp.o): In function `NimBLEClient::deleteServices()':
[... cut ...]
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(npl_os_freertos.c.o):npl_os_freertos.c:(.text.npl_freertos_time_ms_to_ticks+0x0): first defined here
.pio/build/m5atom-matrix/lib1d1/libNimBLE-Arduino@src-4e14b22c5ba97577051470b05d0d81bd.a(npl_os_freertos.c.o):(.data.ble_port_mutex+0x0): multiple definition of `ble_port_mutex'
.pio/build/m5atom-matrix/libaef/libNimBLE-Arduino.a(npl_os_freertos.c.o):(.data.ble_port_mutex+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/m5atom-matrix/firmware.elf] Error 1```

DTTerastar, 2022-02-14:
```
bool BleFingerprint::query()
{
    if (!(allowQuery || rmAsst) || didQuery) return false;
    if (rssi < -90) return false;
    auto now = millis();

    if (now - lastSeenMillis > 5) return false;

    if (now - lastQryMillis < qryDelayMillis) return false;
    didQuery = true;
    lastQryMillis = now;

    bool success = false;

    Serial.printf("%d Query | MAC: %s, ID: %-60s rssi %d\n", xPortGetCoreID(), getMac().c_str(), getId().c_str(), rssi);

    NimBLEClient *pClient = NimBLEDevice::getClientListSize() ? NimBLEDevice::getClientByPeerAddress(address) : nullptr;
    if (!pClient) pClient = NimBLEDevice::getDisconnectedClient();
    if (!pClient) pClient = NimBLEDevice::createClient();
    pClient->setConnectTimeout(5);
    if (pClient->connect(address))
    {
        bool iphone = true;
        if (allowQuery)
        {
            std::string sMdl = pClient->getValue(deviceInformationService, modelChar);
            std::string sName = pClient->getValue(genericAccessService, nameChar);
            iphone = sMdl.find("iPhone") == 0;
            if (!sName.empty() && !sMdl.empty() && sMdl.find(sName) == std::string::npos && sName.compare("Apple Watch") != 0)
            {
                Serial.printf("%d Name  | MAC: %s, ID: %-60s %s\n", xPortGetCoreID(), getMac().c_str(), getId().c_str(), sName.c_str());
                setId(String("name:") + kebabify(sName).c_str(), ID_TYPE_APPLE_NAME);
                success = true;
            }
            else if (!sMdl.empty())
            {
                if (name.isEmpty()) name = sMdl.c_str();
                Serial.printf("%d Model | MAC: %s, ID: %-60s %s\n", xPortGetCoreID(), getMac().c_str(), getId().c_str(), sMdl.c_str());
                setId(String("apple:") + kebabify(sMdl).c_str(), ID_TYPE_APPLE_MODEL);
                success = true;
            }
            else if (!sName.empty())
            {
                if (name.isEmpty()) name = sName.c_str();
            }
        }

        if (rmAsst || iphone) // For some reason we often don't get room asssistants service advertisement
        {
            std::string sRmAst = pClient->getValue(roomAssistantService, rootAssistantCharacteristic);
            if (!sRmAst.empty())
            {
                setId(String("roomAssistant:") + kebabify(sRmAst).c_str(), ID_TYPE_RM_ASST);
                Serial.printf("%d RmAst | MAC: %s, ID: %-60s %s\n", xPortGetCoreID(), getMac().c_str(), getId().c_str(), sRmAst.c_str());
                success = true;
            }
        }
    }

    NimBLEDevice::deleteClient(pClient);

    if (success) return true;

    qryAttempts++;
    Serial.printf("%d QryErr| MAC: %s, ID: %-60s rssi %d, try %d, retry after %dms\n", xPortGetCoreID(), getMac().c_str(), getId().c_str(), rssi, qryAttempts, qryDelayMillis);

    if (qryDelayMillis < 30000)
        qryDelayMillis += (1000 * qryAttempts * qryAttempts);
    else
        qryDelayMillis = 30000;
    didQuery = false;

    return true;
}
```

This bit of code seems to be the root cause of it.

DTTerastar, 2022-02-14:
This one is in findAdvField:

```
1 Query | MAC: 3d09c843b0bd, ID: exp:20                                                       rssi -57
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x400efd6e  PS      : 0x00060330  A0      : 0x800efff2  A1      : 0x3ffea580
A2      : 0x0000005a  A3      : 0x00000002  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffea748  A7      : 0xff000000  A8      : 0x7aad5616  A9      : 0x000000b3
A10     : 0x40000062  A11     : 0x00000001  A12     : 0x00010ee6  A13     : 0x00000000
A14     : 0x3f403d58  A15     : 0x3ffef17c  SAR     : 0x00000008  EXCCAUSE: 0x00000003
EXCVADDR: 0x40000062  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400efd6e:0x3ffea580 0x400effef:0x3ffea5a0 0x400f4456:0x3ffea5c0 0x400f5be2:0x3ffea7d0 0x400d2758:0x3ffea7f0 0x400e0b89:0x3ffea810 0x400e40a6:0x3ffea850 0x400e8fc9:0x3ffea8c0 0x400e9075:0x3ffea900 0x400e92c1:0x3ffea920 0x400e7bc5:0x3ffea940 0x400eec82:0x3ffea960 0x400e015f:0x3ffea980 0x40090c76:0x3ffea9a0
  #0  0x400efd6e:0x3ffea580 in NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned int*) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #1  0x400effef:0x3ffea5a0 in NimBLEAdvertisedDevice::getServiceUUIDCount() at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:753
  #2  0x400f4456:0x3ffea5c0 in BleFingerprint::fingerprint(NimBLEAdvertisedDevice*) at .pio/libdeps/m5atom-matrix/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #3  0x400f5be2:0x3ffea7d0 in BleFingerprint::seen(NimBLEAdvertisedDevice*) at .pio/libdeps/m5atom-matrix/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #4  0x400d2758:0x3ffea7f0 in BleFingerprintCollection::onResult(NimBLEAdvertisedDevice*) at .pio/libdeps/m5atom-matrix/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #5  0x400e0b89:0x3ffea810 in NimBLEScan::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:412
  #6  0x400e40a6:0x3ffea850 in ble_gap_disc_report at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:6359
      (inlined by) ble_gap_rx_adv_report at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:1523
  #7  0x400e8fc9:0x3ffea8c0 in ble_hs_hci_evt_le_adv_rpt at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:501
  #8  0x400e9075:0x3ffea900 in ble_hs_hci_evt_le_meta at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:301
  #9  0x400e92c1:0x3ffea920 in ble_hs_hci_evt_process at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:859
  #10 0x400e7bc5:0x3ffea940 in ble_hs_event_rx_hci_ev at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:120
  #11 0x400eec82:0x3ffea960 in ble_npl_event_run at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:128
      (inlined by) nimble_port_run at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:79
  #12 0x400e015f:0x3ffea980 in NimBLEDevice::host_task(void*) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEDevice.cpp:923 (discriminator 1)
  #13 0x40090c76:0x3ffea9a0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-02-14:
```
1 Query | MAC: fc039f4cfaeb, ID: samsung:fc039f4cfaeb                                         rssi -76
***ERROR*** A stack overflow in task BLE Scan has been detected.
abort() was called at PC 0x4008fc7c on core 1
ets Jun  8 2016 00:22:57
```

h2zero, 2022-02-15:
Thanks, looks like there are a few things to deal with. Looking at your repo I see:
```
build_flags =
	-D CONFIG_BT_NIMBLE_PINNED_TO_CORE=1
```
Any particular reason to set this to core 1? It would be better in my opinion to change the tasks that make calls into the BLE stack run on core 0 and leave this on core 0 so that the wifi and ble core tasks don't have as many sync issues.

```
me-no-dev/AsyncTCP@^1.1.1
```
This is a red flag, async and BLE do not work well together, known and ongoing issue. Not certain it could cause your issues but it is a factor to consider.

It appears that your "BLE scan" task is overflowing it's stack in these logs, bumping that up a bit should resolve a few of these.

thorrak, 2022-02-15:
Are you stopping and restarting scanning? Are you running an active scan? If so, you may be running [into the issue](https://github.com/h2zero/NimBLE-Arduino/issues/302#issuecomment-1031677969) that I had, which resulted in a similar stack trace to the heap corruption one you quoted.

h2zero, 2022-02-15:
Just found a bug in the getService/characteristic/descriptor calls that can cause the stack overflow. Working on a fix for that.

DTTerastar, 2022-02-15:
You're amazing! I was pretty sure I didn't have any recursion in my code so I'm glad you found it!

But the pin to CORE 0 and getting rid of AsyncTCP has so far made everything but the stack overflow go away!  (It's only been an hour though so I reserve the right to find some more GURU's)

DTTerastar, 2022-02-15:
Well I did get one of these:
```
E (276316) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (276316) task_wdt:  - IDLE0 (CPU 0)
E (276316) task_wdt: Tasks currently running:
E (276316) task_wdt: CPU 0: btController
E (276316) task_wdt: CPU 1: IDLE1
E (276316) task_wdt: Aborting.
abort() was called at PC 0x4018b9fc on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbff60 0x4008f8d5:0x3ffbff80 0x4018b9fc:0x3ffbffa0 0x4008dd01:0x3ffbffc0 0x4000bfed:0x3ffb60b0 0x40091a59:0x3ffb60c0 0x40081c7b:0x3ffb60e0 0x4008d95a:0x3ffb6100 0x40192025:0x3ffb6120 0x40192615:0x3ffb6140 0x400908e6:0x3ffb6170
  #0  0x4008f658:0x3ffbff60 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008f8d5:0x3ffbff80 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4018b9fc:0x3ffbffa0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x4008dd01:0x3ffbffc0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x4000bfed:0x3ffb60b0 in ?? ??:0
  #5  0x40091a59:0x3ffb60c0 in vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #6  0x40081c7b:0x3ffb60e0 in interrupt_restore at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1707
  #7  0x4008d95a:0x3ffb6100 in r_global_int_restore at ??:?
  #8  0x40192025:0x3ffb6120 in r_rw_schedule at ??:?
  #9  0x40192615:0x3ffb6140 in btdm_controller_task at ??:?
  #10 0x400908e6:0x3ffb6170 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

DTTerastar, 2022-02-15:
Here's one where the canary stack hit so you can see where the recursion is:
```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (BLE Scan)
Core 0 register dump:
PC      : 0x4008dc7c  PS      : 0x00060936  A0      : 0x3ffdcc10  A1      : 0x3ffdcb50
A2      : 0x00000000  A3      : 0x40123d84  A4      : 0x00000000  A5      : 0x3f403de8
A6      : 0x00000078  A7      : 0x00000000  A8      : 0x00000009  A9      : 0x0000005a
A10     : 0x00000024  A11     : 0x0000002a  A12     : 0xffffffff  A13     : 0x3f40d414
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000019  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x4008dc7c:0x3ffdcb50 0x3ffdcc0d:0x3ffdcf20 0x400edc65:0x3ffdcfe0 0x400e6624:0x3ffdd030 0x400f1661:0x3ffdd090 0x400f16e1:0x3ffdd0c0 0x400f1742:0x3ffdd100 0x400f1742:0x3ffdd140 0x400f1742:0x3ffdd180 0x400f1742:0x3ffdd1c0 0x400f1742:0x3ffdd200 0x400f1742:0x3ffdd240 0x400f1742:0x3ffdd280 0x400f1742:0x3ffdd2c0 0x400f1742:0x3ffdd300 0x400f1742:0x3ffdd340 0x400f1742:0x3ffdd380 0x400f1742:0x3ffdd3c0 0x400f1742:0x3ffdd400 0x400f1742:0x3ffdd440 0x400f1742:0x3ffdd480 0x400f1742:0x3ffdd4c0 0x400f1742:0x3ffdd500 0x400f1742:0x3ffdd540 0x400f1742:0x3ffdd580 0x400f1742:0x3ffdd5c0 0x400f1742:0x3ffdd600 0x400f1742:0x3ffdd640 0x400f1742:0x3ffdd680 0x400f1742:0x3ffdd6c0 0x400f1742:0x3ffdd700 0x400f1742:0x3ffdd740 0x400f1742:0x3ffdd780 0x400f1742:0x3ffdd7c0 0x400f1742:0x3ffdd800 0x400f1742:0x3ffdd840 0x400f1742:0x3ffdd880 0x400f1742:0x3ffdd8c0 0x400f1742:0x3ffdd900 0x400f1742:0x3ffdd940 0x400f1742:0x3ffdd980 0x400f1742:0x3ffdd9c0 0x400f1742:0x3ffdda00 0x400f1742:0x3ffdda40 0x400f1742:0x3ffdda80 0x400f1742:0x3ffddac0 0x400f1742:0x3ffddb00 0x400f1742:0x3ffddb40 0x400f1742:0x3ffddb80 0x400f1742:0x3ffddbc0 0x400f1742:0x3ffddc00
  #0  0x4008dc7c:0x3ffdcb50 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1118
  #1  0x3ffdcc0d:0x3ffdcf20 in ?? ??:0
  #2  0x400edc65:0x3ffdcfe0 in ble_uuid_to_str at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble/host/src/ble_uuid.c:120
  #3  0x400e6624:0x3ffdd030 in ble_gattc_log_uuid at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:4778
      (inlined by) ble_gattc_log_disc_svc_uuid at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:507
      (inlined by) ble_gattc_disc_svc_by_uuid at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gattc.c:1749
  #4  0x400f1661:0x3ffdd090 in NimBLEClient::retrieveServices(NimBLEUUID const*) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #5  0x400f16e1:0x3ffdd0c0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #6  0x400f1742:0x3ffdd100 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #7  0x400f1742:0x3ffdd140 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #8  0x400f1742:0x3ffdd180 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #9  0x400f1742:0x3ffdd1c0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #10 0x400f1742:0x3ffdd200 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #11 0x400f1742:0x3ffdd240 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #12 0x400f1742:0x3ffdd280 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #13 0x400f1742:0x3ffdd2c0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #14 0x400f1742:0x3ffdd300 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #15 0x400f1742:0x3ffdd340 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #16 0x400f1742:0x3ffdd380 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #17 0x400f1742:0x3ffdd3c0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #18 0x400f1742:0x3ffdd400 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #19 0x400f1742:0x3ffdd440 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #20 0x400f1742:0x3ffdd480 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #21 0x400f1742:0x3ffdd4c0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #22 0x400f1742:0x3ffdd500 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
[... cut ...]
  #47 0x400f1742:0x3ffddb40 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #48 0x400f1742:0x3ffddb80 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #49 0x400f1742:0x3ffddbc0 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
  #50 0x400f1742:0x3ffddc00 in NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32-verbose/NimBLE-Arduino/src/NimBLEClient.cpp:496
```

h2zero, 2022-02-15:
Thanks, PR #355 fixes the recursion issue, give that a try and let me know.

DTTerastar, 2022-02-16:
Should we maybe have some sanity checking on the sizes of stuff in findAdvField so at least it doesn't explode?

h2zero, 2022-02-17:
The code should never have reached that point. Somehow the data vector got corrupted. I don't know why this is happening for you.

DTTerastar, 2022-02-21:
I found Framework 3.2 + -D CONFIG_BT_NIMBLE_PINNED_TO_CORE=1 (which WASN'T actually turned on before) made 1.3.3 work perfectly so I'm sticking w/ that for now.

closed 2022-02-21 by DTTerastar

h2zero, 2022-02-21:
Interesting, thanks for the update. I wish it were easier to pin down the issue but there are so many factors involved. Let me know if you come across any further data that may help.