h2zero/NimBLE-Arduino #357: Compile time option to complete disable the dup mac address array handling

DTTerastar, 2022-02-16:
Random crashes when advertising is stopped/restarted:

```Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble)
Core 0 register dump:
PC      : 0x4000c3f5  PS      : 0x00060636  A0      : 0x800e05d0  A1      : 0x3ffe9f20
A2      : 0x3ffe1688  A3      : 0x3ffe168c  A4      : 0xfffffffc  A5      : 0x003ecc85
A6      : 0x00014723  A7      : 0x00014723  A8      : 0x00007b38  A9      : 0x3ffe91c0
A10     : 0x000000a5  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c3f5:0x3ffe9f20 0x400e05cd:0x3ffe9f40 0x400e0742:0x3ffe9f60 0x400e0901:0x3ffe9f90 0x400e3e06:0x3ffe9fd0 0x400e8d29:0x3ffea040 0x400e8dd5:0x3ffea080 0x400e9021:0x3ffea0a0 0x400e7925:0x3ffea0c0 0x400ee9e2:0x3ffea0e0 0x400dfebf:0x3ffea100 0x40090c76:0x3ffea120
  #0  0x4000c3f5:0x3ffe9f20 in ?? ??:0
  #1  0x400e05cd:0x3ffe9f40 in NimBLEAdvertisedDevice** std::__copy_move<false, true, std::random_access_iterator_tag>::__copy_m<NimBLEAdvertisedDevice*>(NimBLEAdvertisedDevice* const*, NimBLEAdvertisedDevice* const*, NimBLEAdvertisedDevice**) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:412
  #2  0x400e0742:0x3ffe9f60 in NimBLEAdvertisedDevice** std::__copy_move_a<true, NimBLEAdvertisedDevice**, NimBLEAdvertisedDevice**>(NimBLEAdvertisedDevice**, NimBLEAdvertisedDevice**, NimBLEAdvertisedDevice**) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:412
      (inlined by) __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > > std::__copy_move_a2<true, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > > >(__gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algobase.h:436
      (inlined by) __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > > std::move<__gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > > >(__gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >, __gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algobase.h:502
      (inlined by) std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::_M_erase(__gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice**, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/vector.tcc:145
      (inlined by) std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> >::erase(__gnu_cxx::__normal_iterator<NimBLEAdvertisedDevice* const*, std::vector<NimBLEAdvertisedDevice*, std::allocator<NimBLEAdvertisedDevice*> > >) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:1147
      (inlined by) NimBLEScan::erase(NimBLEAddress const&) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:428
  #3  0x400e0901:0x3ffe9f90 in NimBLEScan::handleGapEvent(ble_gap_event*, void*) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEScan.cpp:412
  #4  0x400e3e06:0x3ffe9fd0 in ble_gap_disc_report at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:6359
      (inlined by) ble_gap_rx_adv_report at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c:1523
  #5  0x400e8d29:0x3ffea040 in ble_hs_hci_evt_le_adv_rpt at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:501
  #6  0x400e8dd5:0x3ffea080 in ble_hs_hci_evt_le_meta at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:301
  #7  0x400e9021:0x3ffea0a0 in ble_hs_hci_evt_process at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_hci_evt.c:859
  #8  0x400e7925:0x3ffea0c0 in ble_hs_event_rx_hci_ev at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:120
  #9  0x400ee9e2:0x3ffea0e0 in ble_npl_event_run at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:128
      (inlined by) nimble_port_run at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/nimble/porting/nimble/src/nimble_port.c:79
  #10 0x400dfebf:0x3ffea100 in NimBLEDevice::host_task(void*) at .pio/libdeps/m5atom-matrix/NimBLE-Arduino/src/NimBLEDevice.cpp:923 (discriminator 1)
  #11 0x40090c76:0x3ffea120 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

Could there be a compile time option to complete disable the dup mac address array handling?  I have my own code that basically does something similar so it's just extra headache.

h2zero, 2022-02-16:
That wouldn't work well, the library needs to store the device data at least until the scan response comes to call the callback with the complete data.

DTTerastar, 2022-02-16:
Maybe a ring buffer then instead of adding/erasing all the time?

closed 2022-02-21 by DTTerastar