h2zero/NimBLE-Arduino #358: Crash when using setCallbacks(...) with a member variable

phillipo6, 2022-02-18:
Hi,
i'm trying to use this library and i try to encapsulate it inside a class. Now i can't get it to work when passing a callback pointer from a class member.

BluetoothServer Header:
```cpp
#ifndef H_BLE_SERVER
#define H_BLE_SERVER

#include <NimBLEDevice.h>
#include "../settings/settings.h"
#include "ble-callback.h"

class BluetoothServer
{
public:
    void start();
    BluetoothServer(Settings *settings);

private:
    NimBLEServer *pServer;
    NimBLEService *service;
    BleCallback callback;
    NimBLEAdvertising *advertising;
    NimBLE2904 *descriptor;
};

#endif // !H_BLE_SERVER
```

BluetoothServer cpp:
```cpp
#include "ble-server.h"
#include <Arduino.h>
#include "ble-callback.h"

BluetoothServer::BluetoothServer(Settings *settings) : callback(BleCallback(settings))
{
    /** sets device name */
    NimBLEDevice::init("NimBLE-Arduino");

    NimBLEDevice::setPower(ESP_PWR_LVL_P9);
    NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_SC);

    this->pServer = NimBLEDevice::createServer();

    this->service = this->pServer->createService("DEAD");
    NimBLECharacteristic *pBeefCharacteristic = this->service->createCharacteristic(
        "BEEF",
        /** Require a secure connection for read and write access */
        NIMBLE_PROPERTY::READ_ENC |    // only allow reading if paired / encrypted
            NIMBLE_PROPERTY::WRITE_ENC // only allow writing if paired / encrypted
    );

    pBeefCharacteristic->setValue("Burger");
    this->descriptor = (NimBLE2904 *)pBeefCharacteristic->createDescriptor("2904");
    this->descriptor->setFormat(NimBLE2904::FORMAT_UTF8);
    this->descriptor->setCallbacks(&this->callback);
}

void BluetoothServer::start()
{
    Serial.println("start");
    this->service->start();
    this->advertising = NimBLEDevice::getAdvertising();
    /** Add the services to the advertisment data **/
    this->advertising->addServiceUUID(this->service->getUUID());
    /** If your device is battery powered you may consider setting scan response
     *  to false as it will extend battery life at the expense of less data sent.
     */
    this->advertising->setScanResponse(false);
    this->advertising->start();
}
```

h2zero, 2022-02-19:
What error messages are you seeing?

phillipo6, 2022-02-19:
Thanks for your fast response.
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d2503  PS      : 0x00060130  A0      : 0x800d8ff1  A1      : 0x3ffc8950  
A2      : 0x00000000  A3      : 0x3ffc3524  A4      : 0x3ffc89e4  A5      : 0x3ffc8e50  
A6      : 0x8008fea5  A7      : 0x3ffc8950  A8      : 0x00000008  A9      : 0x3ffc8930  
A10     : 0x3ffc6480  A11     : 0x3ffc8e50  A12     : 0x00000018  A13     : 0x3ffc897c  
A14     : 0x3ffc6340  A15     : 0x9118b301  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x8008fead  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d2503:0x3ffc8950 0x400d8fee:0x3ffc89a0 0x400d90c9:0x3ffc89d0 0x400d4cdb:0x3ffc8a10 0x400d4f99:0x3ffc8a50 0x400d56eb:0x3ffc8a70 0x400d4431:0x3ffc8aa0 0x400db251:0x3ffc8ad0 0x400d9c8d:0x3ffc8b00 0x400d9c9f:0x3ffc8b20 0x400e09ee:0x3ffc8b40 0x400d25bf:0x3ffc8b60 0x4008fea2:0x3ffc8b80
```

That's what i receive at the moment a client connects and the esp32 reboots

h2zero, 2022-02-20:
This works fine for me, without seeing all of your code I am guessing that the instance of your `BluetoothServer` class is being deleted at some point, is it a local variable?

phillipo6, 2022-02-20:
Oh god, yes it was a local variable, thanks that fixed the crash, and also thanks for the Read/Write flags.
But now the callback is only called once and then never again

h2zero, 2022-02-20:
Which callback is not being called? I only see one attached to the descriptor and I can see the read callback being called each time I read it's value.

phillipo6, 2022-02-20:
Got it working now, i mixed up descriptor and characteristic, still new to ble, i'm sorry for wasting your time.

closed 2022-02-20 by phillipo6

h2zero, 2022-02-20:
Great! no problem, happy to help :).