h2zero/NimBLE-Arduino #359: system won't allow BLE name change

societyofrobots, 2022-02-18:
I need to change the name of the device during runtime. But, once init'd, later inits won't change the name.

For example, this results in the BLE name of "happy":
```
NimBLEDevice::init("happy");
NimBLEDevice::init("lucky");
```

Perhaps I need to un-init it first? How would I do that?

societyofrobots, 2022-02-19:
> Do you need to change the advertised name or the device name?, they're different things.

I think the advertised name, whatever my phone displays when I try to connect to the device.

> To change the device name you would need to call deinit/init with the new name.

What would be the proper way to deinit? Is it `ble_hs_deinit()`?

h2zero, 2022-02-19:
`NimBLEDevice::deinit()` is the function you want to use.

sivar2311, 2022-02-19:
Sorry for "hijacking"  this issue.

> ... advertised name or the device name?, they're different things.

Can you give a short explanation about the differences?

h2zero, 2022-02-20:
The advertised name is a "name parameter" that is contained in the advertisement packet. The device name is a string that is stored the the "device name characteristic" value and is read upon connection (this is the name that is set from `NimBLEDevice::init() and also used in the advertisement if no other name is specified`).

sivar2311, 2022-02-20:
Thank you for the explanation!

So an advertising device can have a total of 4 names?!
- 0x1800 Generic Access ("device name")
- Advertisement: name
- ScanResponseData name
- ScanResponseData short-name

It may depend on the scanning / connecting device which name is displayed.

sivar2311, 2022-02-20:
@societyofrobots If your issue is about SinricPro BLE Provisioning, please open a new issue [here](https://github.com/sinricpro/non-sdk-issues)
If this is not the case, I have mixed things up and I am sorry.

h2zero, 2022-02-20:
It's generally only 2 names but yes you can do all kinds of things.

The idea is to have the long name in the characteristic and to advertise the short (incomplete) name and to read the full name from the characteristic. The full name can also be advertised though, all depends on how much data you want to advertise. In some cases you would advertise the short name in the advertisement and have the full name in the scan response etc..

sivar2311, 2022-02-20:
@h2zero Thanks again for the explanation. 
BLE is driving me crazy one day - it's very complex.

closed 2024-06-04 by h2zero