h2zero/NimBLE-Arduino #360: BLE device name length

doktordra, 2022-02-25:
Hello, 
I have a problem with device name limited to 8 characters. 
I tried to edit #define CONFIG_BT_NIMBLE_GAP_DEVICE_NAME_MAX_LEN 31, but no luck, stays locked to 8 characters.
Thanks :)

doktordra, 2022-02-25:
Hello, 
Is it possible to have BLE device name longer than 8 characters AND 32 characters long UUIDs?
Thanks!

h2zero, 2022-02-25:
Hi, the advertising packet is limited to 31 bytes. So a 128bit uuid will use 18 of those bytes available for the name -2 bytes for the fields, so the max name length is 11 bytes in this case. 

The same is true for the scan response data as the second uuid will be placed there.

doktordra, 2022-02-26:
Thanks for your answer, but is there any way to bypass that limitation? I have devices already in the wild, with 10 character names and 128bit uuids, and I want to switch to NimBLE. My mobile app might have some issues if I switch to a different naming standards now.
And, shouldn't these 11 bytes be more than just 8 characters, or my calculation is wrong? :)
Thanks again!

h2zero, 2022-02-26:
There is no workaround for this, it is part of the Bluetooth core specification (except Bluetooth 5.0+). Can you share your code used for this? I'm assuming this is working with the Bluedroid library and not here?

h2zero, 2022-02-26:
I don't see why that would happen, 10 chars should be available. Do you have any other data in the advertisements?

h2zero, 2022-04-08:
@doktordra are these devices connectable? If not I may have an answer for this. 

Otherwise I would suggest using the custom data option to set only the name and service uuids to avoid adding the flags data. If these are connectable devices this procedure will work but it's non-spec (flags required if connectable). The original library, I believe, does not include the flags data unless specified, which will free the additional 3 data bytes needed for your name data.

I will be applying a patch to allow the flags to be removed if the device is not connectable in the next release.

closed 2024-06-04 by h2zero